arendst/Tasmota #7838: For the tasmota_IR image, can there be a checkbox for inverted?

posicat, 2020-03-01:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Yes.  I have PCB's for IR control that use an inverted signal as OpenMQTTGateway uses for it's IR.  This parameter can be passed in on new IRsend(pin[GPIO_IRSEND]); by adding a parameter, "true".

**Describe the solution you'd like**  
Would be nice to have a parameter on the configuration screen to support this. I'm working on re-compiling the code on my own machine, but I haven't figured out the Tasmota ecosystem enough yet to contribute code for adding this feature.

**Describe alternatives you've considered**  
Spend more money and make a second PCB that would support Tasmota.  I would like to be able to use the same board for both.  Tasmota is a much better software current, but OpenMQTTGateway has some great features too.

**Additional context**  
This is the link to the PCB I designed, it has IR recieve and transmit, but relies on the inverted signal that OpenMQTTGateway uses ... https://easyeda.com/posicat/OMG_ESP_01_PCB-445e573315b543108581500a4d4974be

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-03-01:
Hi, this is an uncommon feature, so I'm not sure it will be integrated in the standard version.

Still you can easily patch your own version and self-compile, as you mentioned above.

In file: `xdrv_05_irremote_full.ino`, line 51, change from:

```
  irsend = new IRsend(pin[GPIO_IRSEND]); // an IR led is at GPIO_IRSEND
```

to 

```
  irsend = new IRsend(pin[GPIO_IRSEND], true); // an IR led is at GPIO_IRSEND
```

posicat, 2020-03-01:
I've patched that line, but I haven't figure out where to set the compile to include the ability to transmit raw IR, it just keeps trying to interpret it as JSON.

s-hadinger, 2020-03-01:
Make sure to select tasmota-ir or tasmota-ircustom when you compile

posicat, 2020-03-02:
I still can't get it to compile so that it understands the "IRsend command"

Trying "IRsend 38000,1,69,342,171,22,21,22,21,21,22,21,64,21,21,22,21,22,21,21,21,22,63,22,63,22,63,22,21,22,63,22,63,22,63,22,63,22,63,22,21,21,64,21,21,22,21,22,21,21,21,22,21,22,21,22,63,22,21,21,64,21,64,21,64,21,64,21,64,21,1530,342,85,22,1530"  

and just get 

07:23:06 RSL: stat/tasmota/RESULT = {"IRSend":"Invalid JSON"}

I know this isn't the place to ask for support, but where's the best place to take this conversation to figure this out?

Jason2866, 2020-03-02:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-02 by ascillato2

remcomeer89, 2020-12-28:
Same isue ir leds are always on, if set to relay_i (inverted) the leds wil turn on and of. but the irsend has no box to invert the signal. The inverted signal is used becouse the transistor switching the relay is active low.

cconde, 2021-06-08:
I'm having the same trouble, I need to invert the IR output.
Can we have any option to configure a module to use inverted IR transmission?
This is because I'm using a non inverting buffer instead of a transistor to drive the LED

s-hadinger, 2021-06-08:
You need to compile your own binary and change here https://github.com/arendst/Tasmota/blob/10646ac28e602f34175b3ac7e8cb7bd61d8b68ad/tasmota/my_user_config.h#L786