arendst/Tasmota #7839: RULES - values always uppercase

snakuzzo, 2020-03-01:
### PROBLEM DESCRIPTION

RULES...

I created this rule...

```
ON ZbReceived#0x19E1#Power DO publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":%value%}}} ENDIF
```

But for example, when the rule is triggered...

```
tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":false}}}
tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":FALSE}}}
```

The first is the original publish, second is sent by RULES.
It sends `FALSE` but I need to have exaclty `false` (payload_on and payload_off on homeassistant are `true` and `false`)

I also tried recompiling Tasmota to use IF, ELSE, ESLEIF and trying this rule...

```
ON ZbReceived#0x19E1#Power DO IF (%value%==TRUE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":true}}} ELSEIF (%value%==FALSE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":false}}} ENDIF ENDON
```

but the result is...

```
22:48:51 RUL: ZBRECEIVED#0X19E1#POWER performs "IF (FALSE==TRUE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":true}}} ELSEIF (FALSE==FALSE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":false}}} ENDIF"
22:48:51 MQT: stat/sonoff-z2t/RESULT = {"If":"Done"}```
```

...always `true`

Any suggestion?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 8.1.0.10
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode + PlatformIO
- [x] Flashing tools used: OTA Upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

23:08:18 MQT: stat/sonoff-z2t/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":280,"Rules":"ON ZbReceived#0x19E1#Power DO IF (%value%==TRUE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":true}}} ELSEIF (%value%==FALSE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":false}}} ENDIF ENDON"}
23:08:18 MQT: stat/sonoff-z2t/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:08:18 MQT: stat/sonoff-z2t/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

23:09:07 MQT: stat/sonoff-z2t/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff-z2t","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
23:09:07 MQT: stat/sonoff-z2t/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"z2tasmota","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:27:57","StartupUTC":"2020-03-01T21:41:10","Sleep":50,"CfgHolder":4617,"BootCount":92,"BCResetTime":"2020-02-29T18:03:21","SaveCount":172,"SaveAddress":"F7000"}}
23:09:07 MQT: stat/sonoff-z2t/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(tasmota)","BuildDateTime":"2020-03-01T22:37:34","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"325/699"}}
23:09:07 MQT: stat/sonoff-z2t/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SnakeRM_24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008209","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
23:09:07 MQT: stat/sonoff-z2t/STATUS4 = {"StatusMEM":{"ProgramSize":623,"Free":380,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FDAE397","047683A1","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:09:07 MQT: stat/sonoff-z2t/STATUS5 = {"StatusNET":{"Hostname":"sonoff-z2t-4154","IPAddress":"192.168.1.52","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"80:7D:3A:3E:B0:3A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:09:07 MQT: stat/sonoff-z2t/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3EB03A","MqttUser":"snakuzzo","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:09:07 MQT: stat/sonoff-z2t/STATUS7 = {"StatusTIM":{"UTC":"2020-03-01T22:09:07","Local":"2020-03-01T23:09:07","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:31","Sunset":"18:33"}}
23:09:07 MQT: stat/sonoff-z2t/STATUS10 = {"StatusSNS":{"Time":"2020-03-01T23:09:07"}}
23:09:07 MQT: stat/sonoff-z2t/STATUS11 = {"StatusSTS":{"Time":"2020-03-01T23:09:07","Uptime":"0T00:27:57","UptimeSec":1677,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"SnakeRM_24","BSSId":"10:13:31:BC:5D:59","Channel":1,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

22:48:51 MQT: tele/sonoff-z2t/SENSOR = {"ZbReceived":{"0x19E1":{"Power":false,"Endpoint":1,"LinkQuality":55}}}
22:48:51 RUL: ZBRECEIVED#0X19E1#POWER performs "IF (FALSE==TRUE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":true}}} ELSEIF (FALSE==FALSE) publish2 tele/sonoff-z2t/SENSOR {"ZbReceived":{"0x19E1":{"Power":false}}} ENDIF"
22:48:51 MQT: stat/sonoff-z2t/RESULT = {"If":"Done"}
22:48:52 MQT: tele/sonoff-z2t/SENSOR = {"ZbReceived":{"0x19E1":{"Power":true}}} (retained)
```

### TO REPRODUCE
Create a rule on a topic with a value true or false

### EXPECTED BEHAVIOUR
I expect to have `true` if value is `true` and `false` if value is `false`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-02:
It might be an option to humbly ask Home Assistant support to handle ``true`` and ``false`` as case insensitive. And while you're on it you may ask them also very humbly to handle ``on`` and ``off`` also as case insensitive although that may already be asked too much ;-)

snakuzzo, 2020-03-02:
Yes...you're right!
Just few seconds after this thread opened I asked myself "Why? Maybe I this is an HA issue... TRUE and true are the same" :)
I will open an issue on homeassistant project, but about `on` and `off` I think HA correctly manage values now because they are strings.

I close this issue ;)

closed 2020-03-02 by snakuzzo

snakuzzo, 2020-03-02:
frenck from homeassistant answered...

https://github.com/home-assistant/home-assistant/issues/32396#issuecomment-593355468

arendst, 2020-03-02:
Thx for reporting back.