arendst/Tasmota #7848: Wrong units KWh -> kWh

McGiverGim, 2020-03-03:
### PROBLEM DESCRIPTION
When using MQTT discovery, the units in Home Assistant appear as KWh (first uppercase) and not kWh (first lowercase).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Avatto
- [X] Tasmota binary firmware version number used: 8.1.0.10(a837a8a-tasmota)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tuya Convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
08:44:04 MQT: enchufeavatto/stat/RESULT = {"NAME":"Avatto OT06","GPIO":[17,0,0,0,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":49}
08:44:04 MQT: enchufeavatto/stat/RESULT = {"Module":{"0":"Avatto OT06"}}
08:44:04 MQT: enchufeavatto/stat/RESULT = {"GPIO0":{"17":"Botón1"},"GPIO1":{"0":"Ninguno"},"GPIO2":{"0":"Ninguno"},"GPIO3":{"0":"Ninguno"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"Ninguno"},"GPIO10":{"0":"Ninguno"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"56":"Led1i"},"GPIO14":{"21":"Relé1"},"GPIO15":{"0":"Ninguno"},"GPIO16":{"0":"Ninguno"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
 Not used
```
- [X] Provide the output of this command: `Status 0`:
```
08:45:02 MQT: enchufeavatto/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Enchufe Avatto"],"Topic":"enchufeavatto","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:45:02 MQT: enchufeavatto/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-ES.bin","RestartReason":"Software/System restart","Uptime":"0T00:36:34","StartupUTC":"2020-03-03T07:08:28","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2020-03-02T10:23:55","SaveCount":333,"SaveAddress":"FB000"}}
08:45:02 MQT: enchufeavatto/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(a837a8a-tasmota)","BuildDateTime":"2020-03-02T00:04:06","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"365/699"}}
08:45:02 MQT: enchufeavatto/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PIMINET_N",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
08:45:02 MQT: enchufeavatto/stat/STATUS4 = {"StatusMEM":{"ProgramSize":575,"Free":428,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":0,"Features":["0000040A","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:45:02 MQT: enchufeavatto/stat/STATUS5 = {"StatusNET":{"Hostname":"enchufeavatto-6446","IPAddress":"192.168.100.4","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"CC:50:E3:74:99:2E","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
08:45:02 MQT: enchufeavatto/stat/STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_74992E","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:45:02 MQT: enchufeavatto/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-03-03T07:45:02","Local":"2020-03-03T08:45:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:27","Sunset":"18:37"}}
08:45:02 MQT: enchufeavatto/stat/STATUS9 = {"StatusPTH":{"PowerDelta":101,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:45:02 MQT: enchufeavatto/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-03T08:45:02","ENERGY":{"TotalStartTime":"2020-03-01T19:45:53","Total":1.358,"Yesterday":0.678,"Today":0.046,"Power":4,"ApparentPower":14,"ReactivePower":13,"Factor":0.27,"Voltage":225,"Current":0.061}}}
08:45:02 MQT: enchufeavatto/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-03T08:45:02","Uptime":"0T00:36:34","UptimeSec":2194,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"PIMINET_N","BSSId":"34:DA:B7:F9:A8:44","Channel":1,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
This is the discovery message for example for Energy Today. The value `unit_of_meas` is `KWh` uppercase.
```
08:40:34 MQT: homeassistant/sensor/74992E_ENERGY_Today/config = {"name":"Enchufe Avatto ENERGY Today","stat_t":"enchufeavatto/tele/SENSOR","avty_t":"enchufeavatto/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"74992E_ENERGY_Today","device":{"identifiers":["74992E"],"connections":[["mac","CC:50:E3:74:99:2E"]]},"unit_of_meas":"KWh","dev_cla":"power","frc_upd":true,"val_tpl":"{{value_json['ENERGY']['Today']}}"} (Grabado)
```

### TO REPRODUCE
Use MQTT discovery with Home Assistant.

### EXPECTED BEHAVIOUR
The unit must be kWh lowercase.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

effelle, 2020-03-03:
Will be changed on next update, thanks for reporting.

ascillato2, 2020-03-12:
Closing as has been added. Thanks.

closed 2020-03-12 by ascillato2