arendst/Tasmota #7855: WebGui Channel1 instead of Channel3

catalintabirca, 2020-03-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
SetOption68 1. Configured relay1, relay2, pwm1, pwm2
In the console pwm1 is power3 / channel3
In the WebGui the slider for pwm1 issues the command Channel1 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____nodeMCU, sonoffBasic, Arilux03
- [x ] Tasmota binary firmware version number used: _____8.1.0.10 custom ( olso tested on 8.1.0.3(338499d-sensors))
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: _____gitpod
- [x ] Flashing tools used: _____tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
22:32:07 MQT: stat/esp8266/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"18":"Button2"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"37":"PWM1"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"29":"Relay1i"},"GPIO13":{"30":"Relay2i"},"GPIO14":{"39":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"11":"Switch3"}}
stat/esp8266/RESULT = {"NAME":"ZatarraTest","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
stat/esp8266/RESULT = {"Module":{"0":"ZatarraTest"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:33:14 MQT: stat/esp8266/STATUS = {"Status":{"Module":0,"FriendlyName":["zatarracustom","zatarracustom2","zatarracustom3","zatarracustom4","",""],"Topic":"esp8266","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":180,"SaveState":0,"SwitchTopic":"0","SwitchMode":[2,2,2,2,2,2,2,2],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
22:33:14 MQT: stat/esp8266/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://tabirca/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:36:54","StartupUTC":"2020-03-04T19:56:20","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-03-04T19:58:04","SaveCount":19,"SaveAddress":"3F8000"}}
22:33:14 MQT: stat/esp8266/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(zatarra)","BuildDateTime":"2020-03-04T17:40:48","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"340/699"}}
22:33:14 MQT: stat/esp8266/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rebel",""],"TelePeriod":300,"Resolution":"A58180C0","SetOption":["0400822A","2805C8000100060000005A00000000000000","00040000","00000000"]}}
22:33:14 MQT: stat/esp8266/STATUS4 = {"StatusMEM":{"ProgramSize":607,"Free":2464,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FAAA38F","041131A1","820114C5","010004F1","70000B21","00000004"],"Drivers":"1,2,3,4,5,8,9,10,13,14,17,20,21,24,25,27,28,29,34,37","Sensors":"1,2,3,5,6,9,10,13,22,28,29,33,63"}}
22:33:14 MQT: stat/esp8266/STATUS5 = {"StatusNET":{"Hostname":"esp8266-2448","IPAddress":"192.168.1.126","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:C2:C9:90","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
22:33:14 MQT: stat/esp8266/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1884,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C2C990","MqttUser":"zatarra","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:33:14 MQT: stat/esp8266/STATUS7 = {"StatusTIM":{"UTC":"2020-03-04T20:33:14","Local":"2020-03-04T22:33:14","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+02:00","Sunrise":"06:47","Sunset":"18:07"}}
22:33:14 MQT: stat/esp8266/STATUS10 = {"StatusSNS":{"Time":"2020-03-04T22:33:14","Switch3":"ON","ANALOG":{"A0":8}}}
22:33:14 MQT: stat/esp8266/STATUS11 = {"StatusSTS":{"Time":"2020-03-04T22:33:14","Uptime":"0T00:36:54","UptimeSec":2214,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Channel3":40,"POWER4":"OFF","Channel4":85,"POWER5":"OFF","Channel5":100,"Color":"66D9FF","Wifi":{"AP":1,"SSId":"arielrebel","BSSId":"E8:94:F6:D4:28:1B","Channel":1,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:10"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
22:34:25 CMD: channel3
22:34:25 SRC: WebConsole from 192.168.1.147
22:34:25 CMD: Group 0, Index 3, Command "CHANNEL", Data ""
22:34:25 MQT: stat/esp8266/RESULT = {"Channel3":46}

//moving first slider, for pwm1 (power3 / channel3):
22:34:44 CMD: Group 0, Index 1, Command "CHANNEL", Data "70"
22:34:44 MQT: stat/esp8266/RESULT = {"Command":"Error"}

//moving third slider, for pwm3:
22:35:43 CMD: Group 0, Index 3, Command "CHANNEL", Data "76"
22:35:43 SRC: Light
22:35:43 MQT: stat/esp8266/RESULT = {"POWER3":"ON","Channel3":76,"POWER4":"OFF","Channel4":85,"POWER5":"OFF","Channel5":100,"Color":"75D9FF"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
configure relay1, relay2, pwm1, pwm2 and setoption68 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
In the WebGui pwm1's channel number should be the same with power number (power3 -> channel3) just like in the console

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-05:
iirc as designed. If there are relais the get the lower "Power" numbers AFTER that the counting of the PWM channels does begin.

ascillato2, 2020-03-05:
Closing as it has been fixed. Thanks.

closed 2020-03-05 by ascillato2

catalintabirca, 2020-03-05:
Compiled new version. Tested. All ok. Thank you Theo! Thank you Tasmota!