arendst/Tasmota #7856: Is it possible to strobe/flash/blink a RGBW bulb

xbmcnut, 2020-03-05:
**Have you looked for this feature in other issues and in the docs?**  
Couldn't find anything and searching for 'flash' returns a few too many results :)
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I have just flashed a Arilux E27 bulb with 8.1.0 and it works great in Home Assistant with the standard Ailight (27) module settings but I'd like to be able to flash the bulb for a few sections as an alert. I was doing this with the Ailight firmware but it was very unreliable so I've moved to Tasmota like all my other devices.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._
Previous MQTT config had this:
```
    effect: true
    effect_list: [colorfade_slow, colorfade_fast, flash]
    optimistic: false
```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-05:
You can achieve this using rules. See docu for more information.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

xbmcnut, 2020-03-05:
Thanks. I used `cmnd/office_rgb/backlog` with a payload of `color 3; power 3` and that achieved my desired result.

closed 2020-03-05 by xbmcnut