arendst/Tasmota #7857: Tasmota Counter big jumps, not sure why.

jackkitley, 2020-03-05:
### PROBLEM DESCRIPTION
I setup a counter on tasmota , a counter1 for my pulse water meter. All was running fine for first 2 weeks and i kept adjusting the counterdebounce so it could slow down to match the municipal water meter as i wanted accurate meter readings "real time". 

The problem is now that the counter jumps as crazy increments. I have tried rolling back to version 6.4.0, i have then gone to recent version 8 again but same thing happens all the time. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [ ] Tasmota binary firmware version number used: Tasmota 8.1.0.8 
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: NodeMCU PyFlasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
07:47:26 MQT: stat/water_meter/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
07:47:26 MQT: stat/water_meter/RESULT = {"Module":{"18":"Generic"}}
07:47:26 MQT: stat/water_meter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"42":"Counter1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
07:47:28 MQT: tele/water_meter/STATE = {"Time":"2020-03-05T07:47:28","Uptime":"0T01:59:13","UptimeSec":7153,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"3_stanley_road","BSSId":"18:E8:29:64:5C:09","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

07:48:03 MQT: stat/water_meter/STATUS = {"Status":{"Module":18,"FriendlyName":["Water Meter"],"Topic":"water_meter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:48:03 MQT: stat/water_meter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:59:48","StartupUTC":"2020-03-05T04:48:15","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-03-04T18:37:00","SaveCount":41,"SaveAddress":"FB000"}}
07:48:03 MQT: stat/water_meter/STATUS2 = {"StatusFWR":{"Version":"8.1.0.8(643d3d0-tasmota)","BuildDateTime":"2020-02-16T17:00:13","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"369/699"}}
07:48:03 MQT: stat/water_meter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["3_stanley_road",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
07:48:03 MQT: stat/water_meter/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
07:48:03 MQT: stat/water_meter/STATUS5 = {"StatusNET":{"Hostname":"water_meter-6631","IPAddress":"192.168.88.38","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:82:19:E7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:48:03 MQT: stat/water_meter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.88.54","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8219E7","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:48:03 MQT: stat/water_meter/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 05 06:48:03 2020","Local":"Thu Mar 05 07:48:03 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:23","Sunset":"18:40"}}
07:48:03 MQT: stat/water_meter/STATUS10 = {"StatusSNS":{"Time":"2020-03-05T07:48:03","COUNTER":{"C1":176954}}}
07:48:03 MQT: stat/water_meter/STATUS11 = {"StatusSTS":{"Time":"2020-03-05T07:48:03","Uptime":"0T01:59:48","UptimeSec":7188,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"3_stanley_road","BSSId":"18:E8:29:64:5C:09","Channel":6,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect the counter to go up by 1 per my 5 litres of water used.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

https://www.youtube.com/watch?v=vyXqRemdknI

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I have checked all the wiring of the pulse counter and it is correct, i have double checked it 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-05:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2020-03-05:
Closing as it is a Hardware issue. Noise on measurement. Please address this to the Tasmota support chat for help. Thanks.

closed 2020-03-05 by ascillato2