arendst/Tasmota #7861: Interlock wil not accept 3 groups with 6 relays

martin072, 2020-03-06:
### PROBLEM DESCRIPTION
Interlock does not work beyond relay 4. I have 3 shutters, 6 relays, and trying the command Interlock 1,2 3,4 5,6 gives me the following output: 

```
11:49:51 CMD: interlock 1,2 3,4 5,6
11:49:51 RSL: stat/tasmota/RESULT = {"Interlock":"ON","Groups":"1,2 3,4"}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 8.1.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino IDE
- [x] Flashing tools used: Arduino IDE
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  11:57:09 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:57:10 RSL: stat/tasmota/RESULT = {"Module":{"18":"Generic"}}
11:57:10 RSL: stat/tasmota/RESULT = {"GPIO0":{"25":"Relay5"},"GPIO1":{"0":"None"},"GPIO2":{"21":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"26":"Relay6"},"GPIO13":{"0":"None"},"GPIO14":{"24":"Relay4"},"GPIO15":{"0":"None"},"GPIO16":{"22":"Relay2"}}

```
- [- ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
11:57:55 RSL: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["TASMOTA","TASMOTA2","TASMOTA3","TASMOTA4",""],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:57:55 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://192.168.73.10:80/api/arduino/tasmota.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:16","StartupUTC":"2020-03-06T10:48:39","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":46,"SaveAddress":"3F6000"}}
11:57:55 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2020-03-06T11:00:14","Boot":31,"Core":"2_6_3","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"319/699"}}
11:57:55 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.73.10","LogPort":514,"SSId":["UPC975567",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
11:57:55 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":3448,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":2,"Features":["00000809","87CAA397","003C83A1","000004C5","050000C4","00007981"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,20,21,22,24,27,29","Sensors":"1,2,3,4,5,6,9,31,45"}}
11:57:55 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7744","IPAddress":"192.168.73.132","Gateway":"192.168.73.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.4.4","Mac":"3C:71:BF:31:5E:40","Webserver":2,"WifiConfig":5}}
11:57:55 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_315E40","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:57:55 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 06 10:57:55 2020","Local":"Fri Mar 06 11:57:55 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99,"Sunrise":"07:14","Sunset":"18:30"}}
11:57:55 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-06T11:57:55","Shutter1":{"Position":5,"direction":0},"Shutter2":{"Position":0,"direction":0},"Shutter3":{"Position":0,"direction":0}}}
11:57:55 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-06T11:57:55","Uptime":"0T00:09:16","UptimeSec":556,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"UPC975567","BSSId":"DE:0D:17:AD:89:38","Channel":7,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
11:58:59 CMD: weblog 4
11:58:59 RSL: stat/tasmota/RESULT = {"WebLog":4}
11:59:00 CFG: Saved to flash at 3F5, Count 47, Bytes 4096
11:59:09 WIF: Checking connection...
11:59:09 WIF: Connected
11:59:09 CMD: interlock 1,2 3,4 5,6
11:59:09 SRC: WebConsole from 192.168.73.102
11:59:09 CMD: Group 0, Index 1, Command "INTERLOCK", Data "1,2 3,4 5,6"
11:59:09 RSL: stat/tasmota/RESULT = {"Interlock":"ON","Groups":"1,2 3,4"}
```

### TO REPRODUCE
Define 6 relays, user interlock command to create 3 groups

### EXPECTED BEHAVIOUR
Should be able to create 3  separate locking groups with 6 relays

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

martin072, 2020-03-06:
my Bad, one relay was not configured.. closing the issue

closed 2020-03-06 by martin072

catalintabirca, 2020-03-06:
11:57:10 RSL: stat/tasmota/RESULT = {"GPIO0":{"25":"Relay5"},"GPIO1":{"0":"None"},"GPIO2":{"21":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"26":"Relay6"},"GPIO13":{"0":"None"},"GPIO14":{"24":"Relay4"},"GPIO15":{"0":"None"},"GPIO16":{"22":"Relay2"}}

you do not have relay3 defined