arendst/Tasmota #7866: device Sonoff Pow R2 Modul or Gosund SHP5 Modul restart sporadicly after switch on load

Ajimaru, 2020-03-07:
Please DO NOT OPEN AN ISSUE:
  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
  - If you have an issue when flashing was done via Tuya Convert, please address it to Tuya Convert Team
  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://tasmota.github.io/docs/#/help/FAQ) and [Troubleshooting](https://tasmota.github.io/docs/#/help/Troubleshooting)

 Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): **Sonoff Pow R2 Modul** & **Gosund SHP5 Modul**
- [x] Tasmota binary firmware version number used: 8.1.0.10
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: ESPEasy-tasmota-dot-info
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```

  Configuration output here for **Sonoff Pow R2 Modul**:

08:27:40 CMD: Backlog Template; Module; GPIO 255
08:27:40 SRC: WebConsole from 192.168.1.113
08:27:40 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
08:27:40 SRC: Backlog
08:27:40 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
08:27:40 MQT: stat/Tasmota-Entertainment/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:27:41 SRC: Backlog
08:27:41 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
08:27:41 MQT: stat/Tasmota-Entertainment/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
08:27:41 SRC: Backlog
08:27:41 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
08:27:41 MQT: stat/Tasmota-Entertainment/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

  Configuration output here for **Gosund SPH5 Modul**:

08:29:21 CMD: Backlog Template; Module; GPIO 255
08:29:21 SRC: WebConsole from 192.168.1.113
08:29:21 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
08:29:21 SRC: Backlog
08:29:21 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
08:29:21 MQT: stat/Tasmota-GamingPC/RESULT = {"NAME":"SHP5","GPIO":[57,145,56,146,0,22,0,0,0,0,21,0,17],"FLAG":0,"BASE":18}
08:29:22 SRC: Backlog
08:29:22 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
08:29:22 MQT: stat/Tasmota-GamingPC/RESULT = {"Module":{"0":"SHP5"}}
08:29:22 SRC: Backlog
08:29:22 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
08:29:22 MQT: stat/Tasmota-GamingPC/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"56":"Led1i"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
[... 67 lines cut ...]
08:16:17 MQT: stat/Tasmota-Entertainment/RESULT = {"POWER":"ON"}
08:16:17 MQT: stat/Tasmota-Entertainment/POWER = ON
08:16:17 QPC: Reset
08:16:19 APP: Anzahl Startvorgänge 29
08:16:19 CFG: in Flash gespeichert am F8, zählen 208, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Reboot issue occures if the load is not switched on for some hours (like over night)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

That the relais stays on

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-08:
I cannot reproduce but it might be because of the load. If it's large like 1500W it may surge your power lines leading to some software error in the SHP or Pow.

The once was an issue about this but I cannot find it anymore.

Ajimaru, 2020-03-08:
The load is < 200W. No peak during switching on the relais.
I have more Sonoff and gGsund devices with Tasmota 8.1.0, they runs flawless.
Its an odd behaviour. Let me know if you need a weblog with "more details" log option enabled.

ascillato, 2020-03-08:
The **exception 4** is a problem with interrupts and it is sometimes related to erroneous usage of IRAM memory or interrupts/nointerrups commands. So, as the rest of devices are not having this issue, it seems that your firmware is faulty or had a problem while flashing (flashing problems are common, so don't worry) 

Please, try:

1- Type `reset 5` on the console to erase all flash (but leaving wifi credentials)
2- Upload by OTA latest Tasmota with latest core ( http://thehackbox.org/tasmota/pre-2.7/tasmota.bin ) 
3- Then type `reset 3` (to reset wifi calibration)
4- Do a power cycle in order to let your device recalibrate the wifi. (you need to de-energize your device for some seconds and energize it again) 
5- And finally configure again your device and try again. 

Thanks.

Jason2866, 2020-03-09:
I have several Gosund SP1 (there is no Gosund SHP5!, it is Blitzwolf SHP5)
running latest dev 8.1.0.10 and there is no restart.
As @ascillato already wrote, only two possibilities:
- bad flash / wrong settings
- bad hardware
It is not a bug in Tasmota generating this issue.

ascillato2, 2020-03-09:
Please, when you have time, tell us if the proposed procedure solve your issue. Thanks.

closed 2020-03-09 by ascillato2

Ajimaru, 2020-03-09:
Well Gosund SHP5 was a typo, the Gosund devices are SP1 which are configured as Blitzwolf SHP5 in the Tasmota Firmware.
This behaviour occured with dev 8.1.0.9 & 8.1.0.10 uploaded by OTA.
I have set the loglevel of the serial to 0, the Sonoff and Gosund device with the dev 8.1.0.10 Firmware have not showed the reset issue since.

trichromic-ab, 2022-10-30:
> I cannot reproduce but it might be because of the load. If it's large like 1500W it may surge your power lines leading to some software error in the SHP or Pow.
> 
> The once was an issue about this but I cannot find it anymore.

No large loads at all. Its been happening on 8 different switches with various loads, some with none.

I'm currently following Theo's 2 suggestions and will report back after a bit of soak-testing time.