arendst/Tasmota #7867: Sonoff TH10 sensor problem

BrianDurward, 2020-03-07:
Uploaded the development bin file 8.0.0.10 to a Sonoff TH10 and lost the sensor data.
Reverted to 8.1.0 and the temperature data now returned to normal. 
Module settings Sonoff TH (4)
Sensor DS18x20 (4)

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-07:
Please, could you be so kind on completing the troubleshooting template in order to have more debug information so as to properly help you? Please post the information of the version that don't work for you. Thanks.

Jason2866, 2020-03-07:
Please read the changelog. 
tasmota.bin in development branch does not include sensors anymore.

BrianDurward, 2020-03-07:
The development tasmota-sensors bin didn't work either.

ascillato, 2020-03-07:
@BrianDurward 

Please, could you be so kind on completing the troubleshooting template in order to have more debug information so as to properly help you? Please post the information of the version that don't work for you. Thanks.

closed 2020-03-08 by ascillato2