arendst/Tasmota #7872: MqttLog 0 (default) and MqttLog 2 produce same MQTT traffic

denisfrench, 2020-03-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
MqttLog 0 is the default MQTT logging level (#define MQTT_LOG_LEVEL LOG_LEVEL_NONE in my_user_config.h). MqttLog 0 produces MQT output similar to WebLog 2 RSL messages, which are suppressed with WebLog 0. Switching between MqttLog 0 and 2 produces no change in MQTT output, leaving no way to suppress automatic MQTT info/stat output if required.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 R2
- [ ] Tasmota binary firmware version number used: 8.1.0.10(tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: TasmoCompiler
- [x] Flashing tools used: Tasmotizer 1.1
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
13:30:35 RSL: RESULT = {"NAME":"proSmart BBoil","GPIO":[17,255,255,255,255,255,0,0,21,255,4,255,157],"FLAG":15,"BASE":18}
13:30:36 RSL: RESULT = {"Module":{"0":"proSmart BBoil"}}
13:30:36 RSL: RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"94":"Counter1n"},"GPIO14":{"4":"DS18x20"},"GPIO15":{"0":"None"},"GPIO16":{"157":"LedLink"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
13:32:43 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":15,"Rules":"ON Power1#Boot DO IF (Mem1==0) Event setPoint=75 ENDIF ENDON ON Power1#Boot DO IF (Mem2==0) Event setTime=360 ENDIF ENDON ON Time#Minute|2 DO Event Report ENDON ON Event#setPoint DO Mem1=%value% ENDON ON Event#setTime DO Backlog Mem2=%value%; Mem3=%value%-60; Mem4=%value%-120; Mem5=%value%-180; Mem6=%value%-240 ENDON ON Event#Report DO Publish losant/%topic%/state {"data":{"hwsTemp":%Var1%,"hwsPower":%Var3%,"hwsRelay":%Var2%,"hwsBoost":%Var4%,"hwsSetPoint":%Mem1%, "hwsSetTime":%Mem2%}} ENDON"}
13:32:43 RSL: RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":47,"Rules":"ON Power1#Boot DO Backlog Var1 0; Var2 0; Var3 0; Var4 0 ENDON ON DS18B20#Temperature!=%Var1% DO Var1 %value% ENDON ON Power1#State DO Backlog Var2 %value%; Var4=%Var3%*%value%; Event Report ENDON ON Analog#A0div10<=10 DO Backlog Var3 0; Var4 0; Power OFF; Event Report ENDON ON Analog#A0div10>=90 DO Backlog Var3 1; Var4=%Var2%*1; Event Report ENDON ON Counter#C1!=%Var6% DO IF (%value%==4200) Var5 0; Var6 %value% ELSE Var5=3600/%value%; Var6 %value% ENDIF ENDON"}
13:32:43 RSL: RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Free":15,"Rules":"ON DS18B20#Temperature>%Mem1% IF (Power#state=1) Power OFF; Event Report ENDIF ENDON ON Time#Minute=%Mem5% IF (Power1#state=0) AND DS18B20#Temperature<(%Mem1%-20) Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem4% IF (Power1#state=0) AND DS18B20#Temperature<(%Mem1%-10) Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem3% IF (Power1#state=0) AND DS18B20#Temperature<%Mem1% Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem2% IF (Power1#state=1) Power OFF; Event Report ENDIF ENDON"}
```
- [x] Provide the output of this command: `Status 0`:
```
13:32:43 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":15,"Rules":"ON Power1#Boot DO IF (Mem1==0) Event setPoint=75 ENDIF ENDON ON Power1#Boot DO IF (Mem2==0) Event setTime=360 ENDIF ENDON ON Time#Minute|2 DO Event Report ENDON ON Event#setPoint DO Mem1=%value% ENDON ON Event#setTime DO Backlog Mem2=%value%; Mem3=%value%-60; Mem4=%value%-120; Mem5=%value%-180; Mem6=%value%-240 ENDON ON Event#Report DO Publish losant/%topic%/state {"data":{"hwsTemp":%Var1%,"hwsPower":%Var3%,"hwsRelay":%Var2%,"hwsBoost":%Var4%,"hwsSetPoint":%Mem1%, "hwsSetTime":%Mem2%}} ENDON"}
13:32:43 RSL: RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":47,"Rules":"ON Power1#Boot DO Backlog Var1 0; Var2 0; Var3 0; Var4 0 ENDON ON DS18B20#Temperature!=%Var1% DO Var1 %value% ENDON ON Power1#State DO Backlog Var2 %value%; Var4=%Var3%*%value%; Event Report ENDON ON Analog#A0div10<=10 DO Backlog Var3 0; Var4 0; Power OFF; Event Report ENDON ON Analog#A0div10>=90 DO Backlog Var3 1; Var4=%Var2%*1; Event Report ENDON ON Counter#C1!=%Var6% DO IF (%value%==4200) Var5 0; Var6 %value% ELSE Var5=3600/%value%; Var6 %value% ENDIF ENDON"}
13:32:43 RSL: RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Free":15,"Rules":"ON DS18B20#Temperature>%Mem1% IF (Power#state=1) Power OFF; Event Report ENDIF ENDON ON Time#Minute=%Mem5% IF (Power1#state=0) AND DS18B20#Temperature<(%Mem1%-20) Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem4% IF (Power1#state=0) AND DS18B20#Temperature<(%Mem1%-10) Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem3% IF (Power1#state=0) AND DS18B20#Temperature<%Mem1% Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem2% IF (Power1#state=1) Power OFF; Event Report ENDIF ENDON"}
13:33:22 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(tasmota)","BuildDateTime":"2020-03-08T01:55:34","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"552/699"}}
13:33:22 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WitsEnd","WitsEndN"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008000","2805C8000100060000005A00000000000000","00001000","00000000"]}}
13:33:22 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":551,"Free":2520,"Heap":30,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F8863C3","043603A1","000000C5","010000C0","C000F981","00000004"],"Drivers":"1,2,3,4,6,8,9,10,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
13:33:22 RSL: STATUS5 = {"StatusNET":{"Hostname":"59292a4a1ba24b0007ac7527-1424","IPAddress":"192.168.1.15","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A0:20:A6:01:C5:90","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:33:22 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-03-08T02:33:22","Local":"2020-03-08T13:33:22","StartDST":"2020-10-04T02:00:00","EndDST":"2020-04-05T03:00:00","Timezone":99,"Sunrise":"06:56","Sunset":"19:30"}}
13:33:22 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-03-08T13:33:22","COUNTER":{"C1":123.371212},"ANALOG":{"A0":4},"DS18B20":{"Id":"000473554DFF","Temperature":23.3},"TempUnit":"C"}}
13:33:22 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-03-08T13:33:22","Uptime":"0T00:17:03","UptimeSec":1023,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WitsEnd","BSSId":"10:DA:43:1A:BC:94","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:30"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_

EXAMPLE (WebLog DEFAULT, WebLog 0, then back to WebLog 2):
```
13:36:21 CMD: WebLog
13:36:21 RSL: RESULT = {"WebLog":2}
13:36:32 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 23.8"
13:36:32 RSL: RESULT = {"Var1":"23.8"}
13:36:33 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 24.7"
13:36:33 RSL: RESULT = {"Var1":"24.7"}
13:36:35 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 25.6"
13:36:35 RSL: RESULT = {"Var1":"25.6"}
13:36:38 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 26.1"
13:36:38 RSL: RESULT = {"Var1":"26.1"}
13:36:40 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 26.2"
13:36:40 RSL: RESULT = {"Var1":"26.2"}
13:36:49 CMD: WebLog 0
<NOTE GAP IN LOGGING WITH NIL OUTPUT>
13:37:10 RSL: RESULT = {"WebLog":2}
13:37:13 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 27.8"
13:37:13 RSL: RESULT = {"Var1":"27.8"}
13:37:15 RUL: DS18B20#TEMPERATURE!=%VAR1% performs "Var1 27.7"
13:37:15 RSL: RESULT = {"Var1":"27.7"}
```
[... cut ...]
Likely many users are depending on the default MqttLog 0 behaviour to produce MQTT messages required for their automation systems. I suggest that if MqttLog 0 is rectified to produce no MQTT output, this use case can be accommodated by the default MqttLog level being changed to 2 (#define MQTT_LOG_LEVEL LOG_LEVEL_INFO in my_user_config.h)

I haven't bothered to test MqttLog 1 in my investigation of this issue. Presumably it too is equivalent to MqttLog 2.

I had a look at xdrv_02_mqtt.ino to see if I could resolve this myself, but unfortunately the workings are beyond my basic understanding of C++.

**(Please, remember to close the issue when the problem has been addressed)**

denisfrench, 2020-03-08:
So I thought I should at least look at what MqttLog 3 and 4 actually output.

They too produce no difference in MQTT traffic from the default MqttLog 2 (now confirmed with MQTT.fx). I expected to see the same debug message over MQTT as per WebLog 3 and 4.

Am I missing something, or is MqttLog totally broken?

arendst, 2020-03-08:
You're missing something.

MqttLog allows sending of debug information over Mqtt, hence MqttLog. Like weblog it sends ALL information based on it's log level over MQTT. If you monitor ``stat/device/LOGGING`` you'll see what I mean.

As you seem to have MQTT pricing I suggest you turn off MQTT logging as it adds a lot of extra MQTT traffic.

Take a look TDM which has a MQTT console showing you exactly what I mean.

There is no means of controlling the amount of normal MQTT traffic as there was never a usecase for it. Perhaps you should look for a better MQTT server solution.

closed 2020-03-08 by ascillato2