arendst/Tasmota #7877: BME280 and possibly others: Calculate Dewpoint from temperature and relative humidity

mweinelt, 2020-03-09:
**Have you looked for this feature in other issues and in the docs?**  

Yes, I have.

**Is your feature request related to a problem? Please describe.**  

No, it is not.

**Describe the solution you'd like**  

Calculate the dew point when temperature and relative humidity are known.

**Describe alternatives you've considered**  

Doing the calculation in other parts of my infrastructure, but that would require doing the calculation at multiple places.

**Additional context**  

> If you are interested in a simpler calculation that gives an approximation of dew point temperature if you know the observed temperature and relative humidity, the following formula was proposed in a 2005 article by Mark G. Lawrence in the Bulletin of the American Meteorological Society:
> 
> Td = T - ((100 - RH)/5.)
> 
> where Td is dew point temperature (in degrees Celsius), T is observed temperature (in degrees Celsius), and RH is relative humidity (in percent). Apparently this relationship is fairly accurate for relative humidity values above 50%. 

https://iridl.ldeo.columbia.edu/dochelp/QA/Basic/dewpoint.html

The actual paper is available here: https://journals.ametsoc.org/doi/pdf/10.1175/BAMS-86-2-225

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-09:
Like the idea, but a calculation that is reasonably plausible only for air humidity above 50% is not useful for many users. Better nothing than wrong.
A better approximation can be found at https://en.wikipedia.org/wiki/Dew_point

ascillato2, 2020-03-09:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-09 by ascillato2

mweinelt, 2020-03-09:
Not sure where the duplicate issue actually is. This is the only issue I can find for the terms "dew" and "dewpoint".

ascillato2, 2020-03-09:
@mweinelt 

Yes, you are right. I was confusing this with IAQ. Correcting labels. Thanks.