arendst/Tasmota #7879: Temperature Sensor DS18B20 not detected

peret2000, 2020-03-09:
### PROBLEM DESCRIPTION
I am using several NodeMCU V3 boards, and I have noticed that very frequently (under unknown circumstances) the temperature sensor (DS18B20) is not detected at all, although it is correctly connected (D2 GPIO04, pull-up resistor 4.7k), and the software is configured: Module type: Generic(18), and D2 GPIO4 with DS18x20 (4)

Checks: with standard OneWire code example, it is detected and temperature is measured. Switching to tasmota, it is not detected. But, sometime ago, it was. Maybe the release, or maybe it is detected until you change the sensor with a different one (I have found some blog alerting about it). I am sorry, but circumstances are not clear.

I thought this could be a common problem, but after searching around tasmota project, and out of tasmota, I have not found clear information.

I think tasmota is not relying on external libraries as much as possible, perhaps for a space efficiency, so it is difficult to reproduce the code out of tasmota; specially at the beginning, when it is looking for sensors:
`uint8_t OneWireSearch(uint8_t *newAddr) in xsns_05_ds18x20.ino`

After some test and errors, I have found that adding some debugging code in function `void OneWireReadBit1(void)`, the sensor is detected.

Also, undoing commit df3441 (february, 20th), it also works.

I do not understand very well the changes, maybe a race condition is happening, or who knows. It is complex for me, maybe somebdy that knows better OneWire library can assist.

Thanks!

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used: NodeMCU V3
- [X] Tasmota binary firmware version number used: 8.2.0 (commit c65cc9f1)
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: platformio core (Ubuntu 19.10)
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
18:13:05 MQT: tasmota/stat/test/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
18:13:06 MQT: tasmota/stat/test/RESULT = {"Module":{"18":"Generic"}}
18:13:06 MQT: tasmota/stat/test/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4":"DS18x20"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [X] Provide the output of this command: `Status 0`:
```
18:13:56 MQT: tasmota/stat/test/STATUS = {"Status":{"Module":18,"FriendlyName":["test"],"Topic":"test","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
18:13:56 MQT: tasmota/stat/test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.1.120:8000/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:30","StartupUTC":"2020-03-09T17:12:26","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2020-02-22T18:20:41","SaveCount":80,"SaveAddress":"FA000"}}
18:13:56 MQT: tasmota/stat/test/STATUS2 = {"StatusFWR":{"Version":"8.2.0(c65cc9f1)","BuildDateTime":"2020-03-09T18:11:04","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"394/699"}}
18:13:56 MQT: tasmota/stat/test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.1.120","LogPort":514,"SSId":["MyWifi","MyWifi2"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008229","2805C8000100060000005A00000000000000","00000001","00000000"]}}
18:13:56 MQT: tasmota/stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
18:13:56 MQT: tasmota/stat/test/STATUS5 = {"StatusNET":{"Hostname":"test","IPAddress":"192.168.1.169","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.105","Mac":"2C:3A:E8:14:CD:0B","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
18:13:56 MQT: tasmota/stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.flespi.io","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_14CD0B","MqttUser":"51841","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:13:56 MQT: tasmota/stat/test/STATUS7 = {"StatusTIM":{"UTC":"2020-03-09T17:13:56","Local":"2020-03-09T18:13:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:34","Sunset":"19:14"}}
18:13:56 MQT: tasmota/stat/test/STATUS10 = {"StatusSNS":{"Time":"2020-03-09T18:13:56"}}
18:13:56 MQT: tasmota/stat/test/STATUS11 = {"StatusSTS":{"Time":"2020-03-09T18:13:56","Uptime":"0T00:01:30","UptimeSec":90,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MyWifi","BSSId":"8C:B6:9E:A3:BA:CC","Channel":4,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:

### TO REPRODUCE
Compile, I guess, any version later than commit df3441 (february, 20th); although it is not deterministic

### EXPECTED BEHAVIOUR
DS18B20 should be detected and temperature should be shown

Jason2866, 2020-03-09:
Can you link to the commit you think it is the reason for the issue.
There is no commit df3441 on 20.02.2020.

peret2000, 2020-03-10:
Sure. This is the commit: [Fix DS18x20 driver](https://github.com/arendst/Tasmota/commit/df3441786186cc2155ba76b0e1d2a2bb035cc93e#diff-1ee864f3d2bd5e53b506f4af0a42e49d).

I was done on the 18th. Sorry.

It seems it was done to correct an already existing issue. Maybe it behaves right or wrong depending on a very specific property of the sensor...

Jason2866, 2020-03-10:
@arendst  Sensor fixing fun ;-)

arendst, 2020-03-10:
Try again pls.

peret2000, 2020-03-10:
Unfortunately, it is not working

arendst, 2020-03-10:
Noted the same. Will research (again)

peret2000, 2020-03-10:
With only these changes (undoing that [commit of the 18th](https://github.com/arendst/Tasmota/commit/df3441786186cc2155ba76b0e1d2a2bb035cc93e#diff-1ee864f3d2bd5e53b506f4af0a42e49d)) it works:

[image]

arendst, 2020-03-10:
I guess. But the problem in that case it doesn't work on my hardware.

I rolled back to a 20191107 version and that runs fine. Since then Shelly 2 wire code was added. Still investigating where it goes wrong.

arendst, 2020-03-10:
Pls try the latest one...

peret2000, 2020-03-10:
Not working :(

arendst, 2020-03-10:
I'm out of ideas.

peret2000, 2020-03-10:
I understand. There must be a complex problem, related to how the compiler translates source code into machine instructions, because depending on where the code is placed, there is a different result. So, it hides some problem (race condition?) really complex to detect, I'm afraid.
I can only say that adding log prints the code worked fine.

k0bar, 2020-03-10:
Perhaps I can add to the confusion.  I have two Witty Cloud modules with DS18B20 sensors attached.  That is the only sensor added to them.  Under version 8.1.0.7 both sensors were detected and worked fine.  However, after building and uploading 8.1.0.10 the DS18B20 is detected and works fine on one module but is not detected at all on the other module.  I think this reduces the chance of it being related to the compiler.  My gut tells me a timing issue....but that's my gut.  I will try to swap sensors and see what happens.

trojanobelix, 2020-03-11:
Might an interrupt while reading be the problem?
That is the only differnce I can see to other implentations like

`// Read a bit. Port and bit is used to cut lookup time and provide more certain timing.
uint8_t OneWire::read_bit(void)
{
	IO_REG_TYPE mask IO_REG_MASK_ATTR = bitmask;
	volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
	uint8_t r;

	noInterrupts();
	DIRECT_MODE_OUTPUT(reg, mask);
	DIRECT_WRITE_LOW(reg, mask);
	delayMicroseconds(3);
	DIRECT_MODE_INPUT(reg, mask);	// let pin float, pull up will raise
	delayMicroseconds(10);
	r = DIRECT_READ(reg, mask);
	interrupts();
	delayMicroseconds(53);
	return r;
}`

jansene, 2020-03-11:
Nodemcu single DS18b20
No issue when using STAGE version of 8.1.0.10: http://thehackbox.org/tasmota/pre-2.7/tasmota.bin
Program Version	8.1.0.10(eaf0e35-tasmota)
Build Date & Time	2020-03-10T18:09:58
Core/SDK Version	**STAGE/2.2.2-dev(38a443e)**

Developement version 8.1.0.10, DS18b20 not detected: http://thehackbox.org/tasmota/tasmota.bin
Program Version	8.1.0.10(eaf0e35-tasmota)
Build Date & Time	2020-03-10T18:00:11
Core/SDK Version	2_6_1/2.2.2-dev(38a443e)

k0bar, 2020-03-11:
Interesting.  I'm going to recap my situation again.  Two stock Witty 
Cloud modules with a single DS18B20 attached to each.  Both worked fine 
under 8.1.0.7 but not under 8.1.0.10 (sorry, didn't use any versions in 
between).   Other salient points: Program Version    8.1.0.10(tasmota), 
Build Date & Time 2020-03-10T17:35:21 (my build using platformio),  and 
Core/SDK Version    STAGE/2.2.2-dev(38a443e).

I always seem to be in a prototype stage with my ideas so my DS18B20's 
are in sockets.  Last night I had some time to explore. Findings (which 
are only going to confuse people more)...older DS18B20's (about 5 years 
old) are NOT recognized.  Yes, they do work.  Newer DS18B20's and 
MAX31820's (about a year old) ARE recognized.

My guess would be something in the chip masks/process changed over 
time.  We'll never know, but seems to indicate to me a timing issue.  
Are the newer chips are faster?  I don't know, just pure speculation 
here.  But it is very repeatable.  Old not recognized, new recognized, 
both work in 8.1.0.7 and earlier versions.

What can I do to help besides flap my fingers and pass along my 
observations?

Jason2866, 2020-03-11:
Could be a timing issue. Sensors from different generations and  brands, core with little changes which can affect timing too.

ascillato, 2020-03-12:
@k0bar 

Please, can you test latest Tasmota but with latest arduino core http://thehackbox.org/tasmota/pre-2.7/tasmota.bin

Thanks

k0bar, 2020-03-12:
That worked fine for me with all of the D@18B20 sensors...old, newer, 
and MAX31820.  Reverted to my build and no joy on the older sensors.

Trivia...sensor IDs (less leading 0's) 400CC16, 400DA92 (old) ACC2E13, 
ACB2AF4, and A23E2CD

That build has far more enabled then I use in my build.

ascillato2, 2020-03-12:
Great to know that latest Tasmota with the new arduino core works fine for you.

So, the issue seems to be in the old core. The new core is going to be in the next release of Tasmota. 

The fact that the Tasmota default has more sensors that your own build is not going to affect anything in your device. You can use it. Anyway, you can build your own version also with latest core.

closed 2020-03-12 by ascillato2

peret2000, 2020-03-12:
Thanks a lot to everybody, with the help of several people giving different ideas and symptoms it has been solved. I will test the last commit in a few hours (not possible now). Thanks again

peret2000, 2020-03-12:
Confirmed that it works in my infrastructure with 8.2.0, changeset f9181811 (STAGE/2.2.2-dev(38a443e)

h4nc, 2020-03-23:
@arendst Unfortunately it doesn't work for me. I have a DS18B20 connected to ESP8266 USB Stick lick device. All works fine when I go back to 8.1.0 but the temperature value doesn't show up in 8.2.0.
Tried restarting and power cycling it.

martinren87, 2020-05-02:
Same problem here... in Version 8.2.0 is not working anymore. Same in configuration in 8.1.0 works ok!

kugelkopf123, 2020-05-03:
@h4nc @martinren87  as mentioned here: [https://github.com/arendst/Tasmota/issues/7879#issuecomment-597980940](comment)
Latest Version with the latest Core. So use the [Link](https://github.com/arendst/Tasmota/issues/7879#issuecomment-597947651) or complie your own Version from Dev Branch and try again =)!

ascillato, 2020-05-03:
> in Version 8.2.0 is not working anymore

Use Tasmota-sensors.bin instead.

In tasmota.bin there is no more support for this sensor. It was moved to Tasmota-sensors.bin

h4nc, 2020-05-03:
@ascillato thanks I will try that.

BTW @kugelkopf123 your first link doesn't work correctly.

kugelkopf123, 2020-05-03:
> @ascillato thanks I will try that.
> 
> BTW @kugelkopf123 your first link doesn't work correctly.

Ok! But it was just a comment 2-3 post above yours ;)! So, you just had to read back this thread a Bit to get a solution.

h4nc, 2020-05-03:
@kugelkopf124 you are right, thanks. I found the post you referenced (if you don’t click it but copy and paste your link), but I thought I let you know in case you want to edit it to point to the posting you wanted to mention (this way it would make more sense for other readers). Anyway thanks for your input :)

Trilis29, 2020-05-16:
Hey guys sorry for opening, seems I have run is same problems, was on 8.2.0.3 was working all ok, restarted wemos couple of times and all sensors GONE. Tried reflashing firmware, reconnecting sensors, sometimes 1 or 2 sensors show up of 4, on another WEMOS device no sensor shows up, can't figure out the problem

h4nc, 2020-05-26:
> Use Tasmota-sensors.bin instead.

@ascillato I updated to Tasmota_sensors.bin version 8.3.1.

I get the sensor value using mqtt. However I can't see the temperature value in UI any more. Did this change in the newer versions or is this a bug?

arendst, 2020-05-26:
There are no bugs.

You might want to try command ``WebSensor`` and report it's output.

h4nc, 2020-05-26:
In the meantime I was curious if Tasmota-DE.bin (v8.3.1) would bring back the sensor value (via mqtt) too and indeed it did. This means that I don't need to use Tasmota_sensors.bin. 

Also with Tasmota-DE.bin v8.3.1 I got the temperature value back in the UI, so everything's fine again.

So to conclude I only missed the sensor value (in both the UI and via mqtt) in 8.2.0. 

@arendst I read your reply after updating to Tasmota-DE.bin so I'm not able to try the command any more. However I double checked and restarted the device and it did not show the value in the UI. But I'm interested what are you referring to? How should the output look like?

BTW: Two of my sonoff basics which don't have the best RSSI value (around 40%) don't update reliable over the air. I experienced this the last few updates.
Updating with the OTA-URL button and uploading the (minimal) bin file several times from my Mac (using Safari) didn't work. 
However uploading the bin file was always successful using Chrome on a Win10 PC. So this isn't an issue for me any more, but thought it would be good to mention it.

h4nc, 2020-06-09:
> But I'm interested what are you referring to? How should the output look like?

@arendst am I right that the output of `WebSensor` should output something like this: `{"WebSensor":"1,2,3,4,5,6"}`. 
I guess this list only shows the enabled control displays, right? 

And regarding to this [Link](https://docs.google.com/spreadsheets/d/e/2PACX-1vRBFqaP5QAFmT8iIYBxrn-EQp7-SrkxL6tb0OjpqtBxDEsmkiWCFei3sFyA_UXoqlbs3CqtEdOfC6Pa/pubhtml?gid=1029939700&single=true) the output has to contain a `5` and this would mean that the sensor should show up in the web UI display. 
Is that correct?

arendst, 2020-06-09:
Correct.

```
15:02:33 CMD: websensor
15:02:33 SRC: WebConsole from 192.168.2.1
15:02:33 CMD: Group 0, Index 1, Command "WEBSENSOR", Data ""
15:02:33 MQT: stat/shelly1pm/RESULT = {"WebSensor":"2,3,5,6,8,9,10,15"}
15:03:06 CMD: websensor5 0
15:03:06 SRC: WebConsole from 192.168.2.1
15:03:06 CMD: Group 0, Index 5, Command "WEBSENSOR", Data "0"
15:03:06 MQT: stat/shelly1pm/RESULT = {"WebSensor":"2,3,!5,6,8,9,10,15"}
15:03:07 CFG: Saved to flash at F9, Count 334, Bytes 4096
15:03:18 CMD: websensor5 1
15:03:18 SRC: WebConsole from 192.168.2.1
15:03:18 CMD: Group 0, Index 5, Command "WEBSENSOR", Data "1"
15:03:18 MQT: stat/shelly1pm/RESULT = {"WebSensor":"2,3,5,6,8,9,10,15"}
15:03:18 CFG: Saved to flash at F8, Count 335, Bytes 4096
```
When !5 is displayed it won't show on GUI

h4nc, 2020-06-09:
@arendst thank you. 

> There are no bugs.

Just to conclude: This issue appeared with Tasmota-sensors.bin (8.3.1) and it works fine with Tasmota.bin (8.3.1) for me. 

Out of interest I flashed Tasmota-sensors.bin again. This is the output of `WebSensor`:

`{"WebSensor":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}`

It includes a `5`(without `!` in front) while the sensor value doesn't show up in the GUI. So there might be something wrong.

@arendst this isn't an issue for me any more (and I'm already back to Tasmota.bin again) but I wanted to let you know in case this issue reappears for another user using Tasmota-sensors.bin. In this case I would recommend to try Tasmota.bin to see if the sensor shows up in the GUI again.