arendst/Tasmota #7882: Sonoff bridge rf receiver

Rushmed, 2020-03-09:
### PROBLEM DESCRIPTION
Im using a Tasmota flashed Sonoff RF Bridge. I successfully saved all RfKeys using my [remote](https://www.pollin.de/p/funksteckdosen-set-mit-3-steckdosen-550666).
I`m able to switch the plugs via RF bridge.
Now I want to use the codes received via bridge to detect if someone used a remote and to toggle this in ioBroker.

The issue I'am facing is that the codes received from RF modul in my PI do not correspond with the ones received by remote. A similar issue is described [here](https://groups.google.com/forum/#!searchin/sonoffusers/bridge$20receive%7Csort:date/sonoffusers/mkhZjecsm3c/SKxUH7XOCAAJ) but without  a solution.

Example:
```
20:35:12 MQT: tele/tasmota/RESULT = {"Time":"2020-03-08T20:35:12","RfReceived":{"Sync":11050,"Low":390,"High":1080,"Data":"154554","RfKey":2}}
20:35:21 MQT: tele/tasmota/RESULT = {"Time":"2020-03-08T20:35:21","RfReceived":{"Sync":11070,"Low":390,"High":1080,"Data":"154551","RfKey":1}}
20:36:24 MQT: tele/tasmota/RESULT = {"Time":"2020-03-08T20:36:24","RfReceived":{"Sync":9670,"Low":330,"High":950,"Data":"15555F","RfKey":"None"}}
20:36:26 MQT: tele/tasmota/RESULT = {"Time":"2020-03-08T20:36:26","RfReceived":{"Sync":9670,"Low":330,"High":950,"Data":"15555F","RfKey":"None"}}
```

Lines one an two show the Log when switching one plug on and off with the 433 MHz RF modul installed in my Pi.
Lines three and four show the Log when switching the same plug on and off with the Remote. Of course it's the the same remote I used to learn the codes to the bridge.
Both on off switching events are working but the codes of the remote are shown wrong.

Why does the bridge show wrong codes while reciving from remote although it learned the RfKeys right?
Why does the bridge not detect that a RfKey is pressed on the remote althoug it does when the PI sends codes?
Why is the code received via remote the same for on and off when switching different buttons?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff RF bridge
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: FTDI
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
19:06:35 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:06:35 MQT: stat/tasmota/RESULT = {"Module":{"25":"Sonoff Bridge"}}
19:06:35 MQT: stat/tasmota/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
19:08:15 MQT: stat/tasmota/STATUS = {"Status":{"Module":25,"FriendlyName":["SonoffBridge"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:08:15 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T22:19:40","StartupUTC":"2020-03-08T19:48:35","Sleep":50,"CfgHolder":4617,"BootCount":31,"SaveCount":89,"SaveAddress":"FB000"}}
19:08:15 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019.12.25 12:47:07","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"376/699"}}
19:08:15 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Slowdazamba","Rambazamba"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:08:15 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
19:08:15 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-5093","IPAddress":"192.168.178.58","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"2C:F4:32:BE:F3:E5","Webserver":2,"WifiConfig":4}}
19:08:15 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.20","MqttPort":1883,"MqttClientMask":"SonoffBridge","MqttClient":"SonoffBridge","MqttUser":"ioBroker","MqttCount":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:08:15 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 09 18:08:15 2020","Local":"Mon Mar 09 19:08:15 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:14","Sunset":"18:46"}}
19:08:15 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-09T19:08:15"}}
19:08:15 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-09T19:08:15","Uptime":"0T22:19:40","UptimeSec":80380,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"Slowdazamba","BSSId":"E0:28:6D:87:52:85","Channel":11,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
19:09:58 MQT: stat/tasmota/RESULT = {"WebLog":4}
19:09:59 CFG: in Flash gespeichert am FA, zählen 90, Bytes 4096
19:10:03 WIF: Prüfe Verbindung...
19:10:03 WIF: verbunden
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
I expect the bridge to show the correct rf codes when receiving.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-09:
Duplicate issue which is solved. Flash Portisch firmware. The RF protocol you use is not decoded
correctly via the original firmware of the RF mcu. The protocol is called Intertechno.
Portisch firmware can decode this protocol

Rushmed, 2020-03-09:
Ok thanks.
Flashed Portisch using this [tutorial](https://github.com/arendst/Tasmota/wiki/Sonoff-RF-Bridge-433#rf-chip-firmware-upgrade).
It still shows the wrong codes but semm to convert it into the needed one. Thats fine for me.

closed 2020-03-09 by Rushmed