arendst/Tasmota #7883: Tasmota console constantly executing unknown commands every 2 seconds

fermat1357, 2020-03-10:
Hi,
Not sure if I've done anything wrong or there is some bug in the firmware that's not compatible with my device. But the console log is showing constant (every 2 seconds) and invalid command is executed. Please see the partial console log below.

Step to produce:
Hardware: Tuya universal Smart IR Hub remote control
1. Flashed Tasmota via Tuya Convert using Raspberry Pi
2. Configured Wifi
3. Upgrade Firmware (I've tried a few versions of Tasmota, Tasmota-ir, etc.)

I've observed that when configure to Generic (18) module the problem starts to happen. In fact, I've tried a few other modules and noticed that if GPIO14 presents then the problem occurs.  It doesn't matter the GPIO14 is set to NONE or any other devices, the problem still happening. However, when I chose YTF IR Bridge Module module, the problem doesn't occur since this module doesn't have GPIO14. I'm not sure if it's related to GPIO14 or not but the IR needs GPIO14 to map to IRsend.

Partial console log:
06:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 CMD: :47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 CMD: MD:  CMD: }
06:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 CMD:  CMD: L: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 CMD: 47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:47 CMD: Unknown"}
06:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: "Unknown"}
06:56:48 CMD: 6:48 CMD: 6:56:47 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: mota/RESULT = {"Command":"Unknown"}
06:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: nknown"}
06:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: "Unknown"}
06:56:48 CMD: asmota/RESULT = {"Command":"Unknown"}
06:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: mota/RESULT = {"Command":"Unknown"}
06:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: nown"}
06:56:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
06:56:48 CMD: own"}

With module YTF IR Bridge Module the problem doesn't happen. Below is the normal console log.

00:00:00 CFG: Loaded from flash at F4, Count 101
00:00:00 SNS: Hardware Serial
00:00:00 Project tasmota Tasmota Version 8.1.0(ir)-2_6_1
00:00:00 WIF: Connecting to AP1 Fermat_2.4GHz in mode 11N as tasmota-7668...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on tasmota-7668 with IP address 192.168.1.222
07:00:13 RSL: tele/tasmota/INFO1 = {"Module":"YTF IR Bridge","Version":"8.1.0(ir)","FallbackTopic":"cmnd/DVES_F89DF4_fb/","GroupTopic":"cmnd/tasmotas/"}
07:00:13 RSL: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-7668","IPAddress":"192.168.1.222"}
07:00:13 RSL: tele/tasmota/INFO3 = {"RestartReason":"Software/System restart"}
07:00:17 RSL: tele/tasmota/STATE = {"Time":"2020-03-10T07:00:17","Uptime":"0T00:00:13","UptimeSec":13,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Fermat_2.4GHz","BSSId":"60:38:E0:E0:AC:D1","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}

closed 2020-03-10 by close-issue-app[bot]

ascillato, 2020-03-10:
Seems that you have seriallog enabled and RX and TX pins shortcircuited.

Try `seriallog 0` in the console.

Then, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

fermat1357, 2020-03-11:
Thank you so much. You are right. It was manufacture error. I soldered the 5v plus and negative bypassing the usb plug and the issue no longer happening. Thanks again for your help. I'll get on the support chat forum next time