arendst/Tasmota #7884: Get correct shutter position without PowerRetain

snakuzzo, 2020-03-10:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I have a cover on homeassistant with this `position_topic`:

```
position_topic: "stat/sonoff-cletto/SHUTTER1
```

Documentation says that to get the position there's need to activate `PowerRetain`. But...this way when I restart the hub I get the position from MQTT borker, not from the device.
I think this is wrong.

It could be better to get position directly from the device.
This is possible by `ShutterPosition` command, but the published topic is different than `position_topic`...

eg.

```
stat/sonoff-cletto/RESULT = {"Shutter1":{"Position":55,"Direction":0,"Target":55}}
stat/sonoff-cletto/RESULT = {"ShutterPosition1":55}
```

**Describe the solution you'd like**  
Topic to publish on `ShutterPosition` command should be the same topic published on position change...

eg.

```
stat/sonoff-cletto/SHUTTER1 = 64
```

**Describe alternatives you've considered**  
I tried to obtain this using RULES, but without success. I think it's not possible to trigger a RULE on a `stat` topic

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

to-scho, 2020-03-10:
Have you tried „ Another Home Assistant integration example with position update while movement (Tasmota versions >= v8.1.0.5):“ from the shutter docs? 
This would integrate via „ stat/sonoff-cletto/RESUT“ in your case to home assistant which makes updates available during movement and also allows to „ask“ for shutter position like you said.

snakuzzo, 2020-03-10:
That integration is useful just to get position while movement and doesn't solve my problem
I need to get the position "with no movement" on HA restarts  and I need to get the position from the device and not from the MQTT broker.

Should be great the solution suggested in my first post.

For now I'm solving with this method (but I think it's not correct to add another automation to have something stored on device)

Modified announce automation adding `shutterposition` for blinds and shutters ...

```
- alias: Sonoff Announce
  trigger:
  - event: start
    platform: homeassistant
  condition: []
  action:
  - data:
      topic: cmnd/tasmotas/POWER
    service: mqtt.publish
  - data:
      topic: cmnd/tasmotas/SHUTTERPOSITION
    service: mqtt.publish
```

and created another automation to publish a topic like `position_topic`...

```
- alias: 'Posizione tapparella camera da letto'
  trigger:
  - platform: mqtt
    topic: 'stat/sonoff-cletto/RESULT'
  condition:
    condition: template
    value_template: "{{ 'ShutterPosition1' in trigger.payload_json }}"
  action:
  - service: mqtt.publish
    data_template:
      topic: "stat/sonoff-cletto/SHUTTER1"
      payload: "{{ trigger.payload_json.ShutterPosition1 }}"
```

but...I repeat...It works fine but I think this solution is not a great solution

to-scho, 2020-03-11:
I had same issue but changing the integration from „ stat/sonoff-cletto/SHUTTER1“ to „stat/sonoff-cletto/RESULT“ will not need any additional automation to translate the mqtt.

With this integration I ask my shutter at HA Startup for Shutterposition and got result from tasmota and not mqtt broker. The update during movement is a bonus.

snakuzzo, 2020-03-11:
You are right!
Thank you ;)

closed 2020-03-11 by snakuzzo