arendst/Tasmota #7886: Harware restart when using web config

jpadie, 2020-03-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
hardware restart when changing some settings via the web admin screens.  e.g. belkin emulation on/off.
occasional near-complete failure of the web admin such that either nothing appears, or sometimes just a toggle button
generally a very laggy web interface.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): KooGeek Wifi Smart Plug KSP2
- [ ] Tasmota binary firmware version number used: 8.1.0 and 8.1.0.10. both the normal versions and the lite versions
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: web admin
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
rab/bedroomRadiator/stat/RESULT = {"NAME":"W-DEXI","GPIO":[0,90,0,0,134,132,0,0,130,52,21,0,0],"FLAG":0,"BASE":18}
13:40:19 MQT: rab/bedroomRadiator/stat/RESULT = {"Module":{"0":"W-DEXI"}}
13:40:20 MQT: rab/bedroomRadiator/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"90":"Button1n"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"130":"HLWBL SEL"},"GPIO13":{"52":"Led1"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
13:40:20 MQT: rab/bedroomRadiator/tele/STATE = {"Time":"2020-03-10T13:40:20","Uptime":"0T00:03:52","UptimeSec":232,"Vcc":3.442,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gaia-Rabastens-Free","BSSId":"50:0F:F5:D4:77:89","Channel":6,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:28"}}
13:40:20 MQT: rab/bedroomRadiator/tele/SENSOR = {"Time":"2020-03-10T13:40:20","ENERGY":{"TotalStartTime":"2020-01-13T09:10:27","Total":254.838,"Yesterday":7.886,"Today":0.846,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
rab/bedroomRadiator/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":434,"Rules":"on power1#state=1 do ruletimer1 5400 endon on rules#timer=1 do power1 0 endon"}
13:41:24 MQT: rab/bedroomRadiator/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:41:25 MQT: rab/bedroomRadiator/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

rab/bedroomRadiator/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["bedroom radiator"],"Topic":"bedroomRadiator","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-lite.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:05:25","StartupUTC":"2020-03-10T12:36:28","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2020-03-05T19:35:54","SaveCount":639,"SaveAddress":"F7000"}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(c52680f-lite)","BuildDateTime":"2020-03-05T20:01:02","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"422/699"}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gaia-Rabastens-Free","jpadie"],"TelePeriod":15,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00800000","00000000"]}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS4 = {"StatusMEM":{"ProgramSize":464,"Free":536,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":0,"Features":["00000809","8F082383","041083A0","00000006","010001C0","00000000","00000004"],"Drivers":"1,2,3,4,9,10,16,20,21,37","Sensors":"3"}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS5 = {"StatusNET":{"Hostname":"bedroomRadiator-2676","IPAddress":"192.168.0.134","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"CC:50:E3:7A:6A:74","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS6 = {"StatusMQT":{"MqttHost":"****","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A6A74","MqttUser":"**","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-03-10T12:41:53","Local":"2020-03-10T13:41:53","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:12","Sunset":"18:48"}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-10T13:41:53","ENERGY":{"TotalStartTime":"2020-01-13T09:10:27","Total":254.838,"Yesterday":7.886,"Today":0.846,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
13:41:53 MQT: rab/bedroomRadiator/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-10T13:41:53","Uptime":"0T00:05:25","UptimeSec":325,"Vcc":3.439,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gaia-Rabastens-Free","BSSId":"50:0F:F5:D4:77:89","Channel":6,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:28"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
no output as crash occurs.  after device reboots:

0:00 CFG: Loaded from flash at FB, Count 643
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 422/699
00:00:00 SRC: Restart
00:00:00 Project tasmota bedroom radiator Version 8.1.0.10(c52680f-lite)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Gaia-Rabastens-Free in mode 11N as bedroomRadiator-2676...
00:00:01 WIF: Checking connection...
[... 70 lines cut ...]
14:05:51 MQT: rab/bedroomRadiator/stat/RESULT = {"POWER":"OFF"}
14:05:51 MQT: rab/bedroomRadiator/stat/POWER = OFF
14:05:52 UPP: Multicast (re)joined

```

NB sometimes there is a hardware restart logged, rather than software. I have not been able to determine a pattern

### TO REPRODUCE
_Steps to reproduce the behavior:_
interface is generally laggy.  no particular steps required.
to create the crash, turning on and off Belkin emulation seems always to reproduce the issue.  

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

the interface performs as on other tasmotised devices and no crash occurs. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-10:
At first I noticed the same on wemos. Then I changed some code which is probably causing the watchdog. That solved it. Then I reverted the change to fine tune for background code. In the end I have the same unchanged code running as expected.

Weird as I also lost my DS18B20 again.

There is something fishy going on during compilation...

Investigating.

rrussev, 2020-03-10:
I can confirm the hardware restart with wemos and the latest dev didn’t resolve the problem:

Program Version	8.1.0.10(tasmota)
Build Date & Time	2020-03-10T19:50:43
Core/SDK Version	2_6_1/2.2.2-dev(38a443e)
Uptime	0T00:02:45
Flash write Count	707 at 0xF8000
Boot Count	50
Restart Reason	Hardware Watchdog

jpadie, 2020-03-11:
confirmed as per @rrussev.  No perceived change in behaviour after yesterday's commit.

ascillato2, 2020-03-12:
Please, try latest Tasmota version but with the new arduino core:

http://thehackbox.org/tasmota/pre-2.7/tasmota.bin

After flashing, please, type `reset 5` in the console in order to clean the flash (your settings will be erased) and then type `reset 3` in order to recalibrate the wifi. After that, please de-energize your device, wait some seconds and re-energize it again to allow the firmware to recalibrate the sdk wifi settings.

Then, please, reconfigure your device and try again. 

This wipe out should solve your issue. Please report back. Thanks.

Jason2866, 2020-03-12:
Activating and Deactivating WeMo and HueEmulation does a planned restart.
Please do as @ascillato2  suggested. The issue you see is often generated from fragments
left in flash and/or wifi set up which has gone to the south.
Most of the time we had success for users who asked for help doing "just" this.

The latest development builds available from thehackbox has optimizations in the Arduino Core.

Next release version of Tasmota will be based on this Arduino Core version

jpadie, 2020-03-12:
I have reflashed with the current release from pre-2.7  as per the above; and done a reset 5 (a few times).  

I am now finding the web interface unusable in its lagginess.  most page changes cause a restart event.

jpadie, 2020-03-12:
and I am still getting hardware watchdog resets on configuration changes.

Jason2866, 2020-03-12:
Have you done `reset 3` too? Only this command starts wifi calibration. After the command finished you have to power off the device and on again.
Reboot you router too.  The scenario you describe we do have now and than from user in Discord.
For further help please join our Discord server.

Jason2866, 2020-03-12:
On how many devices do you have this issue? If it is just one it is probably a hardware defect!

arendst, 2020-03-12:
What browser do you use. Try another one.

jpadie, 2020-03-12:
no.  i have not done a reset 3.  

How do you reach (even a tentative) conclusion that wifi calibration plays a part in the lagginess of a web interface?  can you indicate where that is caused in the code and I will take a look at writing a fix.  

The router is not the issue; or at least until it is proven that the codebase is not at fault I am not about to go on a wild goose chase blaming other aspects of the network; it's easier just to abandon these devices. 

For the same reason I do not consider that the browser is an issue.  The issue occurs across multiple client devices (e.g. mobile phone and laptop) and the tasmotised device reports a hardware watchdog restart. 

Given that others have reported the identical issue, and Theo is able to reproduce it, I do not consider that a hardware defect is likely.  I have ten devices to test on.  I can configure a second to test on however again, occam's razor suggests strongly against this and in contrast indicates that this is most likely a client-side bug in the code whose symptoms manifest in a watchdog timeout. 

I am happy, as above, to write the fix if someone can springboard me into the right area to look at.  

I don't participate in Discord.  If that's a barrier to bug-reporting here then I do understand and will drop the issue.

arendst, 2020-03-12:
Oh @jpadie if everything was that simple we had no issues.

As the espressif SDK is closed source it's always trial and error for the (Arduino-Esp8266) core team to get stability. They already move the LwIp code out of the SDK but beleive me there is still a lot being done by the SDK which may in some situations lead to watchdogs. Forget the option of you changing some code to fix this. The offending Tasmota code here hasn't change since months and still some users are reporting only just now having issues with saving settings. What changed is core code.

To help the community I suggest you visit the core site and start solving issues there.

Jason2866, 2020-03-12:
Using `reset 3` is our experience with the behaviour you describe to solve.
We have further experience that some routers do make weird troubles
just with ESP devices.
Furthermore there is the possibility that the config of your device does not fit (wrong)
generating wdt and lagginess.
Since you do not like a more responsive way to talk stop here.

jpadie, 2020-03-12:
Ok. Feel free to close the issue if you consider it resolved to your satisfaction.  

I will abandon the devices and use an alternative to meet my needs.

jpadie, 2020-03-12:
Quote: To help the community I suggest you visit the core site and start solving issues there

Ok. Will do.

Jason2866, 2020-03-12:
Why do you think closing this will give us satisfaction?
You do not consider that any of the suggestions we made maybe can help.
We do had this issues already with devices from users and what we suggest did solve for some.
So not trying this is just left out a chance to find your issue.
If we could reproduce we would dive into. But since we do NOT have this error we can do not bug hunting

jpadie, 2020-03-12:
Why are you referencing satisfaction?  You asked me to stop so I was communicating that from my perspective you could consider the issue closed.   It's that simple. There's no axe to grind. 

As Theo has and can reproduce the issue you already have suitable test cases if you want to continue bug fixing.  But if it is only my devices that are compromised then there is evidently no issue which is causing sufficient problems in the community to require fixing.  

And of course Theo indicates that the fault is in the core and not the firmware.  So this is the wrong forum for me to have addressed in the first place; for which I apologise.

ascillato, 2020-03-12:
Please, at least, try reset 3 (this will not delete any configuration) and then de-energize and re-energize your device. This will recalibrate the internal sdk radio values of the wifi.

ascillato2, 2020-03-13:
@jpadie 

> I am now finding the web interface unusable in its lagginess. most page changes cause a restart event.

That is likely to be a flashing problem. Please, for the sake of testing and burning all the options that have solved these type of issues in the past, at least, try `reset 3` (this will not delete any of your configuration) and then **de-energize** and **re-energize** your device. This power cycle will trigger a recalibration of the internal sdk values of the wifi radio driver of that problematic device.

ascillato, 2020-03-13:
@jpadie 

Hi, any news on this?

jpadie, 2020-03-13:
Hello, 
I appear not to be in the same time zone as you!

After my last post I downgraded to an earlier version of Tasmota (I think using the same core).  the device is now fine; so that resolves the issue for me.  If you wish me to send you a device for testing then feel free to email me your address and I will have a few sent.

Flashing has always been done via the web interface. 

```
Program Version | 7.2.0(tasmota)
-- | --
2019-12-22T11:41:04
2_6_1/2.2.2-dev(38a443e)
```

ascillato, 2020-03-13:
Ok, good test. Please, are you able to update again to latest Tasmota with latest core and do the reset 3 test for wifi radio re-calibration?

ascillato2, 2020-03-13:
@jpadie   

Please, try latest Tasmota version but with the new arduino core:

http://thehackbox.org/tasmota/pre-2.7/tasmota.bin

After flashing, please, type `reset 3` in the Tasmota console in order to recalibrate the wifi. After that, please de-energize your device, wait some seconds and re-energize it again to allow the firmware to recalibrate the sdk wifi settings.

ascillato2, 2020-03-14:
Closing this issue as there is no feedback. Please, when you have time, perform the requested test. Thanks. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-14 by ascillato2

seritos, 2020-05-21:
@jpadie We are interested in the Koogeen KSP2. Could you confirm that uses ESP82XX? 
Where you also able to use the power meter? 
Is easy to flash? 

We can not find almost any info about it, and there's a nice promotion is Amazon ;) 

Thanx!

Jason2866, 2020-05-21:
@seritos  at the time this issue was openend KooGeek Wifi Smart Plug KSP2 uses a ESP82xx
and Tasmota is successfully flashed and working. BUT we dont know if KooGeek Wifi Smart Plug KSP2 still uses NOW the the ESPxx

jpadie, 2020-05-21:
HelloMy units have esp8286EX chips. I used a docker deployment to flash them.  I bought 8 of the units I think.  One got bricked in flashing. The others were ok although I've given four away to different people who have all stopped using them as the tasmota firmware caused so many lockups on the devices.  If I buy more I will stick with the original firmware or write some simple stuff of my own to use. It's frustrating having devices fail to respond to Alexa or go dark when you're not in the house.  Undermines confidence in the device as a whole.  Yes - I use the power metering function on one of them.  A resistive load so it's relatively accurate and consistent. I've not tried taking these to bits.  They seem plasti-welded so it would be a destructive investigation.

seritos, 2020-05-21:
Thanx jpadie!!