arendst/Tasmota #7887: Tasmota should *not* default to pool.ntp.org

scottchiefbaker, 2020-03-10:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
No

**Describe the solution you'd like**  
Tasmota defaults to **pool.ntp.org** for obtaining NTP time. Per the [documentation](https://www.ntppool.org/en/vendors.html) on pool.ntp.org:

> You must **absolutely not use the default pool.ntp.org zone names** as the default configuration in your application or appliance. 

pool.ntp.org will create a subdomain of something like **tasmota.pool.ntp.org** that will serve the same purpose, but also protect the pool from abuse. 

* [Flawed Routers Flood University of Wisconsin Internet Time Server](http://pages.cs.wisc.edu/~plonka/netgear-sntp/)
* [D-Link misconfiguration incident](https://people.freebsd.org/~phk/dlink/)

**Describe alternatives you've considered**  
We can keep using the default pool, but we risk angering the pool admins by not following their requests for getting a vendor domain.

**Additional context**  
N/A

scottchiefbaker, 2020-03-10:
I run several NTP servers, and have an account at pool.ntp.org. I'd be happy to request the vendor domain and get the process started if it would help. I don't want to step on toes, if that's not my place though.

arendst, 2020-03-10:
See #6995.

Applied for vendor domain but never ever received any response. I think pool.ntp.org is routed to local ntp servers in countries so the pressure on some servers isn't that high.

Also latest library supports dhcp ntp servers so users may receive time from their router too.

In short, not an issue anymore as long as I do not hear anything from them.

scottchiefbaker, 2020-03-10:
OK... let me make the request of the pool one more time, and see if I can get any traction from them. If I don't hear anything I'll close the issue.

ascillato2, 2020-03-12:
Hi, if I do a traceroute of pool.ntp.org I receive a different IP of ntp server each time I search. In the pool page explains that they assign a random server when your device uses this service.

They ask users/programmers to not abuse of this service. They ask that if you have several devices using NTP Sync that you make only one of those to sync the time in your network and the rest to sync to that.

Also, for time accuracy, they recommend to use a pool close to your area. Like es.pool.ntp.org for spain, etc. 

You can have a vendor id for ntp but in the end is going to use the same servers (any of the 4000 that are in the ntp pool). That vendor id is only useful for statistics.

Another test shows that if you do a traceroute of pool.ntp.org and 0.pool.ntp.org and 1.pool.ntp.org at the same time, you reach for the 3 addresses the same ip. So, if you want to lower the usage of the whole pool, you should add other addresses not related to that pool as alternative like time.windows.com (that ntp is not available all the time) time.google.com (that ntp don't use the exact same correcting factors than ntp so your device may see a little jump of ms if switching from the other 2 pools).

So, following the ntp pool recommendation, the best option is to configure your router (or other device) as your local time network (that syncs with the pool) and make the rest of your devices to sync with it.

Other option that is the more accurate is to have a gps module attached to Tasmota and make one of your Tasmotas to be the NTP server of your network (already supported). That is called stratum 1 time server and is the more precise and reliable time source.

closed 2020-03-12 by ascillato2