arendst/Tasmota #7890: Sensor MQTT not triggerd on PowerDelta

s0170071, 2020-03-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _Shelly 2.5____
- [x] Tasmota binary firmware version number used: __8.1.0.10___
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: __OTA from 8.1.0__
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:30:04 CMD: Backlog Template; Module; GPIO 255
12:30:04 MQT: stat/shelly25_123/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
12:30:04 MQT: stat/shelly25_123/RESULT = {"Module":{"0":"Shelly 2.5"}}
12:30:04 MQT: stat/shelly25_123/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}
12:30:06 MQT: tele/shelly25_123/STATE = {"Time":"2020-03-11T12:30:06","Uptime":"0T00:00:12","UptimeSec":12,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MNET","BSSId":"5C:49:79:2C:0C:61","Channel":1,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

hook up a 60W lamp, configure powerlow 1; powerhigh 10; powerdelta 500 (5W)
Switch lamp on and off with relay (MQTT SENSOR message is sent only once )
Switch lamp on and off by disconnecting it (leave relay on) MQTT message fired as expected.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
If I switch the relay on, the power consumption MQTT messages are triggerd correctly. 
If I disconnect the lamp (leave relay on) and re-connect it, a MQTT message is sent.

So far, so good, so expected.  But...

If I switch the relay off, there should be a message stating a consumption of 0W. 
If I switch the relay back on, there should be another message stating 60W.
 
(I know that the documentation states that it is sent only when the relay is on. Read below why this is an issue)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

If I route the MQTT power messages to an OpenHap channel/Item, these items are only updated if the MQTT message arrives. When I turn the lamp on, the power gets updated to 60W. When I turn it off, it stays at 60W.

A workaround within OpenHAB:

- use a rule for setting the power to 0 when the relay is switched off
- Use another rule to set that power consumption to the previous value if the relay is turned back on. Unless there is a MQTT meassage saying otherwise in between.

Possible solution: send SENSOR message each time the relay is switched. Maybe delay it by a few ms to get a proper power reading.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-11:
Powerdelta 500 = 400W !

s0170071, 2020-03-11:
> Powerdelta 500 = 400W !

right. Set it to 105 now. 
Anyway, it does not change the issue that the SENSOR message is not triggered when the relay state changes.

Jason2866, 2020-03-11:
As designed and described:

> Set maximum delta for energy monitoring devices to report on active power load change while the power is ON. PowerDelta will not report when the power turns off.  »5.13.0
> 0 = disable reporting on power change
> 1..100 = set reporting on percentage power change to send an MQTT telemetry message
> 101..32000 = set reporting on absolute power change to send an MQTT telemetry message (offset by 100, e.g., 101=1W, 207=107W)

s0170071, 2020-03-11:
I know that it works like designed. No complaints to in regard ;-)
My point is:  Another **SENSOR message triggered shortly after relay action** is needed for reasonable OpenHAB integration. Otherwise you get false/outdated readings.

Otherwise the effect described in the first post happens.

Jason2866, 2020-03-11:
There is no message when toggling relais. This will not be changed.
For reliable working of powerdelta relais has to be always on.
It makes no sense at all to measure or send a mqtt message when a relais is off.

s0170071, 2020-03-11:
> There is no message when toggling relais. This will not be changed.

that's unfortunate.

> It makes no sense at all to measure or send a mqtt message when a relais is off.

but it makes sense to update when the reading changes. And it **does** change when the relay toggles.

> You can make a rule in OH after toggling relais request the power values

That means dozens of additional rules -  just as a workaround. 

> I have a different aproach in OH. I dont use powerdelta. All logic i need is done in OH.
> Way more flexible...

Would you mind to share ?

Jason2866, 2020-03-11:
A small low resources device like a Esp82xx can handle only a limited amount of features.
It is not the question what would be nice too have, it is the question what can be done
with very very limited resources.
For examples how to implement Tasmota stuff in OH you can visit https://community.openhab.org/

s0170071, 2020-03-11:
> what can be done
with very very limited resources.

never mind, I do it myself then. Thought you were  interested that there is a flaw in the design :-)

closed 2020-03-11 by s0170071