arendst/Tasmota #7895: Time of first blink is always shorter than the others

Th0maz, 2020-03-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using POWER 3 to blink a relay or LED, the first blinktime is always shorter than the others.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S20, Sonoff Basic, AILight
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Blinktime 5
Blinkcount 5
Power1 3

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Relay or LED blinks five times with a steady interval.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-12:
Hi,

Sorry, using your commands it works fine for me. I couldn't reproduce your issue. I have tried starting from relay OFF and also from relay ON and the time is steady.

Please, can you complete the template in order to have more debug information? Also add the output of the console when you run these commands. Remember to set weblog to 4. Thanks.

ascillato, 2020-03-13:
@Th0maz 

Hi, any news on this?

ascillato2, 2020-03-13:
Closing as there is no feedback.

When you have time, please add the logs. Thanks.

closed 2020-03-13 by ascillato2