arendst/Tasmota #7903: tx23 wind sensor isn't working in 8.1.10

NBB07, 2020-03-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

i've enabled the support for this sensor in my_user_config and build the image for my wemos d1 mini pro.

there are only 2 sensors connected. 1x BME280 (D1 & D2) and tx23 (D5).
No sensor were recognized :-(
if the tx23 is not activated in my_user_config, the BME280 is working fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): wemos d1mini pro
- [ ] Tasmota binary firmware version number used: 8.1.10
  - [x] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: arduino ide 1.8.12
- [x] Flashing tools used: esptool / arduino ide
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

2 sensors connected. 1x BME280 (D1 & D2) and tx23 (D5)
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-13:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

NBB07, 2020-03-13:
sorry, now i've completed the template.

curzon01, 2020-03-13:
Since you are already using a self-compiled version activate sensor debugging

`#define DEBUG_TASMOTA_SENSOR`

enable logging by command `Weblog 4` and observe the logs.

additionally use a pull-up on the GPIO14 (D5) of 10k, the sensor has none and needs it. The internal pull-up was activated on the TX23 port, but this is often not enough because it is very high-impedance

btw: I use exact the same config as yours on same ports with a Nodemcu and a Wemos d1, both are working perfect.

ascillato2, 2020-03-14:
Hi, any news on this?

NBB07, 2020-03-14:
Thanks a lot! The tx23 is working, now. I am happy!
Only the BME280 would not be recognized anymore.

15:30:10 CMD: Backlog Template; Module; GPIO 255
15:30:10 SRC: WebConsole from 192.168.1.205
15:30:10 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
15:30:10 TX23: sa=0x1b sb=5 (0x05), sc=33 (0x021), sd=0x08, se=5, sf=33
15:30:10 TX23: dir stat - counter=296, actint=5, avgint=6, avg=136.8 (cosx=-0.5, siny=0.5), min -1, max 14
15:30:10 SRC: Backlog
15:30:10 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
15:30:10 MQT: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:30:10 SRC: Backlog
15:30:10 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
15:30:11 MQT: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
15:30:11 SRC: Backlog
15:30:11 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
15:30:11 MQT: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"104":"TX2x"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
15:30:11 TX23: sa=0x1b sb=5 (0x05), sc=33 (0x021), sd=0x08, se=5, sf=33

and status 0:

15:31:46 CMD: status 0
15:31:46 SRC: WebConsole from 192.168.1.205
15:31:46 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
15:31:46 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["WetterNB7"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:31:46 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:44","StartupUTC":"2020-03-14T14:20:02","Sleep":50,"CfgHolder":4617,"BootCount":45,"BCResetTime":"2020-03-12T23:37:03","SaveCount":73,"SaveAddress":"FF8000"}}
15:31:46 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(tasmota)","BuildDateTime":"2020.03.13 19:28:25","Boot":31,"Core":"2_6_3","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"359/699"}}
15:31:46 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Nettelbreak",""],"TelePeriod":300,"Resolution":"558180C4","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:31:46 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":629,"Free":1416,"Heap":21,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840EF","FlashMode":2,"Features":["00000407","8FDAE397","003683A0","000000CD","010033C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,35"}}
15:31:46 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"WetterNB7","IPAddress":"192.168.1.134","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:D7:2E:C9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:31:46 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D72EC9","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:31:46 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"2020-03-14T14:31:46","Local":"2020-03-14T15:31:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:04","Sunset":"18:54"}}
15:31:46 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-03-14T15:31:46","TX23":{"Speed":{"Act":8.6,"Avg":8.2,"Min":0.0,"Max":15.8},"Dir":{"Card":"SSO","Deg":157.5,"Avg":145.5,"AvgCard":"SO","Min":112.5,"Max":292.5,"Range":180.0}},"SpeedUnit":"km/h"}}
15:31:46 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-03-14T15:31:46","Uptime":"0T00:11:44","UptimeSec":704,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Nettelbreak","BSSId":"30:B5:C2:2D:A2:82","Channel":9,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:06"}}}

But that's an other story.

@curzon01: it is possible to get your binary? BME280 is recognized fine with pre-compiled bins. but there is no tx23 in any pre-compiled bin.

thanks for your time and your support!

curzon01, 2020-03-14:
which wire do you use to your BME280 (type, len)? Any pullups on SDA/SCL?

experince my last change for greenhouse sensors I'd used an >1 m old network cable and this worked stable only using additonal pull-ups on SDA/SCL on Wemos side (used finally 2k7).

what'S cmnd `I2CScan` showing (try also pull-up on SDA/SCL)?

NBB07, 2020-03-14:
Len of wire: 5cm. i've got only 2 sensors. BME280 and tx23.
here the output of I2CScan:

16:18:09 CMD: I2CScan
16:18:09 SRC: WebConsole from 192.168.1.205
16:18:09 CMD: Gruppe 0, Index 1, Befehl "I2CSCAN", Daten ""
16:18:09 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x76"}

The sensor is found at 0x76 but there are no values..

curzon01, 2020-03-14:
Your BME sensor is not enabled for displaying on Web page, your Status 4 shows `"Sensors":"1,2,3,4,5,6,35"`

have a look on your tele topic (every 300 sec). This may contain "BME280" and then your sensor is working and only disabled for displaying on Web page - see cmnd `WebSensor<x>`

NBB07, 2020-03-14:
That's the trick. all stuff is running now -> you are my hero :-)
Thanks!

closed 2020-03-14 by NBB07