arendst/Tasmota #7906: Weird recconnect to WIFI and MQTT behavior

dbvcode, 2020-03-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [X] Tasmota binary firmware version number used: 8.1.0.10(lite)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode with PlatformIO
- [x] Flashing tools used: PlatformIO + Wemos D1 mini over USB
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:55:52 MQT: stat/tasmota_F87AC3/RESULT = {"NAME":"MyTemplate","GPIO":[255,255,255,255,7,161,255,255,138,139,255,255,255],"FLAG":15,"BASE":18}
11:55:52 MQT: stat/tasmota_F87AC3/RESULT = {"Module":{"0":"MyTemplate"}}
11:55:52 MQT: stat/tasmota_F87AC3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"7":"WS2812"},"GPIO5":{"161":"Buzzeri"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"138":"PN532 Tx"},"GPIO13":{"139":"PN532 Rx"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
11:57:09 MQT: stat/tasmota_F87AC3/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":443,"Rules":"ON EVENT#PN532_UID DO backlog color 0728fc; delay 1; power off ENDON"}
11:57:09 MQT: stat/tasmota_F87AC3/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
11:57:09 MQT: stat/tasmota_F87AC3/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
11:57:48 MQT: stat/tasmota_F87AC3/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_F87AC3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:05:23","StartupUTC":"2020-03-13T10:52:25","Sleep":0,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-03-13T07:58:05","SaveCount":703,"SaveAddress":"F5000"}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(lite)","BuildDateTime":"2020-03-13T10:25:54","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"320/699"}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Aaaa",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008008","28050A000000060000005A00000000000000","00000000","00000000"]}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS4 = {"StatusMEM":{"ProgramSize":439,"Free":564,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8D08A183","040003A1","00000006","011000C0","00000001","00000004"],"Drivers":"1,2,3,4,9,10,24,37","Sensors":"3,40"}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS5 = {"StatusNET":{"Hostname":"tasmota_F87AC3-6851","IPAddress":"192.168.2.138","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.2","Mac":"5C:CF:7F:F8:7A:C3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.25","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F87AC3","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS7 = {"StatusTIM":{"UTC":"2020-03-13T10:57:48","Local":"2020-03-13T11:57:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00"}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS10 = {"StatusSNS":{"Time":"2020-03-13T11:57:48"}}
11:57:48 MQT: stat/tasmota_F87AC3/STATUS11 = {"StatusSTS":{"Time":"2020-03-13T11:57:48","Uptime":"0T00:05:23","UptimeSec":323,"Vcc":2.981,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER":"OFF","Dimmer":99,"Color":"0728FC","HSBColor":"232,97,99","Channel":[3,16,99],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Aaaa","BSSId":"CE:2D:E0:BD:AB:27","Channel":7,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
10:00:14 WIF: Connecting to AP1 Aaaa in mode 11N as tasmota_F87AC3-6851...

Exception (29):
epc1=0x4023b21b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000019 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffecb0 end: 3fffffb0 offset: 0190
3fffee40:  0000004c 40237563 00000018 00000000  
3fffee50:  3fffee70 4023b357 3ffed5fc 1234cfa9  
3fffee60:  40255f0e 00000000 00000019 00000003  
3fffee70:  00000018 3fff4464 00000004 3ffedd40  
3fffee80:  3ffe8ed6 3ffe8ed6 3fff3f57 4023d31f  
3fffee90:  3ffedcf4 4023cef0 1234cfa9 60000600  
3fffeea0:  4023ca6c 40254908 3ffed6a4 00000000  
3fffeeb0:  00000001 00000004 3ffebaa8 3ffebb02  
3fffeec0:  4023cf2b 3ffedde8 3ffedde8 60000600  
3fffeed0:  4023eea0 3ffedde8 4bc6a7f0 00000000  
3fffeee0:  3ffed6a4 4023ee64 3ffedd0c 3ffee530  
3fffeef0:  4023cee0 4023ceb4 3ffed6a4 3ffee530  
3fffef00:  4023ce7f 3ffedd0c 3ffee530 4022fddc  
3fffef10:  4023d3ed 05b05bbb 3ffee530 60000600  
3fffef20:  4024d735 3ffedd0c 3ffee530 1234bc62  
[... 352 lines cut ...]

For 2. and 3. sometimes one reset(from the button) is not enough as it gets back into the same loop. The second reset seems to work. Also issues 2. and 3. happen out of the blue(i don't mess with the wifi), after the devices has been active for a while.

**The strange things:**
_Out of the 8 Wemoses I've flashed one did not connect to the Wifi from the get go._ Check out the behavior on `What I've noticed` point `2`. 

**Other stuff I've done:**
I thought this might be a power issue and the whole esp8266 gets confused and doesn't work properly. So I strapped a 10000uF electrolytic cap over the Wemos's 3.3V and ground terminals. Seems it works more stable now but this may only be in my head as the same software issues as described happen again. Seems after a bit longer than without the cap.

**My questions:**
1. After a WIFI failure, how can I make Tasmota reconnect instantly to MQTT after it has reconnected successfully to the WIFI? (And not wait the timeout for MQTT). I guess I want the same behavior Tasmota has when it boots up. First it looks for Wifi, then instantly for MQTT.

2. How can I restart the device after 5 consecutive WIFI reconnect failures?

3. How can I restart the device after 5 consecutive MQTT reconnect failures?

4. How can I prevent Tasmota to retain any type of state and assume it starts fresh on each reboot?

**PS.** I'd willingly go after these things in the code myself but unfortunately I don't have any experience with C, low level stuff, hardware programming, etc so I'm hopping my post yields some results, tips, pointers, maybe some additional rules to Tasmota or maybe even some bug fixes?

What I also guess is that most people use Tasmota for low power usage, sensors that send data every 10 minutes or something like this. This way this type of behavior may not even show up. But when I started stress testing it this it came up.

**As I was writing my request, the error I've pasted in the Console Log section popped up and the device restarted.**

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-13:
Do you have original Wemos mini or clones? Nearly ALL wemos mini (clones) dont have
a 3.3V vreg which can NOT deliver enough current. You will find infos if you do a quick search with
your favourite search engine.
Adding a cap does NOT solve this problem.
If the ESP82xx is not powered well with 3.3V it gets unreliable. Typical indicators are wifi connect
problems, reconnects, disconnects and WebFrontend lag until completly no response.

I do use the PN532 with a NodeMCU too and i have none of the issues you describe.

dbvcode, 2020-03-13:
Thanks for the quick reply. I can't tell if it's a clone or the real deal! It's from a 'reputable' source supposedly.
Indeed power was the first thing I was thinking of and I do know what you're saying.
The cap does smooth things out a lot, as I've also checked the 3.3 current on the oscilloscope. Not having enough power, indeed it cannot fix.
I'll try and get a 3.3v regulator and feed the board with that one.

In the meantime, my software wise questions remain:

1. After a WIFI failure, how can I make Tasmota reconnect instantly to MQTT after it has reconnected successfully to the WIFI? (And not wait the timeout for MQTT). I guess I want the same behavior Tasmota has when it boots up. First it looks for Wifi, then instantly for MQTT.
2. How can I restart the device after 5 consecutive WIFI reconnect failures?
3. How can I restart the device after 5 consecutive MQTT reconnect failures?
4. How can I prevent Tasmota to retain any type of state and assume it starts fresh on each reboot?

I am sure they make sense and in my case they would fix the issue and prevent the loops from crashing everything.

Jason2866, 2020-03-13:
There is just one reliable source. https://www.aliexpress.com/store/1331105?spm=a2g0o.productlist.0.0.1f9429e4y7pRJT
You dont run in the situation that any of the things you asked is needed. You can do by rules.
Recommended power supply is 1A when using additional stuff
https://www.reddit.com/r/esp8266/comments/9iizx4/warning_clone_wemos_d1_minis_with_only_150ma_33v/
There will be not a single reconnect with a reliable power supply of the ESP. 
I have devices running over weeks without any.

Jason2866, 2020-03-13:
Just my 2 cents. A board with a bad power supply NEVER runs never reliable. It just seems it works.

dbvcode, 2020-03-13:
@Jason2866 thanks for your suggestions and help.
Unfortunately I am in a tight spot where I don't have time to get new boards so I have to 'repair' what I have.  Even if that means getting an external regulator and hooking the device on the 3.3V pin instead of the 5V.
The next devices I will get will be from a reputable source and I'll watch out for this thing!

Please also point me on the right direction with the rule for counting the mqtt and wifi retries. I can't wrap my head around that one.
I do see I can get the disconnected event `ON wifi#disconnected` but how can I count the disconnects?

Jason2866, 2020-03-13:
As said before you dont need this. If the esp is working correct it has no wifi disconnects.
At the moment it is useless. With a not correct powered Esp a rule will not work as it should.
You cant fix this hardware issue which makes it unreliable with a code running on the device
which has gone to the south...  
Build your extern 3.3V power supply and all your problems will be gone.

dbvcode, 2020-03-13:
@Jason2866 I've just added a 1117 regulator board directly to the 3.3V and first impression is everything feels faster. Logs show up quicker it seems, the web interface is snappier.
Testing is on and will report the findings here!

ascillato2, 2020-03-13:
> Logs show up quicker it seems, the web interface is snappier.

Great to know that now you have solved your hardware issue and it is working for you.

closed 2020-03-13 by ascillato2