arendst/Tasmota #7909: Zigbee connection with CC2530

jwillmer, 2020-03-13:
### PROBLEM DESCRIPTION

I'm using a Wemos D1 Mini and a CC2530. I build my own firmware with the Zigbee flag enabled. I can select the Zigbee parameters in templates so I believe the firmware build was successful. At the moment I have no MQTT broker configured. My CC2530 uses GPIO13/15 as described in the tutorial.
I know like to try test the connections with my CC2530. Sending `ZigbeePermitJoin 1` in the console returns `Command: UNKNOWN`. I don't know what I'm doing wrong and need help to figure it out.

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [-] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: development
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Docker Compiler
- [x] Flashing tools used: Tasmota-PyFlasher-1.0
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

23:38:23 CMD: Backlog Template; Module; GPIO 255
23:38:23 RSL: RESULT = {"NAME":"Zigbee","GPIO":[0,0,0,0,0,0,0,0,0,166,0,165,0],"FLAG":0,"BASE":18}
23:38:23 RSL: RESULT = {"Module":{"0":"Zigbee"}}
23:38:23 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

-- No rules

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

23:39:20 CMD: Status 0
23:39:20 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_6E4669","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:39:20 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:17:05","StartupUTC":"2020-03-13T22:22:15","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-03-13T23:13:52","SaveCount":11,"SaveAddress":"F9000"}}
23:39:20 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0.11(tasmota)","BuildDateTime":"2020-03-13T22:10:00","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"345/699"}}
23:39:20 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["myAP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:39:20 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","047683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,23,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:39:20 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_6E4669-1641","IPAddress":"192.168.1.145","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"5A:AF:AF:AE:A6:A9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:39:20 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-03-13T22:39:20","Local":"2020-03-13T23:39:20","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:06","Sunset":"18:52"}}
23:39:20 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-03-13T23:39:20"}}
23:39:20 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-03-13T23:39:20","Uptime":"0T00:17:05","UptimeSec":1025,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"myAP","BSSId":"1A:A8:A9:AA:AF:A1","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
--

### EXPECTED BEHAVIOUR
--

### SCREENSHOTS
[image]

[image]

### ADDITIONAL CONTEXT
--

Jason2866, 2020-03-14:
The command syntax has changed. Take a look at https://tasmota.github.io/docs/#/Zigbee?id=using-zigbee2tasmota

ascillato2, 2020-03-14:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-14 by ascillato2

jwillmer, 2020-03-14:
Using `ZbPermitJoin 1` returns: 
`RSL: RESULT = {"ZbPermitJoin":"Zigbee not started (yet)"}`

How do I start the zigbee device?

This is my hardware setup - if that helps:
[image]

s-hadinger, 2020-03-14:
The error you're seeing means that Tasmota does not receive anything from the cc2530. 

Did you flash successfully the cc2530?
Have you checked that RX/TX are in the right order? TX to Rx and Rx to TX.

jwillmer, 2020-03-14:
I used [ Z-Stack-firmware/coordinator/Z-Stack_Home_1.2/bin/default/CC2530_DEFAULT_20190608.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2530_DEFAULT_20190608.zip)

I flashed it via CC Debugger and SmartRF. I used the following pin layout except that I took Vcc and GND from the main interface pins. 
[image]

The only odd thing was that it always failed the image verification process after flashing. But erasing and flashing worked without any errors so I ignored this part. I guess my next step is to flash the CC2530 DL-20 with working verification? :(

[image]

s-hadinger, 2020-03-14:
I suggest you try flashing with the Wemos D1 and follow the wiki. It's a proven method.

jwillmer, 2020-03-14:
Ok, I will give it a try.

Jason2866, 2020-03-15:
Flashing using CCLoader works well too.
https://github.com/Jason2866/CCLoader

jwillmer, 2020-03-16:
Turns out my CC2530 module is faulty. I can't erase the default image on it 👎