arendst/Tasmota #7914: Support for more than 8 Ds18b20 sensors

mariusebastian, 2020-03-14:
**Have you looked for this feature in other issues and in the docs?**
Yes, the docs does not mention a max number of sensors, but it is limited to 8 pcs 

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I need to monitor more than 8 temperature probes, tasmota only supports 8.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I want tasmota to support more than 8 sensors.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Alternatively I could settle for a description on how to edit the drivers to make this work. I have seen people trying to fix this by changing the `DS18X20_MAX_SENSORS` parameter in xsns_05_ds18x20.ino, but this apparently messes up the MQTT payload so more changes are required.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-14:
Sorry duplicate. There will no support for more than 8.

mariusebastian, 2020-03-14:
ok, that's a petty.

closed 2020-03-14 by mariusebastian