arendst/Tasmota #7916: German languange texts with default en-GB configuration

tiagofreire-pt, 2020-03-14:


arendst, 2020-03-14:
This is caused by xsns_53_sml.ino. Disable define USE_SML_M and the messages will be gone.

The code owner will need to revisit the functionality of this: 
```
#if MY_LANGUAGE==de-DE
// german web text
#define D_TPWRIN "Verbrauch"
#define D_TPWROUT "Einspeisung"
#define D_TPWRCURR "Aktueller Verbrauch"
#define D_TPWRCURR1 "Verbrauch P1"
#define D_TPWRCURR2 "Verbrauch P2"
#define D_TPWRCURR3 "Verbrauch P3"
#define D_Strom_L1 "Strom L1"
#define D_Strom_L2 "Strom L2"
#define D_Strom_L3 "Strom L3"
#define D_Spannung_L1 "Spannung L1"
#define D_Spannung_L2 "Spannung L2"
#define D_Spannung_L3 "Spannung L3"
#define D_METERNR "Zähler Nr"
#define D_METERSID "Service ID"
#define D_GasIN "Zählerstand"                // Gas-Verbrauch
#define D_H2oIN "Zählerstand"                // H2o-Verbrauch
#define D_StL1L2L3 "Ströme L1+L2+L3"
#define D_SpL1L2L3 "Spannung L1+L2+L3/3"

#else
// other languages (tbd)
#undef D_TPWRIN
#undef D_TPWROUT
#undef D_TPWRCURR
#undef D_TPWRCURR1
#undef D_TPWRCURR2
#undef D_TPWRCURR3
#undef D_Strom_L1
#undef D_Strom_L2
#undef D_Strom_L3
#undef D_Spannung_L1
#undef D_Spannung_L2
#undef D_Spannung_L3
#undef D_METERNR
#undef D_METERSID
#undef D_GasIN
#undef D_H2oIN
#undef D_StL1L2L3
#undef D_SpL1L2L3

#define D_TPWRIN "Total-In"
#define D_TPWROUT "Total-Out"
#define D_TPWRCURR "Current-In/Out"
#define D_TPWRCURR1 "Current-In p1"
#define D_TPWRCURR2 "Current-In p2"
#define D_TPWRCURR3 "Current-In p3"
#define D_Strom_L1 "Current L1"
#define D_Strom_L2 "Current L2"
#define D_Strom_L3 "Current L3"
#define D_Spannung_L1 "Voltage L1"
#define D_Spannung_L2 "Voltage L2"
#define D_Spannung_L3 "Voltage L3"
#define D_METERNR "Meter_number"
#define D_METERSID "Service ID"
#define D_GasIN "Counter"                // Gas-Verbrauch
#define D_H2oIN "Counter"                // H2o-Verbrauch
#define D_StL1L2L3 "Current L1+L2+L3"
#define D_SpL1L2L3 "Voltage L1+L2+L3/3"

#endif
```
As it stands it won't work as pio users will add the language switch as compiler option undetectable by the code.

Only solution is to add translations to all language files and/or re-use already translated words.

tiagofreire-pt, 2020-03-14:
I'm able to translate this to Portuguese (pt-PT) and Spanish (es-ES). How can I contribute to?

tiagofreire-pt, 2020-03-14:
> 
> 
> This is caused by xsns_53_sml.ino. Disable define USE_SML_M and the messages will be gone.
> 

The Smart Meter wiki state that I need to define that var. Is the wiki correct?

ascillato2, 2020-03-14:
Closing as the PR has been made.

PR #7917 (2020-03-15): Translations for xsns_53_sml.ino (en-GB, de-DE and pt-PT)
## Description:

**Related issue (if applicable):** fixes #7916

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core Tasmota_core_stage
  - [X] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
PR #7923 (2020-03-16): Correcting translations and energy unit for them dimensional denominations
## Description:

Correcting translations for its dimensional denominations and energy unit from KWh to kWh.

**Disclaimer: "Current" was corrected to "Active Power", as its current unit in Tasmota is "W", not "A". Could anyone confirm this in the code itself?**

**Related issue (if applicable):** fixes #7916 

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works on c [... cut]
closed 2020-03-14 by ascillato2