arendst/Tasmota #7922: Blitzwolf SHP-2 with bluetooth..

betonishard, 2020-03-15:
**Have you looked for this feature in other issues and in the docs?**  
YES

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
NO

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I have had a 30 minute google search looking for something that I think there is no trace of it. Or it is my google capabilities ofcourse. However the following.

I have multiple Blitzwolf SHP2, working flawlessly with Tasmota. However, since there there are unused GPIO pins on it, I am wondering if someone has already attached more sensors to it. FOr instance, I have some BLE devices and since the Blitzwolf is hanging throughout the house, it seemed obvious to attach a HM-10 bluetooth device to it to pick up BLE signals. They are mostly not in sight and on places that make temperature and humidity not that reliable and motion is ruled out as well. 

Probably someone has an idea about this?

P.S. I know the TX/RX are used for flashing..

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Putting up separate devices for BLE coverage

Well, it would be great to hear from you..

Thanks!!

Jason2866, 2020-03-15:
DO NOT add add any sensors to a Power Monitoring device. 
It is not supported and it will never be supported since
it is dangerous because you have live Mains on the GPIOs.
Please read our docu for further information.

ascillato2, 2020-03-15:
Closing this issue as it is **duplicated** and also because it has been warned in several places of the docs to **not connect anything extra to power measurement devices**. It is very dangerous. Search in issues for photos of the result (burning devices and electroshocks to people.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-15 by ascillato2