arendst/Tasmota #7926: Feit Dimmer (from costco) - physical switch doesn't work as expected

qwertangel, 2020-03-15:
### PROBLEM DESCRIPTION
When using feit dimmer with tasmota, switch works correctly when sending remote commands (over web/mqtt). However when physical switch is pressed, it requires 3 extra pushes to turn switch off.

### REQUESTED INFORMATION
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
Feit Smart Dimmer https://templates.blakadder.com/feit_electric-DIM-WIFI.html
- [x ] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: 
Tuya Convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
21:31:59 CMD: Backlog Template; Module; GPIO 255
21:31:59 SRC: WebConsole from 192.168.1.178
21:31:59 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
21:31:59 SRC: Backlog
21:31:59 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
21:31:59 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:31:59 SRC: Backlog
21:31:59 CMD: Group 0, Index 1, Command "MODULE", Data ""
21:31:59 RSL: stat/tasmota/RESULT = {"Module":{"54":"Tuya MCU"}}
21:31:59 SRC: Backlog
21:31:59 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:31:59 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
21:32:05 WIF: Checking connection...
21:32:05 WIF: Connected

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
No rules

- [ ] Provide the output of this command: `Status 0`:
```
21:33:25 CMD: Status 0
21:33:25 SRC: WebConsole from 192.168.1.178
21:33:25 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:33:25 RSL: stat/tasmota/STATUS = {"Status":{"Module":54,"FriendlyName":["DiningRoom"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:33:25 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:11:47","StartupUTC":"2020-03-15T20:21:38","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":31,"SaveAddress":"F5000"}}
21:33:25 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.0.0.3(tasmota)","BuildDateTime":"2019-11-06T23:07:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
21:33:25 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mhome",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:33:25 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":446,"Free":556,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F400082","04048000","00000005","000000C0","00000080"],"Drivers":"1,2,3,4,9,10,12,16","Sensors":"1,2,3"}}
21:33:25 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3062","IPAddress":"192.168.1.176","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":<REDACTED>","Webserver":2,"WifiConfig":2}}
21:33:25 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AB4BF6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:33:25 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 15 20:33:25 2020","Local":"Sun Mar 15 21:33:25 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:02","Sunset":"18:55"}}
21:33:25 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-15T21:33:25"}}
21:33:25 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-15T21:33:25","Uptime":"0T00:11:47","UptimeSec":707,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":43,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"<REDACTED>","BSSId":"<REDACTED>","Channel":3,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
21:25:41 CMD: weblog 4
21:25:41 RSL: stat/tasmota/RESULT = {"WebLog":4}
21:25:41 CFG: Saved to flash at F7, Count 29, Bytes 4096
21:25:44 WIF: Checking connection...
21:25:44 WIF: Connected
21:25:51 TYA: Send "55aa00000000ff"
21:25:51 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
21:25:51 TYA: Heartbeat
21:26:02 TYA: Send "55aa00000000ff"
21:26:02 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
21:26:02 TYA: Heartbeat
21:26:04 WIF: Checking connection...
21:26:04 WIF: Connected
21:26:05 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","DpId":1,"DpIdType":1,"DpIdData":"01"}}
21:26:05 TYA: fnId=11 is set for dpId=1
21:26:05 TYA: RX Relay-1 --> MCU State: On Current State:Off
21:26:05 SRC: Switch
21:26:05 TYA: Send dim value=431 (id=2)
21:26:05 TYA: Send "55aa0006000802020004000001afc5"
21:26:05 RSL: stat/tasmota/RESULT = {"POWER":"ON"}
21:26:05 RSL: stat/tasmota/POWER = ON
21:26:05 {"TuyaReceived":{"Data":"55AA0307000802020004000001AFC9","Cmnd":7,"CmndData":"02020004000001AF","DpId":2,"DpIdType":2,"DpIdData":"000001AF"}}
21:26:05 TYA: fnId=21 is set for dpId=2
21:26:05 TYA: RX Dim State=431
21:26:06 CFG: Saved to flash at F6, Count 30, Bytes 4096
21:26:13 TYA: Send "55aa00000000ff"
21:26:13 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
21:26:13 TYA: Heartbeat
21:26:14 {"TuyaReceived":{"Data":"55AA0307000802020004000001AEC8","Cmnd":7,"CmndData":"02020004000001AE","DpId":2,"DpIdType":2,"DpIdData":"000001AE"}}
21:26:14 TYA: fnId=21 is set for dpId=2
21:26:14 TYA: RX Dim State=430
21:26:18 {"TuyaReceived":{"Data":"55AA03070005010100010011","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
21:26:18 TYA: fnId=11 is set for dpId=1
21:26:18 TYA: RX Relay-1 --> MCU State: Off Current State:On
21:26:18 SRC: Switch
21:26:18 TYA: Send dim skipped value=0
21:26:18 RSL: stat/tasmota/RESULT = {"POWER":"OFF"}
21:26:18 RSL: stat/tasmota/POWER = OFF
21:26:18 CFG: Saved to flash at F5, Count 31, Bytes 4096
21:26:24 WIF: Checking connection...
21:26:24 WIF: Connected
21:26:24 TYA: Send "55aa00000000ff"
21:26:24 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
21:26:24 TYA: Heartbeat

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
After flashing.. 
Press Button -> Switch lights On, FW updates to On
Press Button -> Switch lights Off, FW stays On
Press Button -> Switch lights On, FW stays On
Press Button -> Switch lights Off, FW updates to Off

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Press Button -> Switch lights On, FW updates to On
Press Button -> Switch lights Off, FW updates to Off

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
Happens over both tasmota 7.x and 8.x . Appranetly this issue doesn't happen in ESPHome (as per this thread https://community.home-assistant.io/t/costco-feit-smart-dimmer-tuya-convert-tasmota/151390 ) so looks like a tasmota specific issue and not hardware issue.

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-03-17:
The logs don't show the behaviour you stated in **TO REPRODUCE**

Please make detailed logs with the sequence stated

ascillato2, 2020-04-02:
Hi, any news on this?

ascillato2, 2020-04-04:
Please, when you have time, provide the requested logs. Thanks.

qwertangel, 2020-04-05:
log I posted is what I got. There was nothing more.

blakadder, 2020-04-05:
if thats all the logs you have this is what i see:
```
21:26:05 SRC: Switch
21:26:05 TYA: Send dim value=431 (id=2)
21:26:05 TYA: Send "55aa0006000802020004000001afc5"
21:26:05 RSL: stat/tasmota/RESULT = {"POWER":"ON"}
21:26:05 RSL: stat/tasmota/POWER = ON
```
switch turns relay on

```
21:26:18 SRC: Switch
21:26:18 TYA: Send dim skipped value=0
21:26:18 RSL: stat/tasmota/RESULT = {"POWER":"OFF"}
21:26:18 RSL: stat/tasmota/POWER = OFF
```
switch turns relay off

i don't see the 3 extra pushes.

give the information  i would suspect its an MCU or switch to MCU issue but given the logs there are no telltale signs of anything working wrong

beren12, 2021-03-25:
This is still an issue. TheEebb at https://github.com/TheEebb/tasmota-tuyamcu-fix/issues/2 made a patch in a separate repo, and they said they would update it for current tasmota. Hopefully we can get it merged or something like it. He seems to understand the TuyaMCU very well.

beren12, 2021-03-25:
ESPHome made similar changes I believe: https://github.com/esphome/esphome/pull/1404