arendst/Tasmota #7927: BME280 TempOffset for humidity and pressure

DerOetzi, 2020-03-15:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
When connecting a BME280 I can adjust temperature by using TempOffset but this is only applied to temperature. But humidity and pressure of the sensor are depending on the temperature too.

**Describe the solution you'd like**  
The TempOffset should be taken into account inside the library when calculating humidity and pressure.

**Describe alternatives you've considered**  
I don't see any alternatives.

**Additional context**  
I have tried to implement an own project, which I didn't finish for missing time, that's why I want to switch back to Tasmota. But there I have written an alternative library with correct TempOffset calibration for BME280 some month ago. You can find the library here [BME280 Library](https://github.com/DerOetzi/SoilMoistureSensor/blob/master/lib/BME280/BME280.cpp) Maybe you can reuse this code.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-03-16:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-16 by ascillato2

DerOetzi, 2020-03-16:
> Closing this issue as it is **duplicated**. Sorry.
>
> Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.
>
> ### [...]

I didn't find the duplicate issue can you give me a hint please, so I can add my code suggestion there?

ascillato, 2020-03-16:
Please, re do the search using other keys. At least there are 4 old issues with this. As a summary, this request has been dismissed after some discussions because an offset is wrong for humidity due to it requires a non linear fit curve for more precise values. Also this fit curve needs to be customized for each sensor based on real measurements. For more information you can check the sensor datasheet.

If you want to code some alternative option for that, a PR is welcome. Thanks.

DerOetzi, 2020-03-16:
I have found this issues as well, but they aren't correct because they apply the TempOffset after calculation before outputing. The operating principle for this kind of sensors (for example BME280) is they use temperature to read humidity and pressure. So if you correct the temperature you have to do this before you read humidity and pressure.

Unfortunately I don't have time to get deeper inside the Tasmota Code to do it on myself. 
I had a quick look at the library and think the correct place to add the TempOffset inside this function [calc_temperature Line 887](https://github.com/arendst/Tasmota/blob/a258296055fed8b3bcc9d63e7d8a82d18c666294/lib/BME680_driver-bme680_v3.5.9/bme680.c#L887) because dev->calib.t_fine is the variable which is used inside calc_pressure and calc_humidity functions to read correct data.