arendst/Tasmota #7932: If statement expression doesn't work with brackets

os400, 2020-03-16:
### PROBLEM DESCRIPTION
_If statement expression doesn't work with brackets_

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NODEMCU
- [ ] Tasmota binary firmware version number used: 8.1.0.8
  - [ ] Pre-compiled
  - [  ] Self-compiled
    - [ ] IDE / Compiler used: __gitpod___
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:36:22 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
11:36:22 MQT: stat/esp69/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:36:22 SRC: Backlog
11:36:22 CMD: Group 0, Index 1, Command "MODULE", Data ""
11:36:22 MQT: stat/esp69/RESULT = {"Module":{"18":"Generic"}}
11:36:23 SRC: Backlog
11:36:23 CMD: Group 0, Index 1, Command "GPIO", Data "255"
11:36:23 MQT: stat/esp69/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"30":"Relay2i"},"GPIO3":{"0":"None"},"GPIO4":{"10":"Switch2"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"0":"None"},"GPIO14":{"1":"DHT11"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
 {"StatusFWR":{"Version":"8.1.0.8(tasmota)","BuildDateTime":"2020-02-19T12:11:37","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"317/699"}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
11:40:01 CMD: if ( (2+0) < 1 ) var1=1 else var1=0 endif
11:40:01 CMD: Group 0, Index 1, Command "IF", Data "( (2+0) < 1 ) var1=1 else var1=0 endif"
11:40:01 MQT: stat/esp69/RESULT = {"If":"Done"}
11:40:02 SRC: Backlog
11:40:02 CMD: Group 0, Index 1, Command "VAR", Data "=1"

(removing brackets works)
11:40:22 CMD: if ( 2+0 < 1 ) var1=1 else var1=0 endif
11:40:22 SRC: WebConsole from 192.168.222.161
11:40:22 CMD: Group 0, Index 1, Command "IF", Data "( 2+0 < 1 ) var1=1 else var1=0 endif"
11:40:22 MQT: stat/esp69/RESULT = {"If":"Done"}
11:40:22 SRC: Backlog
11:40:22 CMD: Group 0, Index 1, Command "VAR", Data "=0"

```

### TO REPRODUCE
_Use second level of brackets in IF expressions _

### EXPECTED BEHAVIOUR
_Brackets to be treated as per the syntax rule:

1+2*2=5
(1+2)*2=6

11:43:12 CMD: if ( 1+2*2 == 5) var1=1 else var1=0 endif
...
11:43:12 CMD: Group 0, Index 1, Command "VAR", Data "=1"
...
11:44:05 CMD: if ( (1+2)*2 == 5) var1=1 else var1=0 endif
...
11:44:05 CMD: Group 0, Index 1, Command "VAR", Data "=1"
_

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Problem arose when I tried (time > (sunrise+20)) which appeared to work until sunrise became greater... a lot of expressions with second level brackets may appear to work without proper testing. Removing the brackets works fine for me but is a gotcha for the unwary.

The discovery is documented in the forum:
https://groups.google.com/d/msg/sonoffusers/zzfNxzhqb_A/a_-l6mVOAgAJ
_

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-16:
If statement is meant to ve used inside a rule. On the command line is going to be ignored.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

os400, 2020-03-16:
The If statement is not ignored on the command line, it is run the same as a rule.

The problem was discovered in, and exists in, a rule.

os400, 2020-03-16:
You're going to leave it closed without reading it? OK, I shan't bother next time...

closed 2020-05-03 by stale[bot]