arendst/Tasmota #7938: Commands over MQTT fail to accept parameters

nasmork, 2020-03-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Trying to pass parametrized commands to tasmota over MQTT but it fail to accept parameters responding with stat/tasmota_D72851/RESULT = {"Command":"Unknown"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini Pro
- [x] Tasmota binary firmware version number used: 8.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [VS Code with Arduino] IDE / Compiler used: _____
- [x] Flashing tools used: VS Code with Arduino
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
01:38:57 MQT: stat/tasmota_D72851/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
None

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
01:37:45 MQT: stat/tasmota_D72851/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"tasmota_D72851","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
01:37:45 MQT: stat/tasmota_D72851/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://www.111.ca/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T01:37:47","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"1970-01-01T00:00:00","SaveCount":3,"SaveAddress":"FF8000"}}
01:37:45 MQT: stat/tasmota_D72851/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-16T19:12:51","Boot":31,"Core":"2_6_3","SDK":"3.0.0-dev(c0f7b44)","Hardware":"ESP8266EX","CR":"326/699"}}
01:37:45 MQT: stat/tasmota_D72851/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"MqttLog":3,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["2WIRE",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["800080E9","2805C8000100060000005A00000000000000","00000000","80000000"]}}
01:37:45 MQT: stat/tasmota_D72851/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":1444,"Heap":23,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840EF","FlashMode":3,"Features":["00000809","CFD8C197","002683A1","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,6,7,8,9,10,12,16,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
01:37:45 MQT: stat/tasmota_D72851/STATUS5 = {"StatusNET":{"Hostname":"tasmota_D72851-2129","IPAddress":"172.16.1.163","Gateway":"172.16.1.1","Subnetmask":"255.255.255.0","DNSServer":"4.2.2.1","Mac":"A4:CF:12:D7:28:51","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:37:45 MQT: stat/tasmota_D72851/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D72851","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:37:45 MQT: stat/tasmota_D72851/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T01:37:45","Local":"1970-01-01T01:37:45","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
01:37:45 MQT: stat/tasmota_D72851/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T01:37:45"}}
01:37:45 MQT: stat/tasmota_D72851/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T01:37:45","Uptime":"0T01:37:47","UptimeSec":5867,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"2WIRE281","BSSId":"F8:32:E4:54:ED:19","Channel":3,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
01:41:39 CMD: Group 0, Index 0, Command "STATUS ", Data ""
01:41:39 CMD: Payload -99
01:41:39 MQT: stat/tasmota_D72851/RESULT = {"Command":"Unknown"}

01:42:24 CMD: Group 0, Index 1, Command "OPTION55 ", Data ""
01:42:24 CMD: Payload -99
01:42:24 MQT: stat/tasmota_D72851/RESULT = {"Command":"Unknown"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
cmnd/tasmota_D72851/Status 0

cmnd/tasmota_D72851/Option55 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-03-17:
Well, you should send commands to your device, that are known. At least "Option55" is no known command. Maybe, you mean "Setoption55"? 
Regarding "Status", I'm not quite sure, if the trailing spaces are a problem. On all my devices, the command works like a charm :-)

blakadder, 2020-03-17:
```
Command "STATUS ", Data ""
```
please send to topics without spaces with a properly defined payload

closed 2020-03-17 by blakadder

nasmork, 2020-03-17:
Not sure how to properly define payload for MQTT command...

Here's the command:
08:14:56 MQT: Received Topic "cmnd/tasmota_D72851/SetOption55 1", Data Size 0, Data ""
08:14:56 SRC: MQTT

Here's the response:
08:14:56 MQT: stat/tasmota_D72851/RESULT = {"Command":"Unknown"}

Here's the trace:
08:14:56 CMD: Group 0, Index 1, Command "SETOPTION55 ", Data ""
08:14:56 CMD: Payload -99

Same command sent via Serial is parsed just fine:
CMD: SetOption55 1
08:21:06 SRC: Serial
08:21:06 CMD: Group 0, Index 55, Command "SETOPTION", Data "1"
08:21:06 CMD: Payload 1
08:21:06 MQT: stat/tasmota_D72851/RESULT = {"SetOption55":"ON"}

Here's the same example with Status:
08:26:29 MQT: Received Topic "cmnd/tasmota_D72851/Status 0", Data Size 0, Data ""
08:26:29 SRC: MQTT
08:26:29 CMD: Group 0, Index 0, Command "STATUS ", Data ""
08:26:29 CMD: Payload -99
08:26:29 MQT: stat/tasmota_D72851/RESULT = {"Command":"Unknown"}

It somewhat works without the space between command and payload: 
08:26:57 MQT: Received Topic "cmnd/tasmota_D72851/Status0", Data Size 0, Data ""
08:26:57 SRC: MQTT
08:26:57 CMD: Group 0, Index 0, Command "STATUS", Data ""
08:26:57 CMD: Payload -99
08:26:57 MQT: stat/tasmota_D72851/STATUS = {"Status":{"Module":1,"Friend...……….}

While it should look like this:
08:31:16 SRC: Serial
08:31:16 CMD: Group 0, Index 1, Command "STATUS", Data "0"
08:31:16 CMD: Payload 0

Unless I read documentation wrong, commands with parameters over MQTT are sent in the following format: "prefix/topic/command payload"....

meingraham, 2020-03-17:
What MQTT client are you using to publish the command? Typically clients have entry fields for the topic (e.g., cmnd/tasmota_D72851/SetOption55) and a separate entry for the data (e.g., 1).

nasmork, 2020-03-17:
> What MQTT client are you using to publish the command? Typically clients have entry fields for the topic (e.g., cmnd/tasmota_D72851/SetOption55) and a separate entry for the data (e.g., 1).

I use MQTT.fx. Could you recommend any alternatives?

meingraham, 2020-03-17:
MQTT.fx should work fine if you follow that pattern (topic vs. data).

I use MQTT Explorer. Of course, I also use the `Publish` command on any Tasmota device web UI Console.

nasmork, 2020-03-17:
> MQTT.fx should work fine if you follow that pattern (topic vs. data).
> I use MQTT Explorer. Of course, I also use the Publish command on any Tasmota device web UI Console.

Thank you