arendst/Tasmota #7940: ESP-01 configuration help with tasmota

sparviero79, 2020-03-17:
Hello to all. I have recently flashed ESP-01 with the tasmota firmware V8.1.0 which is inserted in a 4 relay circuit, as in the photo attached below, but I can't get it to work. I tried with various guides that I found around, like this: https://github.com/arendst/Tasmota/wiki/LC-Technology-WiFi-Relay
but I can't get the relays activated. I hope I haven't flashed my ESP-01 in vain and have to burr it.
Thanks so much for your cooperation.

Link 4Relay: [https://sito.libero.it/mypersonalpage/wp-content/uploads/sites/22239/2020/03/Immagine.jpg](url)

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

sparviero79, 2020-03-17:
[image]

Jason2866, 2020-03-17:
Have you tried https://templates.blakadder.com/LC-ESP01-4R-5V.html

closed 2020-03-17 by sparviero79