arendst/Tasmota #7942: Very slow Tasmota on Sonoff Bridge

kowalczykrafal, 2020-03-17:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
I'm using the Sonoff Bridge tasmota and a door opener like Kerui. When Kerui opens (door opens) - Kerui sends a lot of the same code. And I think the Tasmota is waiting for the last information.
I have logs from the RFSinfer:
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366
Received 7392366

In my opinion, the Tasmota waits until there is silence and then publishes the code in the console. That's causing big delays.
There is no problem when the Tasmota receives only one code, then the information is shown right away.
I wanted to use the Keriu sensor to turn on the light in the room as soon as the door was opened. But such long delays cause discomfort.
Can you change that?

**Describe the solution you'd like**  
In text

**Describe alternatives you've considered**  
In text.

**Additional context**  
I don't have.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-18:
Known behaviour with original firmware of the RF MCU.
Flash 3rd party firmware "Portisch" to get immediatly a response.
Flashing is described https://tasmota.github.io/docs/#/devices/Sonoff-RF-Bridge-433

ascillato2, 2020-03-18:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-18 by ascillato2

pickonedev, 2021-09-22:
I have the same problem... I have a lot of Kerui PIR sensors and DOOR sensors...
When the sensors are triggered in small amount of time, the tasmotized Sonoff RF Bridge, does not show the proper chronologic triggered sensors... 

If I trigger the sensors from 1 to 8 in a semi-fast way or even normal way, it will always mess up the order, in the RF Console... it will be almost random or some of them are even missing, because of the slow receive of the codes... Like (1 3 2 4 7 5 6 8), or (1 3 5 4 7)

Anybody solved this with the RF Bridge?