arendst/Tasmota #7943: Broken links in Tamsota online docs

eku, 2020-03-17:
### PROBLEM DESCRIPTION

Links to pictures in https://tasmota.github.io/docs/#/devices/EXS-2-Channel-Wifi-Dimmer-V4?id=example-setup-with-dht22 are broken.
Since I don't have the pictures I'm no able to repair the documentation.

### REQUESTED INFORMATION

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE

Use a browser to view 
https://tasmota.github.io/docs/#/devices/EXS-2-Channel-Wifi-Dimmer-V4?id=example-setup-with-dht22

### EXPECTED BEHAVIOUR
Pictures a visible.

### SCREENSHOTS
none

### ADDITIONAL CONTEXT
none

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-03-18:
Yes, refering site https://www.ex-store.de/ is down. Let us wait if the site comes back.
Maybe be a victim of canceling LetsEncrypt certificates...

eku, 2020-03-19:
@Jason2866  why does it load the pictures from an external site? Any reason for not embedding the pictures with the docs?

Jason2866, 2020-03-19:
Since the documetation is open source too. We recommend to host the pictures in our doc site.

effelle, 2020-03-22:
Done.

closed 2020-03-22 by effelle

eku, 2020-03-22:
@effelle still two pictures unresolved: https://pi-gate.net/images/menu.png and https://pi-gate.net/images/V4_setup.png.

effelle, 2020-03-22:
If you already knows the pages with the offending links please change them.
Just use the link on top of doc pages (Improve this article).