arendst/Tasmota #7949: IrRecv Unkown protocol when on load

jamucha, 2020-03-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am using esp12 with ws2812b and IR receiver. When I set scheme to any of 5-12 I cant controll device with IR anymore. In provided log, I was pressing same buton all the time.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:11:08 MQT: stat/ledm/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:11:08 MQT: stat/ledm/RESULT = {"Module":{"0":"Generic"}}
15:11:08 MQT: stat/ledm/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"51":"IRrecv"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"7":"WS2812"},"GPIO14":{"22":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
15:11:29 MQT: stat/ledm/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on Power1#state do POWER3 %value% endon"}
15:11:29 MQT: stat/ledm/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
15:11:30 MQT: stat/ledm/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

15:11:43 MQT: stat/ledm/STATUS = {"Status":{"Module":0,"FriendlyName":["LedM","Tasmota2","Tasmota3"],"Topic":"ledm","ButtonTopic":"0","Power":5,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:11:43 MQT: stat/ledm/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:33:55","StartupUTC":"2020-03-18T13:37:48","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":168,"SaveAddress":"F4000"}}
15:11:43 MQT: stat/ledm/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:41:58","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"356/699"}}
15:11:43 MQT: stat/ledm/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETIASPOT-DF8EA0","TURBO"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:11:43 MQT: stat/ledm/STATUS4 = {"StatusMEM":{"ProgramSize":660,"Free":340,"Heap":17,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:11:43 MQT: stat/ledm/STATUS5 = {"StatusNET":{"Hostname":"ledm-1444","IPAddress":"192.168.1.15","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"EC:FA:BC:27:C5:A4","Webserver":2,"WifiConfig":4}}
15:11:43 MQT: stat/ledm/STATUS6 = {"StatusMQT":{"MqttHost":"XXX.XXX.XXX.XXX","MqttPort":1883,"MqttClientMask":"LedM","MqttClient":"LedM","MqttUser":"X","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:11:43 MQT: stat/ledm/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 18 14:11:43 2020","Local":"Wed Mar 18 15:11:43 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"06:56","Sunset":"19:00"}}
15:11:43 MQT: stat/ledm/STATUS10 = {"StatusSNS":{"Time":"2020-03-18T15:11:43"}}
15:11:43 MQT: stat/ledm/STATUS11 = {"StatusSTS":{"Time":"2020-03-18T15:11:43","Uptime":"0T00:33:55","UptimeSec":2035,"Heap":16,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","POWER3":"ON","Dimmer":63,"Color":"A1A1A1","HSBColor":"201,0,63","Channel":[63,63,63],"Scheme":7,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXX","BSSId":"00:12:2A:DF:8E:A8","Channel":10,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:11"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
15:13:41 CMD: Scheme 0
15:13:41 SRC: WebConsole from 192.168.1.128
15:13:41 CMD: Group 0, Index 1, Command "SCHEME", Data "0"
15:13:41 MQT: stat/ledm/RESULT = {"Scheme":0}
15:13:41 CFG: Saved to flash at FA, Count 170, Bytes 4096
15:13:44 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:44 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:45 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:46 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:50 CMD: Scheme 5
15:13:50 SRC: WebConsole from 192.168.1.128
15:13:50 CMD: Group 0, Index 1, Command "SCHEME", Data "5"
15:13:50 MQT: stat/ledm/RESULT = {"Scheme":5}
15:13:50 CFG: Saved to flash at F9, Count 171, Bytes 4096
15:13:52 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":32,"Hash":"0xEDCBD6D8","Repeat":0}}
15:13:52 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:53 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:54 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":18,"Hash":"0xE6AE2099","Repeat":0}}
15:13:54 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:13:55 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FFB24D","DataLSB":"0x00FF4DB2","Repeat":0}}
15:14:00 WIF: Checking connection...
15:14:00 WIF: Connected
15:14:02 CMD: Scheme 6
15:14:02 SRC: WebConsole from 192.168.1.128
15:14:02 CMD: Group 0, Index 1, Command "SCHEME", Data "6"
15:14:02 MQT: stat/ledm/RESULT = {"Scheme":6}
15:14:02 CFG: Saved to flash at F8, Count 172, Bytes 4096
15:14:04 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":26,"Hash":"0x17D45754","Repeat":0}}
15:14:05 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":22,"Hash":"0xC1A24F88","Repeat":0}}
15:14:06 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":25,"Hash":"0xCA7425F7","Repeat":0}}
15:14:06 MQT: tele/ledm/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":29,"Hash":"0xA81579CC","Repeat":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use Scheme 5-12 and try to read ir message data.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct decoding of IR messages when device is on bigger load.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I see that when load is relatively small eg. Clock scheme sometimes messages are received correct. On Bigger load with schemes 6-12 messages are broken always.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-03-18:
Interesting bug. However I doubt we can make it any better. It looks like NeoPixel and IRRemoteESP8266 are competing on interrupts.

Jason2866, 2020-03-18:
You can try (needs self compile) NeoPixel DMA modus. The change is done here:
https://github.com/arendst/Tasmota/blob/development/tasmota/my_user_config.h#L427
As default Tasmota dont use Neopixel DMA because it has unwanted side effects for user usage.

jamucha, 2020-03-18:
Using DMA and changing data pin to gpio3 helped now works flawlessly. My issue is resolved as for me. If you dont plan to do anything about bug, please close thread.

Jason2866, 2020-03-18:
@ascillato2  it is not a bug. Esp82xx cant handle both interrupts since Neopixel generates too
much. Not enough resources left. This is the way how Neopixel does work in bitbang mode.
There is no solution since the ESP has not enough performance.
Only the DMA mode for Neopixel reduces the interrupts but this has many side effects for many
other features and can not be used as default

closed 2020-03-19 by ascillato2