arendst/Tasmota #7950: Sonoff POW R2 restarts every 10 minutes

viktak, 2020-03-18:
### PROBLEM DESCRIPTION
Device restarts with an exception almost every 10 minutes. I mean, sometimes it skips it, but it always happens at the multiple of 10 minute mark.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW R2
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: ESP flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```

18:06:57 CMD: Backlog Template; Module; GPIO 255
18:06:57 MQT: stat/washingmachine/RESULT = {"NAME":"Generic","GPIO":[17,145,0,146,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":43}
18:06:57 MQT: stat/washingmachine/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
18:06:57 MQT: stat/washingmachine/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
18:19:04 CMD: Status 0
18:19:04 MQT: stat/washingmachine/STATUS = {"Status":{"Module":43,"FriendlyName":["Tasmota"],"Topic":"washingmachine","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:19:04 MQT: stat/washingmachine/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:58:06","StartupUTC":"2020-03-18T16:20:58","Sleep":0,"CfgHolder":4617,"BootCount":16,"SaveCount":99,"SaveAddress":"F9000"}}
18:19:04 MQT: stat/washingmachine/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"326/699"}}
18:19:04 MQT: stat/washingmachine/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gunther",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00000000"]}}
18:19:04 MQT: stat/washingmachine/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":255,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:19:04 MQT: stat/washingmachine/STATUS5 = {"StatusNET":{"Hostname":"washingmachine-5712","IPAddress":"192.168.1.145","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"A4:CF:12:B6:16:50","Webserver":2,"WifiConfig":4}}
18:19:04 MQT: stat/washingmachine/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.99","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B61650","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:19:04 MQT: stat/washingmachine/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 18 17:19:04 2020","Local":"Wed Mar 18 18:19:04 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"06:56","Sunset":"19:00"}}
18:19:04 MQT: stat/washingmachine/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:19:04 MQT: stat/washingmachine/STATUS10 = {"StatusSNS":{"Time":"2020-03-18T18:19:04","ENERGY":{"TotalStartTime":"2020-03-17T17:03:00","Total":0.245,"Yesterday":0.026,"Today":0.219,"Power":94,"ApparentPower":94,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.414}}}
18:19:04 MQT: stat/washingmachine/STATUS11 = {"StatusSTS":{"Time":"2020-03-18T18:19:04","Uptime":"0T00:58:06","UptimeSec":3486,"Heap":25,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"Gunther","BSSId":"48:F8:B3:62:7A:D3","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":2,"Downtime":"0T00:00:06"}}}
18:19:04 MQT: stat/washingmachine/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
18:16:06 CMD: weblog 4
18:16:06 MQT: stat/washingmachine/RESULT = {"WebLog":4}
18:16:06 DMP: 55 5A 02 D4 38 00 03 2E 00 3D 0E 00 93 92 4D 26 00 00 D2 E6 71 3B 19 9F
18:16:06 CFG: Saved to flash at FA, Count 98, Bytes 4096
18:16:06 DMP: 55 5A 02 D4 38 00 03 2E 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 1B C2
18:16:06 DMP: 55 5A 02 D4 38 00 03 2E 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 1D C4
18:16:06 DMP: 55 5A 02 D4 38 00 03 2E 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 1F C6
18:16:06 DMP: 55 5A 02 D4 38 00 03 2E 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 21 C8
18:16:06 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 23 C7
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 25 C9
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 27 CB
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 29 CD
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 2B CF
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 2C D0
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D3 06 71 3B 2E D2
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D2 BE 71 3B 30 8B
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D2 BE 71 3B 32 8D
18:16:07 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D2 BE 71 3B 34 8F
18:16:08 DMP: 55 5A 02 D4 38 00 03 2B 00 3D 0E 00 93 92 4D 26 00 00 D2 BE 71 3B 36 91
[... 95 lines cut ...]
18:16:27 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 96 4D 26 00 00 D3 24 71 3C A1 69
18:16:27 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 96 4D 26 00 00 D3 24 71 3C A3 6B
18:16:27 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 A5 4D 26 00 00 D3 24 71 3C A5 7C
18:16:27 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 A5 4D 26 00 00 D3 24 71 3C A7 7E
18:16:27 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 A5 4D 26 00 00 D3 1C 71 3C A9 78
18:16:28 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 A5 4D 26 00 00 D3 1C 71 3C AB 7A
18:16:28 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 A5 4D 26 00 00 D3 1C 71 3C AD 7C
18:16:28 DMP: 55 5A 02 D4 38 00 03 2C 00 3D 0E 00 93 A5 4D 26 00 00 D3 1C 71 3C AF 7E

```

### TO REPRODUCE
No special steps required. It is a brand new Sonoff POW R2, my first. I have flashed and used other devices with Tasmota successfully.

### EXPECTED BEHAVIOUR
No resets every 10 minutes.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
During researching this issue I came across issue #3385, which is closed, but my problem is a bit different, so I opened a new issue.
#3385 claims that the problem is caused by compiling it on Linux. Well, I didn't compile it at all, but used the publicly released binaries.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-18:
Hi,

This type of issue has been addressed before several times. Most of the time is due to not erasing before flashing. 

Please, try as recommended:

Go to the console and type:
`Reset 5` (this will erase all your settings except wifi credentials and will clean the flash)
Then type:
`Reset 3` (this will recalibrate the sdk settings of the wifi radio) after it restart you need to de-energize the device, wait few seconds and re-energize it again. 

After that, please update to latest Tasmota from development branch ( http://thehackbox.org/tasmota/tasmota.bin ) This version has a lot of fixes. 

Please, report back if this solves your issue.

ascillato2, 2020-03-19:
Please, when you have time, tell us if that solved your issue.

If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-19 by ascillato2

viktak, 2020-03-19:
Since last night I let it run, and all night long it only restarted once.
Tomorrow I will do all the steps you suggested above and will report back.

Thanks!

ascillato, 2020-03-19:
Please, try it. Thanks.

viktak, 2020-03-19:
ok, I had some time now and I have done all the steps suggested above:

Reset 5
Reset 3
Updated to the latest version (8.1.0.11)

Set it up again for my mqtt server, etc, and now I am leaving it for overnight.

viktak, 2020-03-20:
Overnight results:

Approximately 20 times it happened:
```
tele/washingmachine/LWT "Online"
```
however, this time, on the `Information` page it says this:
```
Uptime | 0T13:52:46
Restart Reason | Software/System restart
```

So this means it was NOT restarted. For some reason, it still sent a few `Online` messages.
What could cause it?

Again, all these messages on MQTT came at intervals of an integer multiple of 10 minutes, i.e. 20:45, 20:55, 21:25, 21:35, ...

ascillato, 2020-03-20:
Ok, so if your uptime is high and you don't have exceptions, your device is working fine. Your problem is on the wifi/mqtt connection.

If you type `status 0` on the console, you will see **mqttcount** for the times that mqtt have reconnected and **linkcount** for the times that wifi have reconnected.

If linkcount is high, your problem is wifi. You need to check your network structure and settings. If mqttcount is high you need to check your mqtt broker config.