arendst/Tasmota #7951: MCP23017 output not working on new release

megamarco833, 2020-03-18:
### PROBLEM DESCRIPTION
MCP23017 not respond if a pin is configurated as output

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nodemcu
- [x] Tasmota binary firmware version number used: 8.1.0.11
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: gitpod
- [x] Flashing tools used: ESP8266Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```21:08:56 CMD: Backlog Template; Module; GPIO 255
21:08:56 RSL: stat/tasmota_81E0EB/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:08:56 RSL: stat/tasmota_81E0EB/RESULT = {"Module":{"18":"Generic"}}
21:08:57 RSL: stat/tasmota_81E0EB/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}```

- [ ] Provide the output of this command: `Status 0`:
```
21:09:51 CMD: Status 0
21:09:51 RSL: stat/tasmota_81E0EB/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"tasmota_81E0EB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:09:51 RSL: stat/tasmota_81E0EB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:06","StartupUTC":"2020-03-18T20:01:45","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-03-18T21:01:10","SaveCount":12,"SaveAddress":"F8000"}}
21:09:51 RSL: stat/tasmota_81E0EB/STATUS2 = {"StatusFWR":{"Version":"8.1.0.11(tasmota)","BuildDateTime":"2020-03-18T19:49:32","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"312/699"}}
21:09:51 RSL: stat/tasmota_81E0EB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:09:51 RSL: stat/tasmota_81E0EB/STATUS4 = {"StatusMEM":{"ProgramSize":584,"Free":416,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FDAE397","043683A0","000000CD","010013F1","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,29"}}
21:09:52 RSL: stat/tasmota_81E0EB/STATUS5 = {"StatusNET":{"Hostname":"tasmota_81E0EB-0235","IPAddress":"192.168.0.33","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.8","Mac":"84:0D:8E:81:E0:EB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:09:52 RSL: stat/tasmota_81E0EB/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_81E0EB","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:09:52 RSL: stat/tasmota_81E0EB/STATUS7 = {"StatusTIM":{"UTC":"2020-03-18T20:09:52","Local":"2020-03-18T21:09:52","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:56","Sunset":"19:00"}}
21:09:52 RSL: stat/tasmota_81E0EB/STATUS10 = {"StatusSNS":{"Time":"2020-03-18T21:09:52","MCP230XX":{"D0":1,"D1":0,"D2":0,"D3":0,"D4":0,"D5":0,"D6":0,"D7":0,"D8":1,"D9":1,"D10":1,"D11":0,"D12":0,"D13":0,"D14":0,"D15":1}}}
21:09:52 RSL: stat/tasmota_81E0EB/STATUS11 = {"StatusSTS":{"Time":"2020-03-18T21:09:52","Uptime":"0T00:08:07","UptimeSec":487,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"80:3F:5D:6E:37:67","Channel":13,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
21:02:08 CMD: sensor29 8,5,2
21:02:08 RSL: stat/tasmota_81E0EB/RESULT = {"Command":"Error"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
just compile including MCP23017 with input and output

### EXPECTED BEHAVIOUR
it works if pin is setted as input
not works if pins (every one of pin from 0 to 15) is setted as output

Jason2866, 2020-03-19:
@andrethomas  Can you look into? Thx!

effelle, 2020-03-19:
I still had on MCP in my magic drawer and I tested it now.
The docs are a little misleading but the output work as expected for me.
`sensor29 8,5,2` means enable pin 8 as normal output (5) and the last parameter (2) will rise an error because the value can be just `0` or `1`. 
When using OUTPUT it specify if the pin must be OFF or ON on reset/power-up.
The syntax returns this structure: `sensor29 pin,pinmode,pullup,intmode` and this is what you should use for OUTPUT:
```
15:18:55 CMD: sensor29 8,5,0,0
15:18:55 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
15:19:02 CMD: sensor29 8,5,0,1
15:19:02 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
15:19:06 CMD: sensor29 8,5,0,2
15:19:06 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
```
Now a test using just 3 parameters, meaning we will not specify `intmode`:
```
15:20:33 CMD: sensor29 8,5,1
15:20:33 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
15:20:41 CMD: sensor29 8,5,0 
15:20:41 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
```

```
15:20:46 CMD: sensor29 8,5,2 
15:20:46 MQT: stat/tasmota_2350B0/RESULT = {"Command":"Error"}
```
Can't work because the index of the third param goes from 0 to 1.
```
15:20:56 CMD: sensor29 8,5,0,0
15:20:56 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
15:20:59 CMD: sensor29 8,5,0,1
15:20:59 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
15:21:01 CMD: sensor29 8,5,0,2
15:21:01 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"OFF","INT_MODE":"DISABLED","STATE":"ON"}}
15:21:10 CMD: sensor29 8,5,1,0
15:21:10 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}
15:21:18 CMD: sensor29 8,5,1,1
15:21:18 MQT: stat/tasmota_2350B0/RESULT = {"Sensor29_D8":{"MODE":5,"PULL_UP":"ON","INT_MODE":"DISABLED","STATE":"ON"}}

```
```
15:21:17 CMD: sensor29 8,5,2,0
15:21:17 MQT: stat/tasmota_2350B0/RESULT = {"Command":"Error"}
```
Again, can't work because the index of the third param goes from 0 to 1.

ascillato2, 2020-03-19:
Thanks :+1: 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-03-19 by ascillato2

megamarco833, 2020-03-19:
> I still had on MCP in my magic drawer and I tested it now.
> The docs are a little misleading but the output work as expected for me.
> `sensor29 8,5,2` means enable pin 8 as normal output (5) and the [...]

hi and thanks for reply.
but i'm confused because of inside wiki i read:
`pin,pinmode{,intpullup|outstate{,repmode}}`
> outstate (pinmode 5..6) = set the default state of an OUTPUT pin on reset/power-up. If your device is configured to save state (SetOption0 = 1), the outstate setting will be ignored and the last known state of the pin will be applied during power-up/reset.
> 
>     0/off = set output pin to OFF
>     1/on = set output pin state to ON
>     2/toggle = toggle output pin state

so i understand that wih outstate (5) i can set the outstate 0 or 1 or 2