arendst/Tasmota #7953: Different Log Level for Requests from TasmoAdmin

BBBits, 2020-03-19:
**Have you looked for this feature in other issues and in the docs?**  
briefly 
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Logs are flooded when using tasmoadmin . can only stop flood by setting level to error or off.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
An extra Log level added before INFO called Min
This would log all basic startup info ,IP, IO & config changes Only .
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Somehow filter status requests from tasmoadmin and give them a DEBUG log level
**Additional context**  
_Add any other context or screenshots about the feature request here._
none
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-19:
That is the way TasmoAdmin works. It sends by HTTP the `STATUS 0` command every preconfigured interval (you can change that inside TasmoAdmin).

The logs that are shown in the Tasmota Webconsole are intended because something external is querying information. Anyway those are there only when TasmoAdmin page is being shown and don't affect Tasmota functionality.

If you set WEBLOG to 0, TasmoAdmin won't work. You won't have logs but you won't be able to use TasmoAdmin neither.

If you try other alternative managers like TDM (that uses MQTT) or TasUI (that uses CORS), you will see also the logs of those querying Tasmota. (Links to TDM and TasUI are available in the docs.)

Tasmota produces a response for every command, and that response is shown using the logs system. You can set WEBLOG, SERIALLOG, MQTTLOG and SYSLOG to see those responses. So far, Tasmota has the same response for all channels and no matter where the command comes from.

What you are asking is that Tasmota should show the response of all HTTP commands received but only into debug log level of the web console, however the HTTP responses must continue like before - at info log level - (if not TasmoAdmin won't receive any response). This would require some refactoring due Tasmota internally uses the console to run those commands. The output of the console, mqtt and http responses are the same thing.

So, as it is needed a rework of the webserver driver just to not show a log in the web console but to send the data by http, adding more RAM and FLASH, I don't think is going to be done soon. Sorry.

BBBits, 2020-03-19:
Perhaps simply adding a MIN or BASIC log level between INFO & ERROR .
This could silence the logs a lot for general purpose operation

ascillato, 2020-03-19:
As Tasmota is now, if you don't see a response message in the web console, no response is being transmitted to TasmoAdmin. So, as Tasmota is now (the console is the heart), your request is not feasible without a major refactoring. Too much work and more RAM/FLASH usage for just a log in the console (that is intended and designed to be there) is not feasible to be done. Sorry.

ascillato, 2020-03-19:
If you want TasmoAdmin and others to work, the response of the requests they do, must be shown in the console. If not, there is no response.

BBBits, 2020-03-19:
OK , just checked the note about weblog.
Dang , this is a tad annoying that the weblog is req for it to work .
Guess it's a nogo then .
Thanks for info

closed 2020-03-19 by BBBits

BBBits, 2020-03-19:
Just had a thought to unclutter the console.
would still require the addition of a MIN level between INFO & ERROR & a lot of changes to the new Level.
Perhaps there could be a second syslog port & tasmoadmin could receive status without querying the console.

Jason2866, 2020-03-19:
Will not be done, same reason(s) as explained before AND Tasmota will not add code for a 3rd party tool which is hammering device with http requests just for getting infos. Doing this generates high MCU load. Tasmota is designed to be used via mqtt.