arendst/Tasmota #7955: Enforce Home Assistant auto-discovery as fan

ryanjohnsontv, 2020-03-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes, previously marked as stale in #3424

**Is your feature request related to a problem? Please describe.**  
Currently limited to only being able to change a relay to appear as a light within Home Assistant using auto-discovery

**Describe the solution you'd like**  
Very similar to how SetOption30 sets a relay to appear as a light within Home Assistant, I feel like allowing a relay that controls a single speed fan to appear as a fan would be a welcome implementation.

**Describe alternatives you've considered**  
I am currently routing the switch into a template fan within Home Assistant to aid in customizing my entities. I've also configured the device in YAML within Home Assistant, however this bypasses auto-discovery altogether.

**Additional context**  
Related to: https://github.com/arendst/Tasmota/commit/1c6f752ba8f774d34d5379fe8396663fab9c165a

**(Please, remember to close the issue when the problem has been addressed)**

ThomDietrich, 2020-03-21:
Hello @ryanjohnsontv,

> I am currently routing the switch into a template fan within Home Assistant to aid in customizing my entities.

Why is that not an adequate solution? If you have a single speed fan you can really just turn it on and off, it is thereby logically more a switch than it is a fan. A fan in HA is a device whose speed you can change. You do not need this functionality.

effelle, 2020-03-22:
@ryanjohnsontv thanks for your suggestion but even if `fan` is a valid domain as `light` and `switch` I don't see the advantage here, at least for your user case. 
A fan with just a state is just a switch, no more, no less.
Will be different in case we will start to support differently fan domain but is not the case, at least not in the near future.
For now I'm against and I will not implement it on Discovery. 
Lets see if someone else is interested in or have some ideas to share.

ryanjohnsontv, 2020-03-24:
@effelle @ThomDietrich I was mostly interested in this added functionality to minimize the number of switches I have that are ultimately fed back into fan templates within Home Assistant. This also causes several automations to not function as intended when I call for all switches to turn off/on but not fans, because my fans are still toggled. Obviously this is incredibly contrived all things considered, but I at least wanted to express interest to see if if could be adopted. I was previously setting up my Sonoff S31 units in ESPHome and would output it to HA as a binary fan which worked perfectly, and this functionality is the only downside I've encountered since switching.

closed 2020-04-23 by stale[bot]