arendst/Tasmota #7956: Flashing LED Status code decoder

RoganDawes, 2020-03-20:
**Have you looked for this feature in other issues and in the docs?**  

I have looked for this in Tasmota, and elsewhere, but have not found anything related. I have also asked in the Discord channel with no responses.

**Is your feature request related to a problem? Please describe.**  
Some devices report their status using a flashing LED. Different patterns mean different things. I'd like to be able to interpret the flashing patterns and report them using a meaningful description.

**Describe the solution you'd like**  
I'd like to be able to configure Tasmota to record (and ideally interpret) the pattern of a GPIO toggling over a period of time. This is different to a counter or a PWM sensor as the gaps between the individual state transitions is also important, not just the number of state transitions, or the average rate of them.

**Describe alternatives you've considered**  
I have looked at Counters and PWM sensors, but I believe that this problem is sufficiently different that those will not be able to address this use case.

**Additional context**  
Status light patterns for Centurion D5 gate motor:
Off - Gate is closed.
Slow continuous flash - Gate is opening.
On - Gate is open.
Fast continuous flash - Gate is closing.
1 Flash/sec - Pillar light on permanently.
2 Flashes/sec - No mains.
3 Flashes/sec - Battery low detection.
4 Flashes/sec - Collision detector operated.
5 Flashes/sec - Microprocessor reset.

I think the above may be slightly misleading. n flashes/sec indicates n flashes at about 4Hz, then a 1 second gap, from what I can make out.

**(Please, remember to close the issue when the problem has been addressed)**

RoganDawes, 2020-03-20:
I'm imagining a configuration where you can specify the maximum constant state time e.g. 1000ms to extract a state from solid on and solid off, as well as to indicate when a non-continuous pattern ends. Then an inter-transition period to allow distinguishing between slow and fast flashes. This would obviously be subject to tuning to get the best results.

arendst, 2020-03-20:
Have a look at buzzer functionality. It is implemented in such a way that you might connect a led instead of a buzzer and have many different flash patterns as you like.

See docs how to use a buzzer.

RoganDawes, 2020-03-20:
I think you misunderstand me. I want to go from watching an LED flash (aka tracking a GPIO) to "Gate is Opening". I'm not interested in making an LED flash myself.

RoganDawes, 2020-03-20:
The closest thing I have found so far is the ESPHome “Duty cycle” mechanism. Unfortunately this won’t work for many situations, for example, distinguishing between fast and slow flashing, which may both have the same 50% duty cycle, but mean different things.

RoganDawes, 2020-03-20:
This is essentially the same problem as decoding radio signals, I suppose, or data from a logic analyser. The main thing as I see it is to figure out what the minimum period is, and then to convert the GPIO state at each period into a 1 or 0, and shift that along in e.g. a 32-bit word, or else an array of bytes.

ascillato, 2020-03-20:
As your problem is quite complex and unique, it would be better that you make the recognition routine on an arduino nano for example and then send by serial to Tasmota the processed data.

Tasmota supports TasmotaSlave to flash by OTA an arduino attached. See docs for more information.

Sorry, but as this is feature request requires a lot of work and it is for just 1 usecase (yours), seems that you will need to develop that yourself. Anyway, as this is open source, you can fork Tasmota and do the modifications you need. Then if you want, you can share your modifications.

Leaving this request open for some days to check if someone else is interested on working on this. Thanks for sharing your ideas.

ascillato2, 2020-04-09:
Closing as nobody is working on this. Thanks for sharing your ideas.

closed 2020-04-09 by ascillato2