arendst/Tasmota #7964: 8.2.0 OTA Update seems to be broken

LiloBzH, 2020-03-21:
**ISSUE HERE : https://github.com/arendst/Tasmota/issues/7963**

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-21:
No problem at all.

```
12:57:07 CMD: otaurl http://thehackbox.org/tasmota/release/tasmota-FR.bin
12:57:07 SRC: Serial
12:57:07 CMD: Group 0, Index 1, Command "OTAURL", Data "http://thehackbox.org/tasmota/release/tasmota-FR.bin"
12:57:07 CFG: CR 352/699
12:57:07 MQT: stat/wemos4/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-FR.bin"}
12:57:08 CFG: Saved to flash at FB, Count 53, Bytes 4096
12:57:13 CMD: upload 1
12:57:13 SRC: Serial
12:57:13 CMD: Group 0, Index 1, Command "UPLOAD", Data "1"
12:57:13 MQT: stat/wemos4/RESULT = {"Upload":"Version 8.1.0.11 from http://thehackbox.org/tasmota/release/tasmota-FR.bin"}
12:57:14 CFG: Saved to flash at FB, Count 54, Bytes 4096
12:57:16 HTP: Web server stopped
12:57:16 UPL: http://thehackbox.org/tasmota/release/tasmota-FR.bin
12:57:17 HTP: Web server active on wemos4 with IP address 192.168.2.223
12:57:18 HTP: Web server stopped
12:57:18 UPL: http://thehackbox.org/tasmota/release/tasmota-minimal.bin
12:57:30 HTP: Web server active on wemos4 with IP address 192.168.2.223
12:57:30 MQT: Attempting connection...
12:57:30 MQT: Connected
12:57:30 MQT: tele/wemos4/LWT = Online (retained)
12:57:30 MQT: cmnd/wemos4/POWER = 
12:57:30 MQT: Subscribe to cmnd/wemos4/#
12:57:30 MQT: Subscribe to cmnd/tasmotas/#
12:57:30 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
12:57:31 OTA: Version 0x08020000, Compatible 0x07010006
12:57:31 MQT: stat/wemos4/UPGRADE = {"Upgrade":"Successful. Restarting"}
12:57:33 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
v372a3ec2
@cp:0
ld
00:00:00 CFG: Loaded from flash at FB, Count 54
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 8.2.0(minimal)-STAGE
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -87, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -38, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on wemos4 with IP address 192.168.2.223
12:57:48 NTP: UTC 2020-03-21T11:57:47, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:57:49 MQT: Attempting connection...
12:57:49 MQT: Connected
12:57:49 MQT: tele/wemos4/LWT = Online (retained)
12:57:49 MQT: cmnd/wemos4/POWER = 
12:57:49 MQT: Subscribe to cmnd/wemos4/#
12:57:49 MQT: Subscribe to cmnd/tasmotas/#
12:57:49 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
12:57:49 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"8.2.0(minimal)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
12:57:49 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
12:57:49 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
12:57:49 MQT: stat/wemos4/RESULT = {"POWER":"ON"}
12:57:49 MQT: stat/wemos4/POWER = ON
12:57:50 HTP: Web server stopped
12:58:45 UPL: http://thehackbox.org/tasmota/release/tasmota-FR.bin
[... 29 lines cut ...]
00:00:02 WIF: Tentative de connexion...
00:00:03 WIF: Tentative de connexion...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -91, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -33, Encryption 1
00:00:04 WIF: Connexion Ã  l'AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 en mode 11N comme wemos4...
00:00:04 WIF: Tentative de connexion...
00:00:05 WIF: ConnectÃ©
00:00:05 HTP: Serveur web actif sur wemos4 avec l'adresse IP 192.168.2.223
12:59:25 NTP: UTC 2020-03-21T11:59:24, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:59:26 MQT: Tentative de connexion...
12:59:26 MQT: ConnectÃ©
12:59:26 MQT: tele/wemos4/LWT = ConnectÃ© (persistant)
12:59:26 MQT: cmnd/wemos4/POWER = 
12:59:26 MQT: S'abonner Ã  cmnd/wemos4/#
12:59:26 MQT: S'abonner Ã  cmnd/tasmotas/#
12:59:26 MQT: S'abonner Ã  cmnd/DVES_83BB10_fb/#
12:59:26 MQT: tele/wemos4/INFO1 = {"Module":"Generic","Version":"8.2.0(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
12:59:26 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
12:59:26 MQT: tele/wemos4/INFO3 = {"RestartReason":"Software/System restart"}
12:59:26 MQT: stat/wemos4/RESULT = {"POWER":"ON"}
12:59:26 MQT: stat/wemos4/POWER = ON
12:59:27 APP: Nombre de boot 19
12:59:27 CFG: EnregistrÃ© en flash Ã  FA, Compte 56, Bytes 4096
12:59:30 MQT: tele/wemos4/STATE = {"Time":"2020-03-21T12:59:30","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:07"}}
```

LiloBzH, 2020-03-21:
yeah i know. It s not really in 404. I can download this file with this URL.

So it's not a bug hum...

i will keep looking

Jason2866, 2020-03-21:
Have you a mesh wifi network or repeaters? Esp82xx is known not to work always with.
This is a SDK (core) issue and can NOT be solved in Tasmota code.

LiloBzH, 2020-03-21:
it's not a wifi problem because upgrate OTA work every time before this version

UPgrade with file don t work hum

Upload Échoué
Erreur de comparaison du buffer de téléchargement

LiloBzH, 2020-03-21:
hum with file, it s space problem i suppose. it's probably normal

LiloBzH, 2020-03-21:
13:49:40 MQT: stat/tasmota/RESULT = {"OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-FR.bin"}
13:49:40 MQT: stat/tasmota/RESULT = {"Upgrade":"Version 8.1.0.3 from http://thehackbox.org/tasmota/release/tasmota-FR.bin"}
13:49:40 MQT: tele/tasmota/STATE = {"Time":"2020-03-21T13:49:40","Uptime":"0T00:00:12","UptimeSec":12,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"lilobzh-Freebox","BSSId":"2C:xx:xx:xx:xx:xx","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:49:43 HTP: Serveur web éteint
13:49:44 HTP: Serveur web actif sur tasmota-0165 avec l'adresse IP 192.168.0.150
13:49:45 HTP: Serveur web éteint
13:49:46 HTP: Serveur web actif sur tasmota-0165 avec l'adresse IP 192.168.0.150
13:49:47 HTP: Serveur web éteint
13:49:48 HTP: Serveur web actif sur tasmota-0165 avec l'adresse IP 192.168.0.150
13:49:49 HTP: Serveur web éteint
13:49:50 HTP: Serveur web actif sur tasmota-0165 avec l'adresse IP 192.168.0.150
13:49:51 HTP: Serveur web éteint
13:49:52 HTP: Serveur web actif sur tasmota-0165 avec l'adresse IP 192.168.0.150
13:49:53 MQT: stat/tasmota/UPGRADE = {"Upgrade":"Failed File Not Found (404)"}

It's not a problem with 8.1.0.3 version ?

LiloBzH, 2020-03-21:
So

I force upgrade with MINIMAL firmware

I upgrade with webinterface to 8.2 Minimal and full firmware was DL and Istalled with OTA url automatically i suppose

closed 2020-03-21 by LiloBzH

aik, 2020-07-31:
Upgrade to minimal works but the minimal firmware cannot upgrade to anything but minimal. NodeMCU v1.0, tasmota-lite.bin 485,552.

aik, 2020-07-31:
the firmware size seems to be the key here, what are the requirements for a 1MB flash device?
tasmota.bin.gz  437,094  or 619,376 unpacked - did not boot.
tasmota.bin.gz  380,315  or 539,696 unpacked - works.
Both packed sizes are less than a half of ROM and both unpacked are more than a half.

marine1988, 2021-12-14:
I have same problem when update truth web server

benziman88, 2022-03-18:
Hello everyone not sure if I'm in the right spot here.
running tasmota 8.1.0 on a sonoff 4 ch pro and can't get it to update to 8.5.1 minimial (and to go from there up the versions as stated in the guide).

I tried everything: minimal.bin, minimal.bin.gz, lite.bin and lite.bin.gz ...
bin.gz gives magic byte is not 0xe9 error and the bin file gives upload buffer error

Even tried OTA Update which just restarts the device.
Here are some infos about my device not sure where the issue is as I simply bought in on the web with tasmota already isntalled

[image]

can anyone help me please!

arendst, 2022-03-18:
How hard can this be.

Following the ReleaseNotes all you have to do is:

1) change your otaurl to ``http://ota.tasmota.com/tasmota/release-8.5.1/tasmota.bin`` and do a firmware upgrade.
2) change your otaurl to ``http://ota.tasmota.com/tasmota/release-9.1.0/tasmota.bin.gz`` and do a firmware upgrade.
3) change your otaurl to ``http://ota.tasmota.com/tasmota/release/tasmota.bin.gz`` and do a firmware upgrade.

No fooling around with minimal, light or whatever.

Ofcourse this assumes you initially installed Tasmota as 1M binary image and not fooled around with a 4M image. This can be seen from the rest of your information page.

Good luck.

benziman88, 2022-03-18:
Thank you for the fast reply.
I did what you suggested but the sonoff says restarting device and thats it.
I also tried downloading the file and uplaoding it which gives upload buffer error.

here is the other info about my device
[image]

I did the minimal.bin because the upgrade path said so due to lack of buffer size.

arendst, 2022-03-18:
Ah I see.

Your available space is too small for even the minimal version. That's strange as the default 8.1.0 tasmota.bin is around 580k instead of your reported 630k.

Let me see if I can make a minimal that fits in your tight memory space (will take some time though).

benziman88, 2022-03-18:
Wow this would ne great! Your help is greatly appreciated!

Jason2866, 2022-03-18:
@benziman88 You can use this https://github.com/Jason2866/ESP8266MinimalHTTPUpdater
Load the sonoff-basic v.8.1.0 with this tool and follow the normal upgrade flow. 

@arendst No need for a special minimal ;-)

arendst, 2022-03-18:
If the above doesn't work retry with the procedure I mentionend earlier.

The minimal binary is now shrinked to 357k so it should work.

benziman88, 2022-03-18:
Hi thanks for the reply. I tried to upload the file in build_output/firmware as mentioned in the usage field of your link.
If I upload the .gz file it says magic byte is not 0xE9 and if I try to upload the bin file it says not comaptible.

I also tried the minimal again (thanks for shrinking it so fast!) error remains magic byte is not 0xE9.

didn't think this would be so tough

arendst, 2022-03-18:
Try this:

- Open the console in the GUI
- Enable more logging with ``weblog 3``
- Verify otaurl is set to ``otaurl http://ota.tasmota.com/tasmota/release-8.5.1/tasmota.bin``
- Perform an upgrade with command ``upgrade 1``

Now wait and watch the screen reporting what happens. The whole operation takes at least TWO minutes before the final image is loaded (it first loads the minim al which in turn loads the final bianry). DO NOT TOUCH THE GUI or KEYBOARD.

- If you see the new version of binary has loaded reporting ``8.5.1(tasmota)`` continue with the upgrade
- Change otaurl with ``otaurl http://ota.tasmota.com/tasmota/release-9.1.0/tasmota.bin.gz``
- Perform an upgrade with command ``upgrade 1``

Now wait and watch the screen reporting what happens. The whole operation takes at least TWO minutes before the final image is loaded (it first loads the minim al which in turn loads the final bianry). DO NOT TOUCH THE GUI or KEYBOARD.

- If you see the new version of binary has loaded reporting ``9.1.0(tasmota)`` continue with the upgrade
- Change otaurl with ``otaurl http://ota.tasmota.com/tasmota/release/tasmota.bin.gz``
- Perform an upgrade with command ``upgrade 1``

By now you should grasp the procedure....

Jason2866, 2022-03-18:
> Hi thanks for the reply. I tried to upload the file in build_output/firmware as mentioned in the usage field of your link. If I upload the .gz file it says magic byte is not 0xE9 and if I try to [...]

Have you used the github download button? Using Browser right click save will download the webpage **not** the file!

benziman88, 2022-03-18:
Yes, downloaded and unzzipped the package.
[image]

Jason2866, 2022-03-18:
Ah, there is a firmware compability check. Switch off with command `SetOption78 1`
before you upload the non tasmota formware

benziman88, 2022-03-18:
> Try this:
>
>     * Open the console in the GUI
>
>     * Enable more logging with `weblog 3`
>
>     * Verify otaurl is set to `otaurl http://ota.tasmota.com/tasmota/release-8.5.1/tasmota.bin`
>
>     * [...]

now this did the trick! will continue to upgrade untill current version
[image]

benziman88, 2022-03-18:
so this is it, thank you everyone for the fast help! Appreciate it, will keep my tasmota up to date now :D
[image]

CNC-at-Home, 2023-01-02:
Thanks also from me, the thread has helped me a lot