arendst/Tasmota #7965: Independent PWM channels via HA automatic discovery

ThomDietrich, 2020-03-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes, didn't find one

**Is your feature request related to a problem? Please describe.**  

This request addresses a limitation of the current Home Assistant MQTT discovery support.

The scenario: A device with more than one independent LED strips, configured via `SetOption68 1` as described in https://tasmota.github.io/docs/#/Lights?id=independent-pwm-channels

The result:

[image]

UI and control works as expected, however,

```json
"homeassistant/light/B6DB61_LI_1/config":
{"name":"Küche Bar-LEDs","stat_t":"B6DB61/tele/STATE","avty_t":"B6DB61/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"B6DB61/cmnd/POWER1","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","uniq_id":"B6DB61_LI_1","dev":{"ids":["B6DB61"]}}

"homeassistant/light/B6DB61_LI_2/config":
{"name":"Küche Unterschrank-LEDs","stat_t":"B6DB61/tele/STATE","avty_t":"B6DB61/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"B6DB61/cmnd/POWER2","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","uniq_id":"B6DB61_LI_2","dev":{"ids":["B6DB61"]},"bri_cmd_t":"B6DB61/cmnd/Dimmer","bri_stat_t":"B6DB61/tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","clr_temp_cmd_t":"B6DB61/cmnd/CT","clr_temp_stat_t":"B6DB61/tele/STATE","clr_temp_val_tpl":"{{value_json.CT}}"}
```

discovery messages are still sent as if the device was configured to `SetOption68 1`

**Describe the solution you'd like**  

```json
"homeassistant/light/B6DB61_LI_1/config":
{"name":"Küche Bar-LEDs","stat_t":"B6DB61/tele/STATE","avty_t":"B6DB61/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"B6DB61/cmnd/POWER1","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","uniq_id":"B6DB61_LI_1","dev":{"ids":["B6DB61"]},"bri_cmd_t":"B6DB61/cmnd/Dimmer1","bri_stat_t":"B6DB61/tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer1}}"}

"homeassistant/light/B6DB61_LI_2/config":
{"name":"Küche Unterschrank-LEDs","stat_t":"B6DB61/tele/STATE","avty_t":"B6DB61/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"B6DB61/cmnd/POWER2","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","uniq_id":"B6DB61_LI_2","dev":{"ids":["B6DB61"]},"bri_cmd_t":"B6DB61/cmnd/Dimmer2","bri_stat_t":"B6DB61/tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer2}}"}

"homeassistant/light/B6DB61_LI_3/config": ...
```

**Describe alternatives you've considered**  

Documentation could capture this limitation

**Additional context**  
Happy to provide more details.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-21:
It's a know issue. A solution will take some time.

ThomDietrich, 2020-03-21:
"Good" to know. Thanks for the answer. You wanna reference and close this issue or keep it for tracking?

arendst, 2020-03-21:
Let's keep it for tracking

effelle, 2020-03-21:
On my todo list.

PR #8370 (2020-05-07): New HAss Discovery (4/4): RELAYS/LIGHTS
## Description:
Changelog:
- New management for Lights discovery, using directly `LIGHT` structure;
- Add support for `SetOption37` for color remapping for led channels and independent handling of RGB and white channels;
- Add support for `SetOption68` for multi-channel PWM instead of a single light;
- Add a failsafe to warning about the wrong order of Relays ID when a Light is present:
 **it will block the MQTT generation for Relays/Lights until the issue is fixed**;
- New management for Relays discovery, using directly the GPIOs map;
- Extended the use [... cut]
closed 2020-05-07 by arendst