arendst/Tasmota #7966: BMP BME Sensors don't work after Update from 8.1.0.10 to 8.2.0

Chaot1, 2020-03-21:
I have different BMP280, BMP 180 and BME280 Sensors attached to my Tasmotas. 
After Update to 8.2.0 my Sensors are not longer recognized and no Data is displayed.
Th last working Firmware is 20200227-190516-fb87a03-tasmota-DE

Do i need to change any settings? I don't find a mistake.

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pfanntec, 2020-03-21:
I can confirm that. Unfortunately the BME280 is not working anymore in 8.2.0.
An I2C scan from the console will find the sensor.
{"I2CScan": "Device(s) found at 0x76"}

arendst, 2020-03-21:
See release notes

Version 8.2.0 Elliot
- Change default my_user_config.h driver and sensor support removing most sensors and adding most drivers to tasmota.bin

helmar74, 2020-03-21:
Just for your information... I'm running an BME680, which is working well:
[image]

I compiled myself with this Options in my_user_config.h:
```
  #define USE_BMP                                // [I2cDriver10] Enable BMP085/BMP180/BMP280/BME280 sensors (I2C addresses 0x76 and 0x77) (+4k4 code)
    #define USE_BME680                           // Enable support for BME680 sensor using Bosch BME680 library (+4k code)
```

Jason2866, 2020-03-21:
Short answer, use tasmota-sensors.bin

Chaot1, 2020-03-21:
Ok, working with self compiled firmware.

Thank you for the Dewpoint. Nice Feature.

Thank you all for your work on Tasmota. Great Job.

closed 2020-03-21 by Chaot1

DaniilMir, 2020-03-23:
> Short answer, use tasmota-sensors.bin

There is a problem of egg and chicken. 
sensors.bin has no MQTT TLS suport

s-hadinger, 2020-03-23:
You can self compile tasmota-sensors with AWS IOT

vk3jme, 2020-06-12:
> Just for your information... I'm running an BME680, which is working well:
> [image]
>
> I compiled myself with this Options in my_user_config.h:
>
> ```
>   #define USE_BMP                                // [...]

Hey helmar74,

I need some help - I'm a newbie to compiling Tasmota.
I also have a Tasmota using a BME680 and a TX20.
Mine is a Sonoff TH.
I have had  hard time trying to get compile a firmware to fit both.
Either I can compile Tasmota for one or the other but not both - how did you do it?
What are your overrides?

Please help..

Jamie

deejaybeam, 2020-06-30:
BME280 not working in Tasmota 8.3.1 - using tasmota-sensors.bin, but "I2CScan":"No devices found"

rick-67, 2020-07-18:
HI I upgraded to 8.3.1 and  i have 2 sensor, AM2301 that works fine and BME280 that does not work. I perform an IC2SCAN and output is "{"I2CScan":"Device(s) found at 0x76"}". Can you help me ?

rick-67, 2020-07-19:
Solved...I used Tasmota_sensor. Thanks

Yuriyvp, 2020-09-17:
Same problem with BME680 after migration Tasmota 8.3.1 sensors > Tasmota 8.5 sensors "I2CScan":"No devices found"

Jason2866, 2020-09-18:
If you encounter no i2c devices found your setup or wiring is bad.
This messages does mean there is no i2c device found on the bus.
This is a early stage error message not related to a specific device or driver.

Yuriyvp, 2020-09-18:
In my case it was working for 8 month and lost after migration Tasmota 8.3.1 sensors > Tasmota 8.5 sensors.
Flashing with self compiled 8.5 //define USE_BMP - does not help.
Enabling Disabling SerialLog do not help. Reset does not help.
will chack if BME680 dead.

Yuriyvp, 2020-09-23:
Ok. Downgrade to 7.2 and BME680 apears again. Than upgrade back to > 8.1 > 8.5 fix BME680 discovery problem. Probably some settings after flashing lead to isue....

kmirny, 2020-12-06:
use this define combination in my_user_confg.h:
  #define USE_HTU                                // [I2cDriver9] Enable HTU21/SI7013/SI7020/SI7021 sensor (I2C address 0x40) (+1k5 code)
  #define USE_BMP                                // [I2cDriver10] Enable BMP085/BMP180/BMP280/BME280 sensors (I2C addresses 0x76 and 0x77) (+4k4 code)
    #define USE_BME680 

just keep USE_BMP only doesn't work

tomekc, 2021-12-29:
> use this define combination in my_user_confg.h: #define USE_HTU // [I2cDriver9] Enable HTU21/SI7013/SI7020/SI7021 sensor (I2C address 0x40) (+1k5 code) #define USE_BMP // [I2cDriver10] Enable [...]

It doesn't make sense at first sight, but it worked for me. Thank you! 

Anyone, of course, should check wiring and do `i2cscan` to make sure the sensor is properly connected first.

HombreTriton, 2022-02-21:
Hello,

The problem is present in latest version, 11.0.0.
Sadly I'm stuck on 8.1.0.10

ascillato, 2022-02-21:
Please, use tasmota-sensors.bin. Thanks.