arendst/Tasmota #7970: Support OCF standard (ISO) via integration iotivity-lite library

jkralik, 2020-03-21:
**Have you looked for this feature in other issues and in the docs?**  
I have looked for this in Tasmota, and elsewhere, but have not found anything related. 

**Is your feature request related to a problem? Please describe.**  
The Open Connectivity Foundation is dedicated to ensuring secure interoperability for consumers, businesses and industries by delivering a standard communications platform, a bridging specification,  an open source implementation and a certification program allowing devices to communicate regardless of form factor, operating system, service provider, transport technology or ecosystem.

**Describe the solution you'd like**  
As user I want to have integrated https://github.com/iotivity/iotivity-lite which satisfy OCF standard with enabled cloud support.

**Describe alternatives you've considered**  
There in no other opensource standard yet with ISO. 

**Additional context**  
Specifications: https://openconnectivity.org/developer/specifications
OCF Cloud: https://github.com/go-ocf/cloud

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-21:
> IoT applications may be built for a wide variety of rich and resource-constrained devices across the IoT landscape. As a general guideline, it should be feasible to deploy applications on class 2 constrained devices (>256KB Flash, >64KB RAM), or better.

Not going to happen any time soon on an ESP8266 I guess.

Jason2866, 2020-03-21:
Website not reachable https://www.iotivity.org/get-involved
Is the project alive?

gerritv, 2020-03-21:
Reachable when you drop the www., a sadly common web site misconfiguration.

ascillato2, 2020-03-21:
Thanks for sharing your ideas but this Standard can't be implemented on an ESP8266 using Arduino core due its requirements are higher than the available. On Tasmota-Lite (the smallest version) it leaves 24Kb of free RAM and OCF requires >64KB.

closed 2020-03-21 by ascillato2