arendst/Tasmota #7971: PWM Dimmer no longer working on 8.2.0 - Is functional 8.1.0.11

mhuca, 2020-03-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Prior to 8.2.0 PWM Dimmer was able to control directly attached bulbs. After upgrade to 8.2.0 from 8.1.0.11 the buttons have no effect on the locally attached bulbs.

I've tried the new pre-complied firmware and my self complied firmware both with the same issue.

I've enabled SetOption85 1 and the remote device does respond to the button presses but the locally attached bulbes does not respond.

When reverting to 8.1.0.11 full functionality returns.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Martin Jerry PWM dimmer 
- [X ] Tasmota binary firmware version number used: _8.2.0 tasmota.bin____
  - [X ] Pre-compiled
  - [ X] Self-compiled
    - [ ] IDE / Compiler used: 8.2.0 Firmware.bin_____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:16:29 CMD: Backlog Template; Module; GPIO 255
10:16:29 MQT: FamilyRoomMain/stat/RESULT = {"NAME":"MJ-SD01 Dimmer","GPIO":[19,18,0,59,158,58,0,0,57,37,56,122,29],"FLAG":0,"BASE":73}
10:16:29 MQT: FamilyRoomMain/stat/RESULT = {"Module":{"73":"PWM Dimmer"}}
10:16:29 MQT: FamilyRoomMain/stat/RESULT = {"GPIO0":{"19":"Button3"},"GPIO1":{"18":"Button2"},"GPIO2":{"0":"None"},"GPIO3":{"59":"Led4i"},"GPIO4":{"158":"LedLinki"},"GPIO5":{"58":"Led3i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"57":"Led2i"},"GPIO13":{"37":"PWM1"},"GPIO14":{"56":"Led1i"},"GPIO15":{"122":"Button1i"},"GPIO16":{"29":"Relay1i"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:17:26 CMD: Status 0
10:17:26 MQT: FamilyRoomMain/stat/STATUS = {"Status":{"Module":73,"FriendlyName":["Family Room Main"],"Topic":"FamilyRoomMain","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[6,5,5,6,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"FamilyRoomLights","OtaUrl":"http://10.0.168.24:88/tasmoadmin/firmwares/firmware.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:46","StartupUTC":"2020-03-21T15:08:40","Sleep":50,"CfgHolder":4617,"BootCount":139,"BCResetTime":"2020-02-28T20:39:42","SaveCount":1360,"SaveAddress":"F8000"}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"418/699"}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Humphrey-NoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A0129","0805C8000100060000005A00000000000000","00040200","00000008"]}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS5 = {"StatusNET":{"Hostname":"FamilyRoomMain-1646","IPAddress":"10.0.168.217","Gateway":"10.0.168.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.168.1","Mac":"DC:4F:22:7A:C6:6E","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.168.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7AC66E","MqttUser":"console","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-03-21T15:17:26","Local":"2020-03-21T10:17:26","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"07:05","Sunset":"19:15"}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-21T10:17:26"}}
10:17:26 MQT: FamilyRoomMain/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-21T10:17:26","Uptime":"0T00:08:46","UptimeSec":526,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"PWM":{"PWM1":0},"Wifi":{"AP":1,"SSId":"Humphrey-NoT","BSSId":"1E:80:63:DF:F4:DE","Channel":1,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
10:18:19 MQT: FamilyRoomMain/stat/BUTTON1T = {"TRIG":"TOGGLE"}
10:18:19 MQT: FamilyRoomMain/tele/STATE = {"Time":"2020-03-21T10:18:19","Uptime":"0T00:09:39","UptimeSec":579,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"PWM":{"PWM1":0},"Wifi":{"AP":1,"SSId":"Humphrey-NoT","BSSId":"1E:80:63:DF:F4:DE","Channel":1,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:07"}}
10:18:19 MQT: FamilyRoomMain/stat/RESULT = {"POWER":"OFF"}
10:18:19 MQT: FamilyRoomMain/stat/POWER = OFF (retained)
10:18:20 MQT: FamilyRoomMain/stat/BUTTON1T = {"TRIG":"TOGGLE"}
10:18:20 MQT: FamilyRoomMain/tele/STATE = {"Time":"2020-03-21T10:18:20","Uptime":"0T00:09:40","UptimeSec":580,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"PWM":{"PWM1":0},"Wifi":{"AP":1,"SSId":"Humphrey-NoT","BSSId":"1E:80:63:DF:F4:DE","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:07"}}
10:18:20 MQT: FamilyRoomMain/stat/RESULT = {"POWER":"ON"}
10:18:20 MQT: FamilyRoomMain/stat/POWER = ON (retained)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Press the buttons will toggle and adjust the locally attached bukbs.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-03-21:
I guess it's related to PR https://github.com/arendst/Tasmota/pull/7928 merged 5 days ago.

@pcdiem might chime in.

mhuca, 2020-03-21:
Any additional information/test I can do to assist in diagnoses?

pcdiem, 2020-03-22:
Sorry, I'm on the road right now so I can't investigate. Please check the
preset values using BriPreset. They should be something like 10,255.

Paul C Diem
PCDiem@FoxValley.net.

pcdiem, 2020-03-22:
Also please check SetOption15. It should be 1

Paul C Diem
PCDiem@FoxValley.net

mhuca, 2020-03-22:
Good News...

So BriPreset iss set to 10 ,255

SetOption15 was set OFF. When is is turned on I have more control of the lights. 

I can use the webconsole to turn on and adjust the locally attached lights. However buttons 2 and 3 will change the LED's on the switch but does not adjust the locally attached lights.

Performed a Reset 5 and reconfigured the device and I am functional.

Thank you for the great work,

Mark

BrnEmilio, 2020-05-10:
Hi,

I have the same random problem in tasmota, when I use version 8.2 on the Sonoff device with 3 keys. I set SetOption15 to 1 and SetOption19 to 1, but the error still persists.

The error occurs with event {"TRIG": "TOGGLE"}
:
18:46:46 MQT: stat / ROOM / BUTTON1T = {"TRIG": "TOGGLE"}
18:46:46 MQT: tele / QUARTO / STATE = {"Time": "2020-05-10T18: 46: 46", "Uptime": "0T00: 00: 20", "UptimeSec": 20, "Heap ": 25," SleepMode ":" Dynamic "," Sleep ": 50," LoadAvg ": 19," MqttCount ": 1," POWER1 ":" ON "," POWER2 ":" OFF "," POWER3 ": "OFF", "Wifi": {"AP": 1, "SSId": "4u2", "BSSId": "0C: 80: 63: 46: 79: ED", "Channel": 5, "RSSI" : 100, "Signal": - 42, "LinkCount": 1, "Downtime": "0T00: 00: 06"}}
18:46:46 MQT: stat / QUARTO / RESULT = {"POWER1": "ON"}[](url)

[
[image]
[image]

](url)

mhuca, 2020-05-11:
@BrnEmilio I was able to get everything functional by doing a "reset 5" and then performing a reconfiguration.

I've since updated to 8.2.0.6 and there is much improvements in the stability.

@pcdiem Was you able to find the route cause to this... I will Reopen as it seems to be more than just myself.

Mark

pcdiem, 2020-05-11:
@BrnEmilio I'm not sure what you mean by "but the error still persists". This thread was about the PWM Dimmer having set option 15 to 0 prior to 8.2.0. If doing SetOption15 1 doesn't solve error you're getting, then it is a different issue.

ascillato2, 2020-05-11:
> I was able to get everything functional by doing a "reset 5" and then performing a reconfiguration.

In that case, this issue should be closed.

> If doing SetOption15 1 doesn't solve error you're getting, then it is a different issue.

In that case, please, address this to the chat, and if can't be solved, please open a new issue with the troubleshooting template complete. 

Thanks.

closed 2020-05-11 by ascillato2