arendst/Tasmota #7973: Add a way to reduce the maximum brightness (dimmer 100)

DerNick, 2020-03-21:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_I have a lamp that gets verry hot if it is on at 100% and then it cuts off because of heat_

**Describe the solution you'd like**  
_i would like to have a way to reduce the maximum brightness. (dimmer should still be 0-100 but the real brigtness would be diffrent)_

**Describe alternatives you've considered**  
_an additional option (something like brightnessrange)_

**Additional context**  
_thx for reading ;)_

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-03-21:
You can do this with `RGBWWTable`

DerNick, 2020-03-21:
How? I can only find tutorials for combinding rgb and whith channels. For example if i want to reduce the max. brightness for every channel to 80% how do i do that? thx bro

Jason2866, 2020-03-22:
`SetOption37` reading the docs :-) https://tasmota.github.io/docs/#/Lights

DerNick, 2020-03-23:
Sorry but i think this is only for remapping the Channels isn´t it? Or do you have an example to reduce the max. brightness?

Jason2866, 2020-03-23:
Have you read this too? https://tasmota.github.io/docs/#/Lights?id=white-blend-mode

DerNick, 2020-03-24:
Yeah, but i don´t want to mix RGB and W i just want to reduce the maximum output of  each channel to like 80%. Because my lamp gets too hot if it runs at 100

s-hadinger, 2020-03-24:
Then just do `RGBWWTABLE 200,200,200,200,200`

DerNick, 2020-03-24:
Thank you! It worked :). I tried that before but it didn´t work, now with the newest firmware it works ;).

closed 2020-03-24 by DerNick