arendst/Tasmota #7974: After 8.2.0 Upgrade "BH1750" is now seen as "AM2301 Dew point" sensor.

Ndrinta, 2020-03-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After the upgrade to 8.2.0 the sensor "BH1750" is now seen as "AM2301 Dew point" sensor.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:30:48 MQT: tasmotaTempSensor/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:30:49 MQT: tasmotaTempSensor/stat/RESULT = {"Module":{"0":"Generic"}}
20:30:49 MQT: tasmotaTempSensor/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"2":"AM2301"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:30:03 CMD: Status 0
20:30:03 MQT: tasmotaTempSensor/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota TempSensor"],"Topic":"tasmotaTempSensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:00","StartupUTC":"2020-03-21T20:11:03","Sleep":100,"CfgHolder":4617,"BootCount":72,"BCResetTime":"2020-03-21T18:59:42","SaveCount":118,"SaveAddress":"FB000"}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"397/699"}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MartinRouterKing",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8209","2805C8000100060000005A00000000000000","00000200","00000000"]}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmotaTempSensor","IPAddress":"192.168.1.59","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:7D:79:AB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"TempSensor","MqttClient":"TempSensor","MqttUser":"MQTT-user","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-03-21T20:30:03","Local":"2020-03-21T20:30:03","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+00:00","Sunrise":"05:49","Sunset":"18:04"}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-21T20:30:03","AM2301":{"Temperature":19.1,"Humidity":64.6,"DewPoint":12.3},"TempUnit":"C"}}
20:30:03 MQT: tasmotaTempSensor/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-21T20:30:03","Uptime":"0T00:19:00","UptimeSec":1140,"Heap":23,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"MqttCount":2,"Wifi":{"AP":1,"SSId":"MartinRouterKing","BSSId":"38:2C:4A:CA:92:80","Channel":1,"RSSI":100,"Signal":-25,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-03-21:
If you look at the first point of the new version you will see:

`Change default my_user_config.h driver and sensor support removing most sensors and adding most drivers to tasmota.bin`

As you hare using `{"StatusFWR":{"Version":"8.2.0(tasmota)"` the BH1750 in not there and you must use Tasmota-sensors.bin to have it working.
See [here](https://github.com/arendst/Tasmota/blob/development/BUILDS.md) for the actual list of supported sensors per firmware.

effelle, 2020-03-21:
And by the way, Dew Point is a new sensor embedded in all humidity sensors.

closed 2020-03-21 by effelle

Ndrinta, 2020-03-21:
Oh shoot!
Thanks for the heads up mate. I will now try and after I'll close the topic.

Cheers, have a great weekend.