arendst/Tasmota #7976: GD-30W 300ml Aromatherapy Diffuser doesn't make any mist

blackskull182, 2020-03-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello folks, I have flashed the device following the documentation and everything works great except the "mist mode", I meant that the device doesn't make any mist.

@blakadder I followed the documentation on your repository. Have you tested this device directly? Or can you help me to ask to the people that write this documentation?

[https://templates.blakadder.com/GD-30W.html](url)

Thanks

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): GD-30W 300ml Aromatherapy Diffuser
- [X] Tasmota binary firmware version number used: 8.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

09:51:54 CMD: Backlog Template; Module; GPIO 255
09:51:54 MQT: stat/diffusore/RESULT = {"NAME":"GD-30W","GPIO":[255,255,255,255,255,255,0,0,255,255,255,255,255],"FLAG":0,"BASE":54}
09:51:55 MQT: stat/diffusore/RESULT = {"Module":{"0":"GD-30W"}}
09:51:55 MQT: stat/diffusore/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

09:53:03 MQT: stat/diffusore/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":57,"Rules":"on TuyaReceived#data=55AA000700056E040001007E do publish2 stat/diffusore/EFFECT rgb_cycle endon on TuyaReceived#data=55AA000700056E040001017F do publish2 stat/diffusore/EFFECT color endon on event#rgb_cycle do tuyasend4 110,0 endon on event#color do tuyasend4 110,1 endon on event#ON do backlog tuyasend4 110,1; tuyasend 11,1 endon on event#off do tuyasend1 11,0 endon on power3#state=1 do tuyasend4 103,1 endon on power3#state=0 do tuyasend4 103,0 endon"}
09:53:03 MQT: stat/diffusore/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
09:53:04 MQT: stat/diffusore/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Free":318,"Rules":"on TuyaReceived#data=55AA000700050D040001011E do tuyasend4 13,0 endon on TuyaReceived#data=55AA000700050D040001021F do tuyasend4 13,0 endon on mqtt#connected do serialsend5 55aa0001000000 endon"}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

09:40:14 MQT: stat/diffusore/STATUS = {"Status":{"Module":0,"FriendlyName":["Diffusore di essenze","Luce diffusore di essenze","Nebbia diffusore di essenze","Acqua diffusore di essenze","Modalità luce diffusore di essenze"],"Topic":"diffusore","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:40:14 MQT: stat/diffusore/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T23:27:40","StartupUTC":"2020-03-21T09:12:34","Sleep":50,"CfgHolder":4617,"BootCount":16,"SaveCount":305,"SaveAddress":"FB000"}}
09:40:14 MQT: stat/diffusore/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"494/699"}}
09:40:14 MQT: stat/diffusore/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Waterloo",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","2805C8000100060000005A00000000000000","00010200","00000000"]}}
09:40:14 MQT: stat/diffusore/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
09:40:14 MQT: stat/diffusore/STATUS5 = {"StatusNET":{"Hostname":"diffusore-7549","IPAddress":"XXXXXXXXXXX","Gateway":"XXXXXXXXXXX","Subnetmask":"XXXXXXXXXX","DNSServer":"XXXXXXXXX","Mac":"50:02:91:C0:5D:7D","Webserver":2,"WifiConfig":4}}
09:40:14 MQT: stat/diffusore/STATUS6 = {"StatusMQT":{"MqttHost":"XXXXXXXX","MqttPort":1883,"MqttClientMask":"diffusore","MqttClient":"diffusore","MqttUser":"XXXXXXXX","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:40:14 MQT: stat/diffusore/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 22 08:40:14 2020","Local":"Sun Mar 22 09:40:14 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"06:47","Sunset":"19:06"}}
09:40:14 MQT: stat/diffusore/STATUS10 = {"StatusSNS":{"Time":"2020-03-22T09:40:14"}}
09:40:14 MQT: stat/diffusore/STATUS11 = {"StatusSTS":{"Time":"2020-03-22T09:40:14","Uptime":"0T23:27:40","UptimeSec":84460,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","POWER5":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Waterloo","BSSId":"18:E8:29:CD:74:C0","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}
09:40:16 MQT: tele/diffusore/RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
09:40:27 MQT: tele/diffusore/RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

09:44:04 MQT: tele/diffusore/STATE = {"Time":"2020-03-22T09:44:04","Uptime":"0T23:31:30","UptimeSec":84690,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","POWER5":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Waterloo","BSSId":"18:E8:29:CD:74:C0","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:44:04 MQT: stat/diffusore/RESULT = {"POWER3":"ON"}
09:44:04 MQT: stat/diffusore/POWER3 = ON
09:44:04 RUL: POWER3#STATE=1 performs "tuyasend4 103,1"
09:44:04 MQT: stat/diffusore/RESULT = {"TuyaSend":"Done"}
09:44:04 MQT: tele/diffusore/RESULT = {"TuyaReceived":{"Data":"55AA00070005670400010178","Cmnd":7,"CmndData":"6704000101","DpId":103,"DpIdType":4,"DpIdData":"01"}}
09:44:07 MQT: tele/diffusore/RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
09:44:18 MQT: tele/diffusore/RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
09:44:29 MQT: tele/diffusore/RESULT = {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Activate the "Mist" switch:

dpID 103 mist strength: 0 = low / 1 = high 
Relay3 mist strength status and control using Rule1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The device should make mist when switched ON.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Thanks

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-03-22:
Seems your expectations are drastically different from reality and you should really read through the instructions which I wrote and am fairly certain are clear enough to an average user.

Furthermore you've completely failed to read the issues guidelines hence the closing of it

closed 2020-03-22 by blakadder

blackskull182, 2020-03-23:
> Seems your expectations are drastically different from reality and you should really read through the instructions which I wrote and am fairly certain are clear enough to an average user.
> 
> Furthermore you've completely failed to read the issues guidelines hence the closing of it

Lovely.