arendst/Tasmota #7978: Led lamp flickers visibly at low brightness

BigMike71, 2020-03-22:
### PROBLEM DESCRIPTION
Led lamp flickers visibly at low brightness, sometimes more sometimes less.
I have several Teckin SB53 (and also other brands) where it stands out with a setting of color # 0000000001 or # 0000000002!
When testing different versions, there is no flickering up to version 8.1.0.4, after which this occurs, even with the
latest 8.2.0 release.
in the Arduino Core I found them here:
https://github.com/esp8266/Arduino/issues/7054
https://github.com/esp8266/Arduino/pull/7057

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ]x Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: release 8.2.0_____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-03-22:
This is very annoying, and I think I understand why it's causing flickering. Can you please try 8.2.0 and remove the file `tasmota/core_esp8266_waveform.cpp`?

The problem is that to remove a flickering when multiple PWM are set, it creates a new flickering with a single PWM.

BigMike71, 2020-03-22:
I don't have the opportunity to build it myself, I only use the releases from https://thehackbox.org/tasmota/ and the github releases ...

effelle, 2020-03-22:
@BigMike71 Please test the annexed build compiled with @s-hadinger instructions and report back, thanks.

[firmware.zip](https://github.com/arendst/Tasmota/files/4365586/firmware.zip)

BigMike71, 2020-03-23:
looks good, no more flickering!
I tested each channel individually with 01 or 02 and watched for several minutes.

s-hadinger, 2020-03-23:
This confirms that your flickering was caused by my anti-flickering patch. Oh boy...

I tried to reproduce with my MagicHome led controller, and I can't see any flickering. Are only Teckin flickering or do you see this phenomenon on all your bulbs?

Can you also paste the result of `Status 0`, I have no clue about your configuration.

BigMike71, 2020-03-24:
I can see it with all brands!
have Teckin, Gosund, AWOW, Bakibo, KingSO, ChangM (all are E27 / E17 from 4.5W to 13W) RGBW and RGBCCT

10:44:03 CMD: status 0
10:44:03 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:44:03 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:00:28","StartupUTC":"2020-03-24T09:43:35","Sleep":50,"CfgHolder":4617,"BootCount":37,"BCResetTime":"2020-03-13T17:44:42","SaveCount":331,"SaveAddress":"F5000"}}
10:44:03 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(tasmota)","BuildDateTime":"2020-03-22T13:27:39","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"354/699"}}
10:44:03 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HFDD",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00108009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
10:44:03 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14400B","FlashMode":0,"Features":["00000809","8FDAE397","043683A1","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:44:03 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"Lampe-42-TECKIN","IPAddress":"192.168.1.242","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:ED:35:42","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
10:44:03 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1884,"MqttClientMask":"Lampe-42","MqttClient":"Lampe-42","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:44:03 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-03-24T09:44:03","Local":"2020-03-24T10:44:03","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:43","Sunset":"19:09"}}
10:44:03 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-03-24T10:44:03"}}
10:44:03 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-03-24T10:44:03","Uptime":"0T00:00:28","UptimeSec":28,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":79,"MqttCount":1,"POWER":"ON","Dimmer":1,"Color":"0000000003","HSBColor":"0,0,0","White":1,"CT":500,"Channel":[0,0,0,0,1],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"HFDD","BSSId":"7C:FF:4D:91:3E:DA","Channel":6,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:06"}}}

s-hadinger, 2020-03-24:
I got back my old Teckin SB50, and indeed I can see some flickering at low brightness, but it's hard to see in plain light. Hmmm. I will try experimenting when it's darker.

BigMike71, 2020-03-24:
ok. 
my Teckin SB53 is a big bulb so its easy to see, my others need a darker room too...

s-hadinger, 2020-03-24:
I removed the cap of the SB50, I can't see the flickering anymore...

Can you try with `Fade 0`. There may be a bug here.

BigMike71, 2020-03-24:
i flashed back to 8.2.0 release, fade 0 but the flickering is visibly

Dlay2017, 2020-04-06:
This appears also with Blitzwolf LT21

s-hadinger, 2020-04-21:
@BigMike71 I switched to a new and better PWM implementation from Arduino Core in #8246.
Once it is merged, can you please try this version?

BigMike71, 2020-04-22:
ok i testet so far, a longer test the next days.
seems to be good!
i only see, if i set the PWM Frequency to 223, the light is brighter than set it to higher than 223 !?

I am afraid that the low frequency can cause headaches or something else. there is some experience due to low frequencies in LED light. Experts say a frequency in the kilo heart is only certain. I have this page that deals with it: https://www.derlichtpeter.de/en/light-flicker/

with my smartphone and camera app you can also see that the 223hz flickers quite a lot (thick bars) at 880hz or more, it is significantly better! But the main problem is solved at first, since they are 2 different things!
thanx

ascillato2, 2020-04-22:
> ok i testet so far, a longer test the next days. seems to be good!

Great. Thanks. :+1:

closed 2020-04-22 by ascillato2