arendst/Tasmota #7983: sserialreceived hex string will (nextion) not be recogniced

thefox69, 2020-03-22:
### PROBLEM DESCRIPTION
Via sserial connected touchdisplay (nextion) creates the following serial hexstring by pressing an button: 65 00 01 01 FF FF FF
But tasmota is not recognicing this hex data via SSerialReceived
The first 4 byte seems to work.
But the last FF blocks feeds to non receiving data

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [  ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Generic (18)_____
- [x ] Tasmota binary firmware version number used: 8.2.0.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: OTA
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:20:18 MQT: stat/Eingang_Disp/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:20:18 MQT: stat/Eingang_Disp/RESULT = {"Module":{"18":"Generic"}}
20:20:18 MQT: stat/Eingang_Disp/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"71":"SerBr Tx"},"GPIO14":{"72":"SerBr Rx"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```

- [x ] Provide the output of this command: `Status 0`:
```
20:21:05 MQT: stat/Eingang_Disp/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"Eingang_Disp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:21:05 MQT: stat/Eingang_Disp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:54:44","StartupUTC":"2020-03-22T17:26:21","Sleep":50,"CfgHolder":4617,"BootCount":46,"BCResetTime":"2020-03-14T22:05:16","SaveCount":105,"SaveAddress":"FA000"}}
20:21:05 MQT: stat/Eingang_Disp/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(5471de6-tasmota)","BuildDateTime":"2020-03-22T19:00:51","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"339/699"}}
20:21:05 MQT: stat/Eingang_Disp/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT-2000",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00408009","2805C8000100060000005A00000000000000","00000080","00000000"]}}
20:21:05 MQT: stat/Eingang_Disp/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":16384,"FlashChipId":"1840C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:21:05 MQT: stat/Eingang_Disp/STATUS5 = {"StatusNET":{"Hostname":"Wemos_Eingang_Disp","IPAddress":"192.168.23.49","Gateway":"192.168.23.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.23.254","Mac":"B4:E6:2D:13:EC:1B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:21:05 MQT: stat/Eingang_Disp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.11.222","MqttPort":1883,"MqttClientMask":"Eingang_Disp","MqttClient":"Eingang_Disp","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:21:05 MQT: stat/Eingang_Disp/STATUS7 = {"StatusTIM":{"UTC":"2020-03-22T19:21:05","Local":"2020-03-22T20:21:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:47","Sunset":"19:06"}}
20:21:05 MQT: stat/Eingang_Disp/STATUS10 = {"StatusSNS":{"Time":"2020-03-22T20:21:05"}}
20:21:05 MQT: stat/Eingang_Disp/STATUS11 = {"StatusSTS":{"Time":"2020-03-22T20:21:05","Uptime":"0T01:54:44","UptimeSec":6884,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"IoT-2000","BSSId":"FC:EC:DA:FD:E7:BB","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:40"}}}

- [ x] Provide the output of the Console log output when you experience your issue; if applicable:

sending hex 65 00:
20:27:25 MQT: tele/Eingang_Disp/RESULT = {"SSerialReceived":e}
sending hex 65 00 01 01 (missing bracket at the end)
20:28:41 MQT: tele/Eingang_Disp/RESULT = {"SSerialReceived":e
sending hex 65 00 01 01 FF FF FF
no reaction in console

```

### TO REPRODUCE
-running tasmota on wemos d1 mini
-received the touchscreen hex string on my windows terminal (hterm)
-connected the windows terminal to serial tasmota and tried the different hex strings
-the first bytes are working but the whole string feeds to no reaction on the console

### EXPECTED BEHAVIOUR
sending commands via sserialsend3 is working perfect by ending with \xFF\xFF\xFF
receiving recongition should work also with hex strings ending with FF FF FF

christophsobel, 2020-03-23:
I got a similar issue with precompliled 8.2.0 over OTA on a sonoff basic. Seems like the quotation marks on the SerialReceived value are missing:
Cannot parse {"SerialReceived":A1XXXX20002B}

I downgraded to 8.1.0 and the quotation marks are back:
09:21:09 CMD: SerialSend1 V
09:21:09 MQT: CUL/stat/RESULT = {"SerialSend":"Done"}
09:21:09 MQT: CUL/tele/RESULT = {"SerialReceived":"V 1.67 nanoCUL868"}

Version 7.2.0 is also working fine.

ascillato, 2020-04-01:
According to https://github.com/arendst/Tasmota/issues/8040#issuecomment-606781875 this issue should be resolved too. Please retry with latest Tasmota version from development branch (http://thehackbox.org/tasmota/). Thanks.

ascillato2, 2020-04-02:
Please, when you have time, tell us if your issue is solved. Thanks.

thefox69, 2020-04-03:
Sorry for late reply
updated  to 8.2.0.3, but behaviour still as before.
sending serial hex 0x65 will create an output in the console, but sending the whole hex  string
65 00 01 01 FF FF FF brings no reaction in the console

ascillato2, 2020-04-03:
Hi,

1- You have defined "GPIO13":{"71":"SerBr Tx"},"GPIO14":{"72":"SerBr Rx"}
Have you tried to defined them as SERIAL TX and SERIAL RX instead? (SerBr means Serial to MQTT bridge)

2- Have you tried to use TX and RX pins instead of "GPIO13,"GPIO14" ?

arendst, 2020-04-05:
Try ``SSerialSend5`` which will change the reception to hex.

thefox69, 2020-04-05:
Can confirm Arendst's fix.
If you send strings with SSerialSend5, you'll receive also hex data back.
Topic closed, finally thanks for such good support

closed 2020-04-05 by thefox69

wymanhns, 2020-12-15:
> Try `SSerialSend5` which will change the reception to hex.

i use SSerialSend5 can receive hex data but system restart then cant receive