arendst/Tasmota #7984: Dallas DS18B20 not detected on Tasmota 8.x

dwernerinformatica, 2020-03-22:
### PROBLEM DESCRIPTION
A set of Dallas 18D20 devices are already in use on various nodeMCU and Wemos D1 mini devices with own firmware. One variant is running with parasite mode (two cables One-wire, GND and Vcc solded together), others with 3 wires as shown in the Tasmota documentation https://tasmota.github.io/docs/#/peripherals/DS18x20. The maximum chain is using even more then 10 devices and 20m cable length. However, because of the better universally I wanted to move to Tasmota 8.1 and also 8.2. Tasmota is compiled and uploaded with Code on Linux, the used variant isTasmota sensors setup.  However all setups with Tasmota does not lead to a working 18D20 device recognition, even if all is reduced to a single device.  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): nodeMCU Amica_____
- [x ] Tasmota binary firmware version number used: 8.2_____
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ x] IDE / Compiler used: Code Linux_____
- [x ] Flashing tools used:Code Linux_____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:27:50 CMD: Backlog Template; Module; GPIO 255
22:27:51 MQT: smartHome/AZG/sonoff/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:27:51 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Module":{"0":"Generic"}}
22:27:51 MQT: smartHome/AZG/sonoff/stat/RESULT = {"GPIO0":{"21":"Relay1"},"GPIO1":{"0":"None"},"GPIO2":{"4":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"4":"DS18x20"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  22:28:53 CMD: Backlog Rule1; Rule2; Rule3
22:28:53 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
22:28:54 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
22:28:54 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [x ] Provide the output of this command: `Status 0`:
```
 22:29:36 CMD: STATUS 0
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["nodemcu1"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:13:43","StartupUTC":"2020-03-22T20:15:53","Sleep":50,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2020-03-21T18:20:15","SaveCount":54,"SaveAddress":"F7000"}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0.8(sensors)","BuildDateTime":"2020.03.22 20:44:29","Boot":5,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"403/699"}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["L3W34-0","7490wz"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","01000000","00000000"]}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000407","8FDAE387","0437A3A4","B7FFBFCD","01DA9BC0","04B6FCC7","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30,32","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,34,37,39,40,42,43,51,52,55,56,58,59,62"}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"nodemcu1","IPAddress":"192.168.2.170","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.2","Mac":"A0:20:A6:0A:CA:9D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.3","MqttPort":1883,"MqttClientMask":"nodemcu1","MqttClient":"nodemcu1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 22 21:29:36 2020","Local":"Sun Mar 22 22:29:36 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"06:47","Sunset":"19:06"}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-22T22:29:36"}}
22:29:36 MQT: smartHome/AZG/sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-22T22:29:36","Uptime":"0T01:13:43","UptimeSec":4423,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"L3W34-0","BSSId":"34:31:C4:0E:94:1D","Channel":1,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:07"}}}
22:29:36 MQT: smartHome/AZG/sonoff/tele/STATE = {"Time":"2020-03-22T22:29:36","Uptime":"0T01:13:43","UptimeSec":4423,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"L3W34-0","BSSId":"34:31:C4:0E:94:1D","Channel":1,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:07"}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
22:31:02 CMD: weblog 4
22:31:02 MQT: smartHome/AZG/sonoff/stat/RESULT = {"WebLog":4}
22:31:03 CFG: in Flash gespeichert am F6, zählen 55, Bytes 4096
22:31:04 WIF: Prüfe Verbindung...
22:31:06 MQT: smartHome/AZG/sonoff/tele/STATE = {"Time":"2020-03-22T22:31:06","Uptime":"0T01:15:13","UptimeSec":4513,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"L3W34-0","BSSId":"34:31:C4:0E:94:1D","Channel":1,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:31:09 WIF: Prüfe Verbindung...
22:31:14 WIF: Prüfe Verbindung...
22:31:19 WIF: Prüfe Verbindung...
```

### TO REPRODUCEr
_Steps to reproduce the behavior:_
Just attach DS18B20 as described in https://tasmota.github.io/docs/#/peripherals/DS18x20

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Normally (and I remember that at in earlier version I have tested this  in 3 wire One-Wire setup) I expect the single or multiple  temperature sensor(s) be detected and temperature values are shown and published

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jansene, 2020-03-22:
your build is not using the latest core: "Core":"2_6_1","SDK":"2.2.2-dev(38a443e)"
See resolution: Temperature Sensor DS18B20 not detected #7879

dwernerinformatica, 2020-03-22:
Meanwhile I have tried again to ensure having my code based platformio  and tasmota branch  updated to the latest, but also that did not lead to a working detection of the 18B20 devices

Here again the updated results

```
23:24:31 CMD: Backlog Template; Module; GPIO 255
23:24:31 SRC: WebConsole from 192.168.2.200
23:24:31 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
23:24:31 SRC: Backlog
23:24:31 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
23:24:31 MQT: smartHome/AZG/sonoff/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:24:32 SRC: Backlog
23:24:32 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
23:24:32 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Module":{"0":"Generic"}}
23:24:32 SRC: Backlog
23:24:32 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
23:24:32 MQT: smartHome/AZG/sonoff/stat/RESULT = {"GPIO0":{"21":"Relay1"},"GPIO1":{"0":"None"},"GPIO2":{"4":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"4":"DS18x20"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
```
23:27:02 SRC: Backlog
23:27:02 CMD: Gruppe 0, Index 1, Befehl "RULE", Daten ""
23:27:02 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:27:02 SRC: Backlog
23:27:02 CMD: Gruppe 0, Index 2, Befehl "RULE", Daten ""
23:27:02 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:27:02 SRC: Backlog
23:27:02 CMD: Gruppe 0, Index 3, Befehl "RULE", Daten ""
23:27:02 MQT: smartHome/AZG/sonoff/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:27:06 WIF: Prüfe Verbindung...
```

```
23:28:03 CMD: Status 0
23:28:03 SRC: WebConsole from 192.168.2.200
23:28:03 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["nodemcu1"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:11:25","StartupUTC":"2020-03-22T22:16:38","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2020-03-21T18:20:15","SaveCount":58,"SaveAddress":"FA000"}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0.1(sensors)","BuildDateTime":"2020.03.22 23:13:06","Boot":5,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"406/699"}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["L3W34-0","7490wz"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","01000000","00000000"]}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000407","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS5 = {"StatusNET":{"Hostname":"nodemcu1","IPAddress":"192.168.2.170","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.2","Mac":"A0:20:A6:0A:CA:9D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.3","MqttPort":1883,"MqttClientMask":"nodemcu1","MqttClient":"nodemcu1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-03-22T22:28:03","Local":"2020-03-22T23:28:03","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:47","Sunset":"19:06"}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS10 = {"StatusSNS":{"Time":"2020-03-22T23:28:03"}}
23:28:03 MQT: smartHome/AZG/sonoff/stat/STATUS11 = {"StatusSTS":{"Time":"2020-03-22T23:28:03","Uptime":"0T00:11:25","UptimeSec":685,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"L3W34-0","BSSId":"34:31:C4:0E:94:1D","Channel":1,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:07"}}}
23:28:06 WIF: Prüfe Verbindung...
```

```
23:28:49 CMD: weblog 4
23:28:49 SRC: WebConsole from 192.168.2.200
23:28:49 CMD: Gruppe 0, Index 1, Befehl "WEBLOG", Daten "4"
23:28:49 MQT: smartHome/AZG/sonoff/stat/RESULT = {"WebLog":4}
23:28:51 MQT: smartHome/AZG/sonoff/tele/STATE = {"Time":"2020-03-22T23:28:51","Uptime":"0T00:12:13","UptimeSec":733,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"L3W34-0","BSSId":"34:31:C4:0E:94:1D","Channel":1,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:07"}}
```

dwernerinformatica, 2020-03-22:
> your build is not using the latest core: "Core":"2_6_1","SDK":"2.2.2-dev(38a443e)"; needs to be
> STAGE/2.2.2-dev(38a443e)
> 
> See resolution: Temperature Sensor DS18B20 not detected #7879

Sorry, I'm not sure how to get from "SDK":"2.2.2-dev(38a443e)" to STAGE/2.2.2-dev(38a443e).  Can you give me a hint for that?

jansene, 2020-03-22:
The first output of status shows "Core":"2_6_1","SDK":"2.2.2-dev(38a443e)".
Looks your newest build satus 0 reports: "STAGE","SDK":"2.2.2-dev(38a443e).
Try 'setoption74 1' to enable internal pull up.

arendst, 2020-03-23:
DS18B20 is only supported on ONE gpio. Remove from gpio2 and try again.

dwernerinformatica, 2020-03-23:
Yes, that finally was it. I meanwhile had already moved to latest Master brach (that was obviosly needed to get to the STAGE version)
```

Program Version | 8.2.0.1(sensors)
-- | --
2020-03-22T23:57:43
STAGE/2.2.2-dev(38a443e)
```
SetOption 74 was also already on, even if I don't really understand the purpose of it as I had soldered a 4,7k resistor from V++ to G4 (data line). Wouldn't that act as such pull up resistor?

I also saw while investigating all this that meanwhile the is a pending pull to support parasite mode as well, which would be great to replace the 3 other  Wemos D1 with self-written Firmware as all the floor heating sensors used parasite mode because of the simpler wireing (2 wire instead of 3).

I really like Tasmota as most of the other stuff is already there in a quality I would need years to add, so with support of parasite mode I could immediately turn the D1 all to Tasmota and MQTT, as the current used http was not 100%  stable.

Is there already an ETA to get that in the main trunk?

dwernerinformatica, 2020-03-23:
Issue is now closed:
Main solution
1. Upgrade to latest Master branch and upgrade all dependent libs in Visual Code
2. Reduce number of defined Ports with 18B20 to only one
(3. Proposal of SetOption74 did not influence that, as finally the device worked with or without option set)
Issue is  closed

closed 2020-03-23 by dwernerinformatica

dwernerinformatica, 2020-03-23:
That is really funny: I used a second nodeMCU but without solderer 4,7k resistor between Vcc and GPIO4 and it worked as well, regardless of SetOption74 setting? May be some of the exports could give a hint why?
And I also have not found what the difference between DS18x20 and DS18x20o is....

effelle, 2020-03-23:
Because the gpio current when Tasmota checked it was in the correct range. 
Trust me, before or after the data pin will fail to send/ receive the correct state. 
> And I also have not found what the difference between DS18x20 and DS18x20o is....

`DS18x20o` and `DHTo` are special sensors for the Shelly Addon.

yann1420, 2020-05-13:
It should be documented in https://tasmota.github.io/docs/  that one  can only use ONE GPIO for a string of DS18B20's and also there is a coded soft-limit of 8x DS18B20's on that string. I wonder if any ever tried what the actual limit is of there temp-sensors. I would like to have >=16 per tasmota.

lalo-uy, 2020-05-13:
With a 820R pullup and external power in the sensors I had got 20 working
on a Wemos D1, but not with Tasmota.
I belive that the philosophy of iot is to have several devices WiFi
connected and not one device wired around the house.

yann1420, 2020-05-13:
I'm using a wemos D1 as well. I have figured out the maximum # of DS18B20 is 8 in tasmota. This is due to a definition in the file _tasmota/xsns_05_ds18x20.ino_
**#define DS18X20_MAX_SENSORS  8**

I'l try to compile a private tasmota.bin with a different limit and good to know to decrease the pull-up (from an electronic point of that is to be expected :)