arendst/Tasmota #7985: Rule doesn't stop on BREAK, all Rules are executed one after another

Cyber1000, 2020-03-22:
### PROBLEM DESCRIPTION
I've adjusted a solution to my use case from here https://github.com/arendst/Tasmota/issues/5897#issuecomment-497786077
Following is a simplified repro:
`Rule1 ON ENERGY#Power>=30 DO Backlog PowerDelta 15 BREAK ON ENERGY#Power>5 DO Backlog PowerDelta 25 ENDON`
I'm testing with a Gosund SP111 and a light bulb at 60W.
I would expect that only the rule with Power>=30 gets executed, but both get executed one after another
I'm using this as a workaround since I get a lot of traffic (my tasmota-device jumps a lot in low power-ranges), I came here from a similar problem here: #7684
Asked on discord but no users with deep knowledge in rules were online.

My log:

> 22:18:24 MQT: stat/DVES_5A0D86/RESULT = {"POWER":"ON"}
> 22:18:24 MQT: stat/DVES_5A0D86/POWER = ON
> 22:18:26 RUL: ENERGY#POWER>=30 performs "Backlog PowerDelta 15"
> 22:18:26 MQT: stat/DVES_5A0D86/RESULT = {"PowerDelta":15}
> 22:18:26 RUL: ENERGY#POWER>5 performs "Backlog PowerDelta 25"
> 22:18:27 MQT: stat/DVES_5A0D86/RESULT = {"PowerDelta":25}
> 22:18:27 MQT: tele/DVES_5A0D86/SENSOR = {"Time":"2020-03-22T22:18:27","ENERGY":{"TotalStartTime":"2020-03-22T21:52:30","Total":0.048,"Yesterday":0.000,"Today":0.048,"Power":62,"ApparentPower":73,"ReactivePower":38,"Factor":0.85,"Voltage":226,"Current":0.324}}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
Even chatted there, but found no solution
- [X] Device used (e.g., Sonoff Basic): Gosund sp111
- [X] Tasmota binary firmware version number used: 8.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: https://github.com/ct-Open-Source/tuya-convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
Configuration output here:
23:56:36 MQT: stat/DVES_5A0D86/RESULT = {"NAME":"Gosund SP111","GPIO":[56,0,57,0,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":18}
23:56:36 MQT: stat/DVES_5A0D86/RESULT = {"Module":{"0":"Gosund SP111"}}
23:56:36 MQT: stat/DVES_5A0D86/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"57":"Led2i"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"134":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"17":"Button1"},"GPIO14":{"132":"HLWBL CF1"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
Rules output here:
23:58:38 MQT: stat/DVES_5A0D86/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":412,"Rules":"ON ENERGY#Power>=30 DO Backlog PowerDelta 15 BREAK ON ENERGY#Power>5 DO Backlog PowerDelta 25 ENDON"}
23:58:39 MQT: stat/DVES_5A0D86/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:58:39 MQT: stat/DVES_5A0D86/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:59:04 MQT: stat/DVES_5A0D86/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"DVES_5A0D86","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:03:18","StartupUTC":"2020-03-22T20:55:46","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2020-03-22T21:52:30","SaveCount":56,"SaveAddress":"F4000"}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":4,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"379/699"}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cyber1000-SmartHome",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS5 = {"StatusNET":{"Hostname":"DVES_5A0D86-3462","IPAddress":"192.168.11.217","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"B4:E6:2D:5A:0D:86","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.11.22","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A0D86","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS7 = {"StatusTIM":{"UTC":"2020-03-22T22:59:04","Local":"2020-03-22T23:59:04","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:47","Sunset":"19:06"}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS9 = {"StatusPTH":{"PowerDelta":15,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS10 = {"StatusSNS":{"Time":"2020-03-22T23:59:04","ENERGY":{"TotalStartTime":"2020-03-22T21:52:30","Total":0.056,"Yesterday":0.000,"Today":0.056,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
23:59:04 MQT: stat/DVES_5A0D86/STATUS11 = {"StatusSTS":{"Time":"2020-03-22T23:59:04","Uptime":"0T02:03:18","UptimeSec":7398,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Cyber1000-SmartHome","BSSId":"00:25:9C:13:C2:46","Channel":13,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:08 MQT: stat/DVES_5A0D86/RESULT = {"POWER":"ON"}
00:00:08 MQT: stat/DVES_5A0D86/POWER = ON
00:00:09 CFG: Saved to flash at F9, Count 59, Bytes 4096
00:00:10 RUL: ENERGY#POWER>=30 performs "Backlog PowerDelta 15"
00:00:10 SRC: Rule
00:00:10 CMD: Group 0, Index 1, Command "BACKLOG", Data "PowerDelta 15"
00:00:10 SRC: Backlog
00:00:10 CMD: Group 0, Index 1, Command "POWERDELTA", Data "15"
00:00:10 MQT: stat/DVES_5A0D86/RESULT = {"PowerDelta":15}
00:00:11 RUL: ENERGY#POWER>5 performs "Backlog PowerDelta 25"
00:00:11 SRC: Rule
00:00:11 CMD: Group 0, Index 1, Command "BACKLOG", Data "PowerDelta 25"
00:00:11 SRC: Backlog
00:00:11 CMD: Group 0, Index 1, Command "POWERDELTA", Data "25"
00:00:11 MQT: stat/DVES_5A0D86/RESULT = {"PowerDelta":25}
00:00:11 MQT: tele/DVES_5A0D86/SENSOR = {"Time":"2020-03-23T00:00:11","ENERGY":{"TotalStartTime":"2020-03-22T21:52:30","Total":0.056,"Yesterday":0.056,"Today":0.000,"Power":65,"ApparentPower":75,"ReactivePower":38,"Factor":0.86,"Voltage":226,"Current":0.333}}
00:00:12 CFG: Saved to flash at F8, Count 60, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Rule1 5
- `Rule1 ON ENERGY#Power>=30 DO Backlog PowerDelta 15 BREAK ON ENERGY#Power>5 DO Backlog PowerDelta 25 ENDON`
- Rule1 1

### EXPECTED BEHAVIOUR
- If I turn on a bulb of 60W I would expect that only rule "ENERGY#Power>=30" is executed, but both are executed (first >=30, than >5)
- I would expect the BREAK stops the second rule

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Cyber1000, 2020-03-22:
@ascillato Since I'm seeing you have done #4500, do you see what's happening here?

Cyber1000, 2020-03-23:
Ok seems that the solution from here https://github.com/arendst/Tasmota/issues/5897#issuecomment-497786077 doesn't work.
If I do Rule1 4 (disable one-shot) it works but triggers all the time, if I do Rule1 5 (use one-shot) it seems to trigger all Rules in the ruleset.

Cyber1000, 2020-03-23:
I've found a workaround:
```
Backlog Rule1 4; Rule2 5
Rule1 ON ENERGY#Power>=30 DO Var1 1 BREAK ON ENERGY#Power>5 DO Var1 2 ENDON
Rule2 ON Var1#State==1 DO PowerDelta 15 ENDON ON Var1#State==2 DO PowerDelta 25 ENDON
Backlog Rule1 1; Rule2 1
```

So the command validating the power is running all the time, which sets a variable Var1, it breaks correctly with BREAK. Rule2 is one-shot, so it detects only changes of Var1 and executes things according to this.

It's not the best workaround in my life, but it works 😄

ascillato2, 2020-04-01:
Hi,

Sorry for the late response.

The rule syntax is wrong in your first post:

```lua
rule1 ON ENERGY#Power>=30 DO Backlog PowerDelta 15 BREAK ON ENERGY#Power>5 DO Backlog PowerDelta 25 ENDON
```
It should be:

```lua
rule1 ON ENERGY#Power>=30 DO PowerDelta 15 BREAK ON ENERGY#Power>5 DO PowerDelta 25 ENDON
```

closed 2020-04-01 by ascillato2

Cyber1000, 2020-05-24:
Hi sorry to come back after a long time, I've a overlooked this response and since my workaround still works it wasn't that important for me ...

And my example was oversimplified, it should look more like:
```
Rule1 ON ENERGY#Power>=30 DO Backlog PowerDelta 15; TelePeriod 3000 BREAK ON ENERGY#Power>5 DO Backlog PowerDelta 25; TelePeriod 60 ENDON
```

I tried your example out (without Backlog) some time ago, I'm not sure if that worked, though I needed more commands executed once, my workaround is more like:
```
Backlog Rule1 4; Rule2 5
Rule1 ON ENERGY#Power>=30 DO Var1 1 BREAK ON ENERGY#Power>5 DO Var1 2 ENDON
Rule2 ON Var1#State==1 DO Backlog PowerDelta 15; TelePeriod 3000 ENDON ON Var1#State==2 DO Backlog PowerDelta 25; TelePeriod 60 ENDON
Backlog Rule1 1; Rule2 1
```

And as you correctly state the "BREAK ON" doesn't work with Backlog as I expected while writing this issue. 
**My workaround splits that into a rule with break on separated ifs (without backlog) and a rule with endon separated ifs (with backlog).**

Just if someone else has similar expectations and wants a workaround ...

dragospascale, 2021-06-03:
I don't understand why the problem is closed because I can confirm everything that Cyber1000 has presented here.
BREAK is so important in RULES when dealing with sensor values in ranges.

ascillato, 2021-06-03:
@dragospascale

Please, update to latest version and try again your rules. If they don't work as expected, please, open a new issue and share them in order to help you. Thanks.

mrneutron42, 2024-02-06:
2.5 years later...it's still an issue in 13.3.0.  
BREAK does NOT stop the execution of the remaining rules that follow within the rule set.

sfromis, 2024-02-06:
As long as you understand the flow of events and what one-shot does, using `break` *will* work; as such there is no issue to be fixed. Sure, rules can be tricky to understand, but that does not mean that trying something, and not seeing it work, means that rules are broken. Only that you need grok to flow around rules, which is indeed not very intuitive.

Worst starting point is to try things the way you're used from programming languages. Rules is a rather different thing.